ARMmbed/mbed-os #11803: The bug of MbedOS CoAP library

TheSilentDawn, 2019-11-04:
### Description of defect
Reference: [https://github.com/ARMmbed/mbed-os/tree/master/features/frameworks/mbed-coap](https://github.com/ARMmbed/mbed-os/tree/master/features/frameworks/mbed-coap)
Function: sn_coap_parser_options_parse
[https://github.com/ARMmbed/mbed-os/blob/d91ed5fa42ea0f32e4422a3c562e7b045a17da40/features/frameworks/mbed-coap/source/sn_coap_parser.c#L301](https://github.com/ARMmbed/mbed-os/blob/d91ed5fa42ea0f32e4422a3c562e7b045a17da40/features/frameworks/mbed-coap/source/sn_coap_parser.c#L301)
The CoAP parser is responsible for parsing received CoAP packets. The function sn_coap_parser_options_parse() parses CoAP input linearly using a while loop. Once an option is parsed in a loop, the current point (*packet_data_pptr) is increased correspondingly. The pointer is restricted by the size of the received buffer, as well as a delimiter byte 0xFF, as shown in line 4 of the code snippet below.
```c
static int8_t sn_coap_parser_options_parse(..., uint8_t **packet_data_pptr, ...)
{
...
    while (message_left && (**packet_data_pptr != 0xff)) {
       	...
       	if (option_len == 13) {
           	option_len = *(*packet_data_pptr + 1) + 13;
           	(*packet_data_pptr)++;
       	}
       	...
    }
...
}
```
Unfortunately, inside each while loop, the check of the value of *packet_data_pptr is not strictly enforced. More specifically, inside a loop, *packet_data_pptr could be increased and then dereferenced without checking. Moreover, there are many other functions in the format of sn_coap_parser_****() that do not check whether the pointer is within the bound of the allocated buffer. All of these lead to heap or stack buffer overflow, depending on how the CoAP packet buffer is allocated.
In the following, we list other locations which cause out-of-bound memory accesses rooted in this vulnerability. 
https://github.com/ARMmbed/mbed-os/blob/d91ed5fa42ea0f32e4422a3c562e7b045a17da40/features/frameworks/mbed-coap/source/sn_coap_parser.c#L660
https://github.com/ARMmbed/mbed-os/blob/d91ed5fa42ea0f32e4422a3c562e7b045a17da40/features/frameworks/mbed-coap/source/sn_coap_parser.c#L331
https://github.com/ARMmbed/mbed-os/blob/d91ed5fa42ea0f32e4422a3c562e7b045a17da40/features/frameworks/mbed-coap/source/sn_coap_parser.c#L257
https://github.com/ARMmbed/mbed-os/blob/d91ed5fa42ea0f32e4422a3c562e7b045a17da40/features/frameworks/mbed-coap/source/sn_coap_parser.c#L310
https://github.com/ARMmbed/mbed-os/blob/d91ed5fa42ea0f32e4422a3c562e7b045a17da40/features/frameworks/mbed-coap/source/sn_coap_parser.c#L313
https://github.com/ARMmbed/mbed-os/blob/d0686fd30b4d3d02efdc7e4d0fbf0dfe173543b6/features/frameworks/mbed-coap/source/sn_coap_protocol.c#L2488

#### Target(s) affected by this defect ?
MbedOS CoAP library

#### Toolchain(s) (name and version) displaying this defect ?
N/A

#### What version of Mbed-os are you using (tag or sha) ?
MbedOS 5.13.2

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
mbed-cli latest version

#### How is this defect reproduced ? 
Using bug_coap_1_1, bug_coap_1_2, bug_coap_1_3, bug_coap_1_4， bug_coap_1_5 and bug_coap_1_6 as input of demo codes below, previous crash examples in sn_coap_parser.c will be triggered.
```c
struct coap_s* coapHandle;
coap_version_e coapVersion = COAP_VERSION_1;
void* coap_malloc(uint16_t size){
    return malloc(size);
}
void coap_free(void* addr){
    free(addr);
}
uint8_t coap_tx_cb(uint8_t *a, uint16_t b, sn_nsdl_addr_s *c, void *d){
    debug_log("coap_tx_cb", sizeof("coap_tx_cb"));
    return 0;
}
int8_t coap_rx_cb(sn_coap_hdr_s *a, sn_nsdl_addr_s *b, void *c){
    debug_log("coap_rx_cb", sizeof("coap_rx_cb"));
    return 0;
}
void fuzz_parser() {
    coapHandle = sn_coap_protocol_init(&coap_malloc, &coap_free, &coap_tx_cb, &coap_rx_cb);
    nsapi_size_or_error_t ret;
    uint8_t* recv_buffer = (uint8_t*)malloc(1280);
    memset(recv_buffer, 0x0, 1280);
    // stack overflow need to change this operation
    // to allocate memory in stack, for example, uint8_t recv_buffer[1280]
    ret = read(0, recv_buffer, 1280);
    sn_coap_hdr_s* parsed = sn_coap_parser(coapHandle, ret, recv_buffer, &coapVersion);
}
```
[bug_coap_1_1.log](https://github.com/ARMmbed/mbed-os/files/3802577/bug_coap_1_1.log)
[bug_coap_1_2.log](https://github.com/ARMmbed/mbed-os/files/3802578/bug_coap_1_2.log)
[bug_coap_1_3.log](https://github.com/ARMmbed/mbed-os/files/3802579/bug_coap_1_3.log)
[bug_coap_1_4.log](https://github.com/ARMmbed/mbed-os/files/3802580/bug_coap_1_4.log)
[bug_coap_1_5.log](https://github.com/ARMmbed/mbed-os/files/3802581/bug_coap_1_5.log)
[bug_coap_1_6.log](https://github.com/ARMmbed/mbed-os/files/3802582/bug_coap_1_6.log)

0xc0170, 2019-11-04:
cc @anttiylitokola

adbridge, 2019-11-06:
@TheSilentDawn Could you please update the title of your issues to actually describe what the bug is rather than just 'bug 1' etc? Thanks

TheSilentDawn, 2019-11-06:
@adbridge No problem.

JanneKiiskila, 2019-12-18:
Thank you for this error report, this will be fixed in an upcoming release.

teetak01, 2020-04-03:
Thanks @TheSilentDawn 

This issue was fixed as part of Mbed OS 5.15.1 release.

https://github.com/ARMmbed/mbed-os/pull/12146

Closing as fixed.

closed 2020-04-03 by teetak01

elanzini, 2020-10-14:
Is it possible to have #12146 as a `Linked pull request` to this issue?
It helps when tracing back what was changed in the fix since [NVD](https://nvd.nist.gov/vuln/detail/CVE-2019-17212) only reports a link to this issue.