ARMmbed/mbed-os #11804: The bug2 of MbedOS CoAP library

TheSilentDawn, 2019-11-04:
### Description of defect
Reference: [https://github.com/ARMmbed/mbed-os/tree/master/features/frameworks/mbed-coap](https://github.com/ARMmbed/mbed-os/tree/master/features/frameworks/mbed-coap)
Function: sn_coap_builder_calc_needed_packet_data_size_2
https://github.com/ARMmbed/mbed-os/blob/d0686fd30b4d3d02efdc7e4d0fbf0dfe173543b6/features/frameworks/mbed-coap/source/sn_coap_builder.c#L355
The CoAP builder is responsible for crafting outgoing CoAP messages. The function sn_coap_builder_calc_needed_packet_data_size_2() is used to calculate the needed memory for the CoAP message from the sn_coap_hdr_s data structure. Both returned_byte_count and src_coap_msg_ptr->payload_len are of type uint16_t. When added together, the result  returned_byte_count will wrap around the maximum as shown in line 4. As a result, insufficient buffer is allocated for the corresponding CoAP message. 
```c
uint16_t sn_coap_builder_calc_needed_packet_data_size_2(const sn_coap_hdr_s *src_coap_msg_ptr, ...)
{
...
       returned_byte_count += src_coap_msg_ptr->payload_len;
...
}
When the data in the sn_coap_hdr_s is copied into the allocated buffer, out-of-bound memory access will happen (line 4).
```c
static int16_t sn_coap_builder_options_build_add_one_option(..., uint16_t option_len, const uint8_t *option_ptr, ...)
{
...
    	memcpy(dest_packet, option_ptr, option_len);
...
}
```
In the following, we list other locations which will cause out-of-bound memory accesses rooted in this vulnerability.
https://github.com/ARMmbed/mbed-os/blob/d0686fd30b4d3d02efdc7e4d0fbf0dfe173543b6/features/frameworks/mbed-coap/source/sn_coap_builder.c#L1090
https://github.com/ARMmbed/mbed-os/blob/d0686fd30b4d3d02efdc7e4d0fbf0dfe173543b6/features/frameworks/mbed-coap/source/sn_coap_builder.c#L710
https://github.com/ARMmbed/mbed-os/blob/d0686fd30b4d3d02efdc7e4d0fbf0dfe173543b6/features/frameworks/mbed-coap/source/sn_coap_builder.c#L524
https://github.com/ARMmbed/mbed-os/blob/d0686fd30b4d3d02efdc7e4d0fbf0dfe173543b6/features/frameworks/mbed-coap/source/sn_coap_builder.c#L527
https://github.com/ARMmbed/mbed-os/blob/d0686fd30b4d3d02efdc7e4d0fbf0dfe173543b6/features/frameworks/mbed-coap/source/sn_coap_builder.c#L528
https://github.com/ARMmbed/mbed-os/blob/d0686fd30b4d3d02efdc7e4d0fbf0dfe173543b6/features/frameworks/mbed-coap/source/sn_coap_builder.c#L718
https://github.com/ARMmbed/mbed-os/blob/d0686fd30b4d3d02efdc7e4d0fbf0dfe173543b6/features/frameworks/mbed-coap/source/sn_coap_builder.c#L746

#### Target(s) affected by this defect ?
MbedOS CoAP library

#### Toolchain(s) (name and version) displaying this defect ?
N/A

#### What version of Mbed-os are you using (tag or sha) ?
MbedOS 5.13.2

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
mbed-cli latest version

#### How is this defect reproduced ? 
Use bug_coap_2_1, bug_coap_2_2, bug_coap_2_3_1, bug_coap_2_3_2, bug_coap_2_3_3, bug_coap_2_4 and bug_coap_2_5 as input of demo code in [Bug 1](https://github.com/ARMmbed/mbed-os/issues/11803) will trigger bugs.
[bug_coap_2_1.log](https://github.com/ARMmbed/mbed-os/files/3802603/bug_coap_2_1.log)
[bug_coap_2_2.log](https://github.com/ARMmbed/mbed-os/files/3802602/bug_coap_2_2.log)
[bug_coap_2_3_1.log](https://github.com/ARMmbed/mbed-os/files/3802601/bug_coap_2_3_1.log)
[bug_coap_2_3_2.log](https://github.com/ARMmbed/mbed-os/files/3802600/bug_coap_2_3_2.log)
[bug_coap_2_3_3.log](https://github.com/ARMmbed/mbed-os/files/3802606/bug_coap_2_3_3.log)
[bug_coap_2_4.log](https://github.com/ARMmbed/mbed-os/files/3802605/bug_coap_2_4.log)
[bug_coap_2_5.log](https://github.com/ARMmbed/mbed-os/files/3802604/bug_coap_2_5.log)

0xc0170, 2019-11-04:
cc @anttiylitokola

JanneKiiskila, 2019-12-18:
Thank you for this error report, this will be fixed in an upcoming release.

teetak01, 2020-04-03:
Thanks @TheSilentDawn 

This issue was fixed as part of Mbed OS 5.15.1 release.

https://github.com/ARMmbed/mbed-os/pull/12147

Closing as fixed.

closed 2020-04-03 by teetak01