ARMmbed/mbed-os #11805: The bug1 of MbedOS Client Cli library

TheSilentDawn, 2019-11-04:
### Description of defect
Reference: [https://github.com/ARMmbed/mbed-os/tree/master/features/frameworks/mbed-client-cli](https://github.com/ARMmbed/mbed-os/tree/master/features/frameworks/mbed-client-cli)
Function: next_command
https://github.com/ARMmbed/mbed-os/blob/c385e141e2c2155e92d88cd99fc0f675d900d102/features/frameworks/mbed-client-cli/source/ns_cmdline.c#L834
The Mbed-Client-Cli library parses the command list from a user, which contains a string with semicolons separating each command.
The library expects a null character at the end of the string. If the command list has no null character, the library will continue parsing characters following the end of the command list. Since the content of the memory after the command list is undefined, unpredictable behavior could be observed. For example, *ptr may not be zero even at the end of string_ptr and the while loop will continue to execute as shown in line 4.
```c
static char *next_command(char *string_ptr, operator_t *oper)
{
...
   	while (*ptr != 0) {
        switch (*ptr) {
        ...
        case ('|'):
        	if (ptr[1] == '|') {
            ...
        	}
        }
       	ptr++;
...
}
```
In the following, we list other locations which will cause out-of-bound memory accesses rooted in this vulnerability.
https://github.com/ARMmbed/mbed-os/blob/d91ed5fa42ea0f32e4422a3c562e7b045a17da40/features/frameworks/mbed-client-cli/source/ns_cmdline.c#L838
https://github.com/ARMmbed/mbed-os/blob/d91ed5fa42ea0f32e4422a3c562e7b045a17da40/features/frameworks/mbed-client-cli/source/ns_cmdline.c#L872
https://github.com/ARMmbed/mbed-os/blob/d91ed5fa42ea0f32e4422a3c562e7b045a17da40/features/frameworks/mbed-client-cli/source/ns_cmdline.c#L858
https://github.com/ARMmbed/mbed-os/blob/d91ed5fa42ea0f32e4422a3c562e7b045a17da40/features/frameworks/mbed-client-cli/source/ns_cmdline.c#L581

#### Target(s) affected by this defect ?
MbedOS Client Cli library

#### Toolchain(s) (name and version) displaying this defect ?
N/A

#### What version of Mbed-os are you using (tag or sha) ?
MbedOS 5.13.2

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
mbed-cli latest version

#### How is this defect reproduced ? 
Use bug_cli_1_1, bug_cli_1_2, bug_cli_1_3 and bug_cli_1_4 as input to demo codes below. And debug program with checking value of command string *ptr
```c
// The code is based on the demo in the Client Cli library source code in the MbedOS
cmd_init( &myprint );              // initialize cmdline with print function
cmd_set_ready_cb( cmd_ready_cb );  // configure ready cb
cmd_add("dummy", cmd_dummy, 0, 0); // add one dummy command
cmd_add("long", cmd_long, 0, 0);   // add one dummy command
uint16_t cmd_len = 0;
read(0, &cmd_len, sizeof(cmd_len));
char* p = (char*)malloc(cmd_len);
read(0, p, cmd_len);
cmd_exe(p);
```
[bug_cli_1_1.log](https://github.com/ARMmbed/mbed-os/files/3802627/bug_cli_1_1.log)
[bug_cli_1_2.log](https://github.com/ARMmbed/mbed-os/files/3802626/bug_cli_1_2.log)
[bug_cli_1_3.log](https://github.com/ARMmbed/mbed-os/files/3802625/bug_cli_1_3.log)
[bug_cli_1_4.log](https://github.com/ARMmbed/mbed-os/files/3802628/bug_cli_1_4.log)

0xc0170, 2019-11-04:
cc @ARMmbed/mbed-os-tools

0xc0170, 2019-11-04:
@artokin Who would be able to help with this one?

artokin, 2019-11-04:
I would propose @RomanSaveljev

adbridge, 2020-10-06:
No longer relevant with the move to a separate CI system

closed 2020-10-06 by adbridge