ARMmbed/mbed-os #11809: BLE connection problems

BramdeBoer1, 2019-11-04:
### Description

Some android devices refuse to connect to our device using bluetooth, whilst other android devices connect just fine.
We are using the cordio driver to implement the bluetooth connectivity.

We've tested with nRF Connect and got the following log from one of the devices that failed to connect.
```
nRF Connect, 2019-11-04
SCS-Env (E6:17:51:A7:7A:5F)
V	16:13:46.391	Connecting to E6:17:51:A7:7A:5F...
D	16:13:46.391	gatt = device.connectGatt(autoConnect = false, TRANSPORT_LE, preferred PHY = LE 1M)
D	16:13:46.529	[Callback] Connection state changed with status: 0 and new state: CONNECTED (2)
I	16:13:46.529	Connected to E6:17:51:A7:7A:5F
D	16:13:46.530	[Broadcast] Action received: android.bluetooth.device.action.ACL_CONNECTED
V	16:13:46.552	Discovering services...
D	16:13:46.552	gatt.discoverServices()
I	16:13:47.014	Connection parameters updated (interval: 20.0ms, latency: 0, timeout: 800ms)
D	16:14:01.524	[Callback] Connection state changed with status: 22 and new state: DISCONNECTED (0)
E	16:14:01.524	Error 22 (0x16): GATT CONN TERMINATE LOCAL HOST
I	16:14:01.524	Disconnected
D	16:14:01.581	[Broadcast] Action received: android.bluetooth.device.action.ACL_DISCONNECTED
```
The following log shows how it should look like.
```
nRF Connect, 2019-11-04
SCS-Env (E6:17:51:A7:7A:5F)
D	16:33:45.590	gatt.close()
D	16:33:45.593	wait(200)
V	16:33:45.796	Connecting to E6:17:51:A7:7A:5F...
D	16:33:45.797	gatt = device.connectGatt(autoConnect = false, TRANSPORT_LE, preferred PHY = LE 1M)
D	16:33:46.040	[Callback] Connection state changed with status: 0 and new state: CONNECTED (2)
I	16:33:46.040	Connected to E6:17:51:A7:7A:5F
D	16:33:46.043	[Broadcast] Action received: android.bluetooth.device.action.ACL_CONNECTED
V	16:33:46.076	Discovering services...
D	16:33:46.076	gatt.discoverServices()
I	16:33:46.669	Connection parameters updated (interval: 20.0ms, latency: 0, timeout: 800ms)
I	16:33:46.932	Connection parameters updated (interval: 7.5ms, latency: 0, timeout: 5000ms)
D	16:33:47.051	[Callback] Services discovered with status: 0
I	16:33:47.051	Services discovered
V	16:33:47.072	Generic Access (0x1800)
- Device Name [R] (0x2A00)
- Appearance [R] (0x2A01)
- Peripheral Preferred Connection Parameters [R] (0x2A04)
Generic Attribute (0x1801)
- Service Changed [I] (0x2A05)
I	16:33:47.126	Connection parameters updated (interval: 20.0ms, latency: 0, timeout: 800ms)
```
It looks like the discoverServices callback doesn't get triggered and we were able to trace it back to a timing issue when setting up the connection.

By enabling WSF_TRACE_ENABLED and LL_TRACE_ENABLED the connection problems went away.

After looking at the debug trace we decided to enable the LL_OP_MODE_FLAG_SLV_DELAY_LLCP_STARTUP flag which we hoped would slow the initialization sequence down.
This also fixed the connection issue.
We've confirmed this by disabling the trace information and keeping just the delay startup flag, after this our device was able to get a bluetooth connection to all devices that it has previously had problems connecting with.

So setting LL_OP_MODE_FLAG_SLV_DELAY_LLCP_STARTUP fixed our connectivity issues, but is this the correct solution or do we need to look at other options?

#### Target(s) affected by this defect ?
NRF52840 custom target

#### Toolchain(s) (name and version) displaying this defect ?
IAR EWB 8.32.3

#### What version of Mbed-os are you using (tag or sha) ?
MBED OS 5.12.1

0xc0170, 2019-11-05:
cc @ARMmbed/mbed-os-pan

0xc0170, 2020-02-24:
@ARMmbed/mbed-os-pan Please review

BramdeBoer1, 2020-10-26:
Could someone please tell me what changes are needed?

pan-, 2020-10-26:
@BramdeBoer1 We haven't made any direct change in that area but the LL stack has been updated to a more recent version. 
Could you try it and tell us if it addresses your issue (I doubt it would) ? If not I will make a PR that adds a configuration flag to configure `LL_OP_MODE_FLAG_SLV_DELAY_LLCP_STARTUP`.

0xc0170, 2020-10-26:
> Could someone please tell me what changes are needed?

Regarding the bot, the issue was all updated, it should be good now.

adbridge, 2020-10-26:
> 
> 
> > Could someone please tell me what changes are needed?
> 
> Regarding the bot, the issue was all updated, it should be good now.
No it wasn't , the mbed OS version requires a specific tag or sha.  "MBED OS 5.12.1" while a valid version name is neither. I will fix for you.

BramdeBoer1, 2020-11-02:
We've updated our application to mbed-os-6.4.0 and can confirm that this flag is still needed to solve the connection issue.