ARMmbed/mbed-os #11826: STM32L476RG stuck at WFI() forever when RTC is used as lpticker

syhoangmai-etteplan, 2019-11-06:
### STM32L476RG stuck at WFI() forever when RTC is used as lpticker

When use RTC as lpticker instead of LPTIM (to achieve long uninterrupt sleep due to LPTIM overflow every few seconds), it seems that the device will stuck at WFI() forever if thread_sleep_for is relatively long (e.g 60000ms). Looks like the device still wakes up according to the energy consumption (a few uA in deepsleep and 6mA when awake and stuck), but do not proceed any further.

#### Target(s) affected by this defect ?
STM32L4, tried on NUCLEO_L476RG

#### Toolchain(s) (name and version) displaying this defect ?
GCC 6 2017 q2 on windows 10

#### What version of Mbed-os are you using (tag or sha) ?
Mbed-os version is 679d24833acf0a0b5b0d528576bb37c70863bc4e (release 5.14.1)

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
Mbed CLI 1.10.1

#### How is this defect reproduced ? 
This issue can be seen by using RTC as lpticker instead of LPTIM ( "target.lpticker_lptim": 0). The program to reproduce this issue for NUCLEO_L476RG is attached here
[rtc_lpticker_stuck.zip](https://github.com/ARMmbed/mbed-os/files/3814458/rtc_lpticker_stuck.zip) . I added a watchdog thread that should kick hardware watchdog periodically if things are still ok. Feel free to adjust the watchdog timer thread priority, seems not to change anything positively. I write this code snippet with ATHandler since my real application use ublox-r412 modem, but a coworker of mine write another application without any cellular modem related at all but still encounter this problem.

0xc0170, 2019-11-07:
cc @ARMmbed/team-st-mcd

syhoangmai, 2019-11-08:
I tried to run the program today and one time according to the debugger it stuck at 
> CLEAR_BIT(SCB->SCR,` ((uint32_t)SCB_SCR_SLEEPDEEP_Msk));
in void HAL_PWREx_EnterSTOP2Mode(uint8_t STOPEntry)

Before today I only see it stuck at:
> if(SLEEPEntry == PWR_SLEEPENTRY_WFI) {
    __WFI();//nearly always stuck here
  }
in void HAL_PWR_EnterSLEEPMode(uint32_t Regulator, uint8_t SLEEPEntry)

The while loop may need a few iteration before it shows the issue, that is why I added the watchdog.
This could be false direction since the program is compiled with develop profile and get size optimized, but likely the root cause is the rtc since using LPTIM does not show this behavior. I will try the GPIO output trick you give provide update later

LMESTM, 2019-11-08:
@syhoangmai - thanks for the extra information. 
May you control a GPIO before and after the DEEP SLEEP / STOP MODE entry to verify that even without a debugger the GPIO state will be high when the target gets stuck. 

Also note that you are trying to use a configuration that is NOT supported as of now. We are not promoting the usage of TICKLESS with RTC timer. The supported functional configuration is based on LPTIM only, (with the current limitation of wake-ups every 2 sec indeed).

LMESTM, 2019-11-08:
More explanation: when you say that you're stuck in 
```
if(SLEEPEntry == PWR_SLEEPENTRY_WFI) {
__WFI();//nearly always stuck here
}
in void HAL_PWR_EnterSLEEPMode(uint32_t Regulator, uint8_t SLEEPEntry)
```
It means that you are not in deep sleep, but in simple WFI. 
In other words, the system is simply IDLE as there is nothing to do ...

LMESTM, 2019-11-08:
And about: 
```
CLEAR_BIT(SCB->SCR,` ((uint32_t)SCB_SCR_SLEEPDEEP_Msk));
in void HAL_PWREx_EnterSTOP2Mode(uint8_t STOPEntry)
```
It just means that debugger gets control on the board when it exits WFI, so it was in WFI before the debugger could access to it and when it got access the next instruction to execute is this one  ...

LMESTM, 2019-11-08:
Proposal: Could you rather remove the Watchdog, then start your test and wait for the STUCK issue to happen - once you see the target is "stuck" check the status with the debugger :-)

syhoangmai, 2019-11-08:
I did this with the watchdog removed. The watchdog thread is just there to show that the rtos is not executing that watchdog kicking thread (and maybe everything), guarantee that device still operating, and I think you will see the stuck problem easier if watchdog which restarts the system every now and then as no one kick it. Otherwise, the behavior of this issue is just nothing, since nothing else is executed. Could you reproduce the problem with my attached example? Just need to compile with develop profile and run.

@LMESTM Is there any plan to support using RTC as lpticker in the future? I think this important for low power devices since waking up every 2 or 4 seconds is not so satisfactory.

Currently I am increasing the prescaler for LPTIM as a workaround (yes, it seems to violate Mbed porting guide), and so far see no problem? Could you tell me is there any potential problem with that approach besides lowering the resolution of timing related functions like thread_sleep_for()?

LMESTM, 2019-11-08:
> I did this with the watchdog removed. The watchdog thread is just there to show that the rtos is not executing that watchdog kicking thread (and maybe everything), guarantee that device still [...]

Ok clear. I'll try to reproduce just to see ... but no guarantee that we will fix this.

> @LMESTM Is there any plan to support using RTC as lpticker in the future? I think this important for low power devices since waking up every 2 or 4 seconds is not so satisfactory.

I am so keen about this idea because : 
- using the RTC induces latencies and there is a complex work to do to overcome this latency 
- using RTC as the main system timer and also as RTC (get/set_time) would require complex implementation, I fear that system timers would get corrupted when application changes time.
- I'd prefer to have LPTIM as base timer, then a way to program RTC alarm for long periods of sleep and suspend scheduler in such situation. This may even also allow for deeper sleep modes. 

> Currently I am increasing the prescaler for LPTIM as a workaround (yes, it seems to violate Mbed porting guide), and so far see no problem? Could you tell me is there any potential problem with that approach besides lowering the resolution of timing related functions like thread_sleep_for()?

The only drawback I see is indeed about low resolution.

kjbracey-arm, 2019-11-08:
> increasing the prescaler for LPTIM 

I would have thought you can go down to at least /8, given the stated 4kHz minimum.

Not sure why you'd need 4kHz anyway - 1kHz should in principle be sufficient to keep the RTOS going, as long as you're capable of setting IRQs on successive counter values. I wonder who came up with that number. (Don't think it was me).  I would expect /32 for a 1kHz clock to be fine, if we can wake at 0003 and set a working IRQ for 0004.

Agree with reserving RTC for "alarm style" wake-up from suspend.

But please have 32-bit LPTIMs in future chips...

LMESTM, 2019-11-08:
> But please have 32-bit LPTIMs in future chips...

I'd love to indeed ! :-)

syhoangmai, 2019-11-12:
So from my understanding, the lpticker (or systick if I called it correctly), for every time the timer matched the compare value the ticker interrupt need to set a new match compare value in order to the keep the RTOS going, is it correct?

Assume the above is correct, could it be that the RTC take too long to set a new matching value? Could an RTC prescaler increase remove this problem?

Anyway, waking up due to timer overflow every 2^16/1024=64s looks good since the average current consumption goes up by a very insignificant amount (likely less than 1uA) and as a software person I do not really ask for better result (ignoring the people who want nA from standby or suspend sleep mode). By the way, isn't increasing the prescaler means that there are more time between matching intervals, so no risk on that latency aspect? And I think maybe reducing the systick rate could improve the system performance since it may avoid excessive jumping between threads?

If this is deemed beneficial, could the options up to /32 be added in the future?

LMESTM, 2019-11-12:
> So from my understanding, the lpticker (or systick if I called it correctly), for every time the timer matched the compare value the ticker interrupt need to set a new match compare value in order to the keep the RTOS going, is it correct?

that's also my understanding.

> Assume the above is correct, could it be that the RTC take too long to set a new matching value? Could an RTC prescaler increase remove this problem?

Increasing the prescaler could actaully make things worse. RTC (and LPTIM) blocks are clocked from 32kHz. With pre-scaler each clock cycle will take longer, so register accesses, and the time it takes to take manage 2 consecutive comparator programming will increase as well. Fortunately, in case of LPTIM, we're managing those 2 consecutive programing case to not impact the system. 

> Anyway, waking up due to timer overflow every 2^16/1024=64s looks good since the average current consumption goes up by a very insignificant amount (likely less than 1uA) and as a software person I [...]

I mentioned latencty above. About impacts on scheduler, I 'm not sure.

> If this is deemed beneficial, could the options up to /32 be added in the future?

Then the mbed-os 4kHz minimum requirement needs to be discussed. 
Then tests need to be updated accordingly, also we need to pass all the mbed-os tests in various configurations, so some more work but I guess this could be considered ...

star297, 2019-11-13:
Please, Please, Please whatever you do, can you not screw up the RTC functions in baremetal OS. Iv'e just spent a week getting ST's to wake up from sleep/deepsleep using the RTC alarm registers, in particular the L476 .

LMESTM, 2019-11-13:
@star297 good to know. Best would be to share your changes to make them available to everyone and then test cases can be added to mbed-os to avoid any regression ...

0xc0170, 2019-11-28:
@LMESTM What is the outcome here? Is this still an issue ? just trying to understand what should be done for this issue.

LMESTM, 2019-11-28:
As @star297 hasn't shared any specific test cases, I don't think there is anything else to track here.
As I would summarize it, this is a Change Request for mbed-os to support long periods of sleep with RTC alarm wake-up.

star297, 2019-11-28:
Sorry guys I did not realize you where waiting for my input here.
I use Eric's RTC alarm based interrupt wake up:

https://os.mbed.com/users/Sissors/code/WakeUp//file/65c04a02ad45/Device/WakeUp_STM_RTC.cpp/

However for some reason I have to set the RTC clock, something like set_time(0); in 'main.cpp' will do otherwise it will not wake up similar to the title of this bug case.
Perhaps your problem could be in this area, however....

> a few uA in deepsleep and 6mA when awake and stuck

That sounds strange, in operational mode the while(1) current consumption (Idd) is 14.1mA (HSE_XTAL) and 11.8mA (HSI).
Sounds to me that the clock sources are not re-initializing after wake up in this bug case? 

Using RTC alarm for long period wake up, from experience is reliable, however watch the 3-4mS resolution characteristics.

0xc0170, 2019-12-04:
RTC request (alarm) is being tracked in another ticket (this issue is just for technical bugs). If there is nothing else, I'll close this.

closed 2019-12-04 by 0xc0170