ARMmbed/mbed-os #11835: Cypress: FPGA: gpio test looks like not valid

yarbcy, 2019-11-07:
### Description of defect
Cypress: FPGA: gpio test looks like not valid.

In file TESTS/mbed_hal_fpga_ci_test_shield/gpio/main.cpp line 165:

`// Initialize GPIO pin as an input, pull-up mode.
    memset(&gpio, 0, sizeof gpio);
    gpio_init_inout(&gpio, pin, PIN_INPUT, PullUp, 0);
    TEST_ASSERT_NOT_EQUAL(0, gpio_is_connected(&gpio));
    tester.gpio_write(MbedTester::LogicalPinGPIO0, 0, false);
    wait_us(HI_Z_READ_DELAY_US);
    TEST_ASSERT_EQUAL_INT(1, gpio_read(&gpio)); // hi-Z, pulled up
    gpio_free(&gpio);`

PSoC6 set 0, tester(FPGA) set 0. But expect 1. It looks like it doesn't make sense.
This test PASSED if set PSOC6 1,  tester(FPGA) set 0.

#### Target(s) affected by this defect ?
Tested on CY8CKIT_062_WIFI_BT

#### Toolchain(s) (name and version) displaying this defect ?
Tested on GCC_ARM

#### What version of Mbed-os are you using (tag or sha) ?
Latest

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
N/A

#### How is this defect reproduced ? 
Always

yarbcy, 2019-11-07:
@0xc0170

0xc0170, 2019-11-07:
@yarbcy What is not valid in the code snippet above?

cc @mprse

yarbcy, 2019-11-07:
> @yarbcy What is not valid in the code snippet above?
> 
> cc @mprse

PSoC6 set 0, tester(FPGA) set 0. But expect 1

0xc0170, 2019-11-07:
Isn't it that you write 0, but because of pull up you expect 1? This comment explains it : `// hi-Z, pulled up`

yarbcy, 2019-11-07:
> Isn't it that you write 0 (fpga side), but because of pull up you expect 1 on the actual pin?
> 
> This comment explains it : `// hi-Z, pulled up`

Yes. But PSoC6 also set initial state 0.

yarbcy, 2019-11-07:
@0xc0170 @mprse 
All the rest GPIO test are passed. Except this one.

0xc0170, 2019-11-07:
gpio api supports value even for input, it writes to gpio as it claims "for future use if you switch to output". Although as soon as you change the direction, you write a new value so I dont understand the use case for that condition there).
My expectation, once you set pin to input, it goes into Hi-Z state (no matter what you write as a value, it shall be ignored. As soon as you change the direction,  the value is preserved and output equals to the value we wrote earlier), and that is what the test also expects. How does it work for failing target? What happens if you set direction to input to the pin itself?

I cant find it in the history why it was there, will need someone else to look at this closer.

yarbcy, 2019-11-07:
From my point of view:

1. PSoC6 set PullUp drive mode (Input)
2. Set gpio_write(0) (I see it on logical analyzer)
3. Set tester (FPGA) 0
4. gpio_read() return 0. But expect 1.

mprse, 2019-11-07:
Test scenario:
1. Set Mbed GPIO to pulled-up input. Provided value, in this case, is irrelevant.
2. Set the corresponding pin on the FPGA side to Hi-Z (something like no connection between GPIOs).
3. Read the Mbed GPIO.
4. Expect logic `1` since the pin is pulled up. 

Maybe pull mode does not work on the cypress board?

yarbcy, 2019-11-07:
@mprse Cypress support pull mode.

This case is passed if replace gpio_init_inout() -> gpio_init_in_ex()

Could you please help me understand why?

mprse, 2019-11-07:
```
void gpio_init_inout(gpio_t *gpio, PinName pin, PinDirection direction, PinMode mode, int value)
{
    if (direction == PIN_INPUT) {
        _gpio_init_in(gpio, pin, mode);
        if (pin != NC) {
            gpio_write(gpio, value);    // we prepare the value in case it is switched later
        }
    } else {
        _gpio_init_out(gpio, pin, mode, value);
    }
}
```

```
void gpio_init_in_ex(gpio_t *gpio, PinName pin, PinMode mode)
{
    _gpio_init_in(gpio, pin, mode);
}
```

`gpio_init_inout()` calls `_gpio_init_in()` and `gpio_write()`.
`gpio_init_in_ex()` calls only `_gpio_init_in()`.

This call to `gpio_write()` is suspicious. Probably this is done to set the value for the future if the port direction is changed (according to https://github.com/ARMmbed/mbed-os/issues/11835#issuecomment-551062580). 
Maybe the `gpio_write()` changes the port direction?

yarbcy, 2019-11-07:
> Maybe the `gpio_write()` changes the port direction?

Looks like no.

But why we have initial state for input pin it should be done for output pin. Right?

0xc0170, 2019-11-07:
That is what I dont understand.

anyway, what happens on the pin if you set it to output (it should be floating, no value set?). Is this documented in the reference manual for that MCU ? What is expected behavior for input pin?

In case the pin has different behavior than we expect (@mprse I havent found the behavior defined in the gpio_api header file for this inout function), you can always check in HAL (if pin is output, and something asks to write to the pin, ignore it and keep it floating).

kjbracey-arm, 2019-11-07:
Looking in `apply_config`, called from `gpio_dir` for PSOC6, I see 

    if (obj->drive_mode == PullUp) {
        gpio_write(obj, 1);
    ...

Suggesting that the `gpio_write` control is used to set the pull - presumably it's acting as a "weak drive".

That test (and the init_inout) seems to be assuming that the `gpio_write` has no effect until the direction is switched.

Seems like achieving that would need a more advanced `gpio_write` implementation - which is unfortunate as we normally like it to be a nice small inline register write.

mprse, 2019-11-07:
```
static inline void gpio_write(gpio_t *obj, int value)
{
    if (obj->direction == CYHAL_GPIO_DIR_OUTPUT) {    <--- set only if output
        MBED_ASSERT(obj->pin != CYHAL_NC_PIN_VALUE);
        cyhal_gpio_write(obj->pin, value != 0);
    }
}

static inline void gpio_set_pull(gpio_t *obj, int value)  <--- new function for setting pull mode
{
    MBED_ASSERT(obj->pin != CYHAL_NC_PIN_VALUE);
    cyhal_gpio_write(obj->pin, value != 0);
}

void apply_config(gpio_t *obj)
{
    MBED_ASSERT(obj->pin != CYHAL_NC_PIN_VALUE);

    cy_rslt_t rslt;
    if (CY_RSLT_SUCCESS != (rslt = cyhal_gpio_configure(obj->pin, obj->direction, obj->drive_mode))) {
        MBED_ERROR(MBED_MAKE_ERROR(MBED_MODULE_DRIVER_GPIO, CY_RSLT_GET_CODE(rslt)), "cyhal_gpio_configure failed");
    }

    if (obj->drive_mode == PullUp) {
        gpio_set_pull(obj, 1);  <--- use new function for pull mode
    } else if (obj->drive_mode == PullDown) {
        gpio_set_pull(obj, 0);  <--- use new function for pull mode
    }
}

```

Maybe something like this?

kjbracey-arm, 2019-11-07:
I think so, but it seems that needs to be extended so that if you're told to write while an input, you should be remembering the written value (in obj, I guess) to write next time you switch to output.

yarbcy, 2019-11-11:
@0xc0170 @mprse Do we have some updates?

0xc0170, 2019-11-11:
@yarbcy Can you review the proposal above https://github.com/ARMmbed/mbed-os/issues/11835#issuecomment-551097567 ?

yarbcy, 2019-11-12:
@0xc0170 Looks good. But I m not expert in mbed_hal.

mprse, 2019-11-12:
@yarbcy Can you test this code?

yarbcy, 2019-11-12:
@mprse Yes.

yarbcy, 2019-11-13:
@mprse I tested and it failed communication with FPGA tester: "An MbedTester communication channel could not be created"

mprse, 2019-11-13:
The communication with the `FPGA-test-shield` is established using bit-banged SPI communication - based on GPIOs (`DigitalInOut`). So dealing with GPIO driver may have influence on this. 
Can you share your code changes?

yarbcy, 2019-11-13:
**I replaced cy_gpio_api.c line 32 apply_config():**

void apply_config(gpio_t *obj)
{
    MBED_ASSERT(obj->pin != CYHAL_NC_PIN_VALUE);

    cy_rslt_t rslt;
    if (CY_RSLT_SUCCESS != (rslt = cyhal_gpio_configure(obj->pin, obj->direction, obj->drive_mode))) {
        MBED_ERROR(MBED_MAKE_ERROR(MBED_MODULE_DRIVER_GPIO, CY_RSLT_GET_CODE(rslt)), "cyhal_gpio_configure failed");
    }

    if (obj->drive_mode == PullUp) {
        gpio_set_pull(obj, 1);  <--- use new function for pull mode
    } else if (obj->drive_mode == PullDown) {
        gpio_set_pull(obj, 0);  <--- use new function for pull mode
    }
}

**Replaced gpio_object.h line 59 gpio_write():**

static inline void gpio_write(gpio_t *obj, int value)
{
    if (obj->direction == CYHAL_GPIO_DIR_OUTPUT) {    <--- set only if output
        MBED_ASSERT(obj->pin != CYHAL_NC_PIN_VALUE);
        cyhal_gpio_write(obj->pin, value != 0);
    }
}

**Added new after gpio_write():** 

void apply_config(gpio_t *obj)
{
    MBED_ASSERT(obj->pin != CYHAL_NC_PIN_VALUE);

    cy_rslt_t rslt;
    if (CY_RSLT_SUCCESS != (rslt = cyhal_gpio_configure(obj->pin, obj->direction, obj->drive_mode))) {
        MBED_ERROR(MBED_MAKE_ERROR(MBED_MODULE_DRIVER_GPIO, CY_RSLT_GET_CODE(rslt)), "cyhal_gpio_configure failed");
    }

    if (obj->drive_mode == PullUp) {
        gpio_set_pull(obj, 1);  <--- use new function for pull mode
    } else if (obj->drive_mode == PullDown) {
        gpio_set_pull(obj, 0);  <--- use new function for pull mode
    }
}

yarbcy, 2019-11-14:
@mprse  **My proposal for code change is:**

mbed-os/hal/mbed_gpio.c remove line 63-65 because initial value doesn't make sense when pin direction is in input direction.

kjbracey-arm, 2019-11-14:
> mbed-os/hal/mbed_gpio.c remove line 63-65 because initial value doesn't make sense when pin direction is in input direction.

As I noted in my example above, there is a use-case for this. Set the output to low, and flip between input and output to manually bit-bang an I2C open-drain output.

That would be efficient on a chip with "OUT" and "DIR" registers where OUT was ignored when DIR was set to input.

I don't know whether any Mbed application in the word is currently doing this, but it does currently work, those lines 63-65 seem to be intended for that, and that test makes sure it works.

kjbracey-arm, 2019-11-14:
Annoyingly, the commit that added lines 63-65 (be8bca4aa06bf63042ac92b79092b67d319c4be2) only has the commit message "proposed change of gpio_api", so it's not 100% clear what they were intending. But it was deliberate.

yarbcy, 2019-11-14:
@kjbracey-arm I don't understand next steps. If you don't agree (line 63-65). Please propose your approach.

kjbracey-arm, 2019-11-14:
My suggestion is roughly as @mprse suggested above, but with the addition as I stated:

> if you're told to write while an input, you should be remembering the written value (in obj, I guess) to write next time you switch to output.

yarbcy, 2019-11-14:
@mprse @0xc0170  Could you please take a look?

0xc0170, 2019-11-14:
@yarbcy @mprse will update the suggestion above with written value storing, it will be good to take this to PR to propose a change. It should fix the failures you have been seeing

> Annoyingly, the commit that added lines 63-65 (be8bca4) only has the commit message "proposed change of gpio_api", so it's not 100% clear what they were intending. But it was deliberate.

👍 , the comment in the PR itself states something similar about the intention https://github.com/ARMmbed/mbed-os/pull/198#commitcomment-5619242 . Sadly, this is not included in the commit itself

yarbcy, 2019-11-14:
@mprse I don't have clear understanding what needs to be done. Can you do these changes?

mprse, 2019-11-14:
I will create a fix proposition.

mprse, 2019-11-14:
The fix can be found here: PR https://github.com/ARMmbed/mbed-os/pull/11867

mprse, 2019-11-15:
Great news! 🎉 

> But please fix in file gpio_object.h line 66: change '==' to '='.

Fixed! Thanks for catching this.

BTW. Probably the previous implementation faulty because the direction check was invalid:
```
static inline void gpio_write(gpio_t *obj, int value)
{
    if (obj->direction == CYHAL_GPIO_DIR_OUTPUT) {    <--- issue: when output direction is set to CYHAL_GPIO_DIR_BIDIRECTIONAL
        MBED_ASSERT(obj->pin != CYHAL_NC_PIN_VALUE);
        cyhal_gpio_write(obj->pin, value != 0);
    }
}
```