ARMmbed/mbed-os #11838: [BUG] AnalogIn reading error for mbed 5.12 & 5.13

Jack-74, 2019-11-07:
### Description of defect

#### Target(s) affected by this defect ?
STM32L432KC

#### Toolchain(s) (name and version) displaying this defect ?

#### What version of Mbed-os are you using (tag or sha) ?
mbed5.12 & 5.13

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
Mbed Online IDE

#### How is this defect reproduced ? 
Just need to use the sample analog program provided in the Mbed OS product page.
Compile and run using Mbed5.11, the result is correct.
When compile and run using Mbed 5.12 & 5.13 using the same sample code, but ADC results are different.

0xc0170, 2019-11-07:
Thanks @Jack-74 . I believe this one was reported earlier on developer page.

cc @ARMmbed/team-st-mcd please review

jeromecoutant, 2019-11-07:
Hi
There was some ADC issue with STM32L4 family before 5.14 version and #11072

0xc0170, 2019-11-08:
@Jack-74 Please can you test latest release if this was fixed in 5.14 release?

adbridge, 2019-11-12:
@Jack-74 when providing the affected release please provide the latest actual release tag (e.g mbed-os-5.13.4) or sha. This allows two things:
1. A more accurate point for us to investigate from 
2. The ability to automate the mirroring of a real tag to our internal defect system
Thanks.

0xc0170, 2019-11-25:
@Jack-74 If you 5.14 did not fix the issue, please reopen with more detail information for this release. 

I'll close this as resolved.

closed 2019-11-25 by 0xc0170