ARMmbed/mbed-os #11852: different behaviour for custom targets under Windows and MacOS

JojoS62, 2019-11-12:
### Description of defect

Under Windows, a linker script in a custom target is not used when there is also a linker script for the same MCU in targets.
Under MacOS, the linker script in the custom target is used.

Another difference was also that under Windows Python 2.7 was used, under MacOS 3.x.

#### Target(s) affected by this defect ?

all custom targets?

STM32F407VE. I created a custom target because ARCH_MAX was similiar, but with different size of flash. The custom target works and is using the custom PinNames. In the meantime, mbed-os memory layout or default linking changed and the custom linker script did not match. But this had no effect because compiling under windows is using the linker script in targets/TARGET_STM/TARGET_STM32F4/TARGET_STM32F407xE/device/TOOLCHAIN_GCC_ARM

#### Toolchain(s) (name and version) displaying this defect ?
GCC_ARM 7/8

#### What version of Mbed-os are you using (tag or sha) ?
5.14

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
mbed-cli 1.10.1

#### How is this defect reproduced ? 
- create custom target with linker script
- compiling gives different results under Windows (wrong) and MacOS (correct). Maybe dependent on Python 2/3

0xc0170, 2019-11-13:
@JojoS62 I do not completely follow the report above.

Can you create small code snippet - targets hierarchy ? how to reproduce the issue

Even your branch could help (with the commands what to run to get wrong linker file).

JojoS62, 2019-11-13:
[testSTM32F407VE_BLACK.zip](https://github.com/ARMmbed/mbed-os/files/3842483/testSTM32F407VE_BLACK.zip)
I have attached a custom target, STM32F407VE_BLACK with the custom_targets.json and a directory with the additional files. The .ld file contains just garbage and the linker should complain. But under windows, this is linking and so obviously not using the custom .ld file.
To reproduce, add this  files to a mbed blinky example and compile with 'mbed compile -m STM32F407VE_BLACK -t GCC_ARM'.

JojoS62, 2020-10-13:
I have not experienced further problems with newer Mbed versions. I would consider this issue as outdated and fixed.

closed 2020-10-13 by JojoS62