ARMmbed/mbed-os #11855: cmsis-pack-manager 0.2.9 installation broken on Alpine linux

korjaa, 2019-11-13:
### Description of defect

Dockerfile to reproduce
```
FROM alpine

# Runtime dependencies
RUN apk add libgcc
RUN apk add libstdc++

# Compilation and Mbed dependencies
#  --no-cache, disables package caching
#  --virtual .build-deps, temporary dependencies named .build-deps (easy cleanup)
# Generic dependencies:
#   build-base
#   cmake
#   git
#   mercurial
#   python3
#   python3-dev
#   py-pip
#   py-virtualenv - Used for easy pip3 cleanup
# cmsis-package-manager dependencies:
#   libffi-dev
#   cargo
# cryptography dependencies
#   openssl-dev
# NOTE: There can be only one "add" call per virtual name
RUN apk add --no-cache --virtual build-deps \
    build-base \
    cmake \
    git \
    mercurial \
    python3 \
    python3-dev \
    py-pip \
    py-virtualenv \
    libffi-dev \
    cargo \
    openssl-dev

# Create virtualenv for easy cleanup
RUN virtualenv /tmp/python-deps -p "$(which python3)"

# Workaround for broken mbed-cli installation
RUN . /tmp/python-deps/bin/activate; pip install git+https://github.com/ARMmbed/cmsis-pack-manager.git@v0.2.9
```

**Current workaround is to preinstall older cmsis-pack-manager of 0.2.8:**
```
RUN . /tmp/python-deps/bin/activate; pip install git+https://github.com/ARMmbed/cmsis-pack-manager.git@v0.2.8
```

**Proposed fix:**
Define cmsis-pack-manager upper version to 0.2.8 instead of 3.0.0.

#### Target(s) affected by this defect ?
mbed toolchain installation

#### Toolchain(s) (name and version) displaying this defect ?
alpine based images (docker)

#### What version of Mbed-os are you using (tag or sha) ?
mbed-os 5.14

But I would assume this affects every mbed-os version that has cmsis-pack-manager dependency listed as <3.0.0
https://github.com/ARMmbed/mbed-os/blob/mbed-os-5.14/requirements.txt#L25

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
Docker 19.03.4
alpine 965ea09ff2eb
mbed-cloud-client-example 4.0.0

#### How is this defect reproduced ? 
With the provided Dockerfile

0xc0170, 2019-11-13:
@ARMmbed/mbed-os-tools Please review

0xc0170, 2019-11-13:
@korjaa How does it fail ? It's not clear to me from the log above.

korjaa, 2019-11-13:
Ah, forgot that one. The failure comes in some rust compilation part.
```
...
       Compiling url v2.1.0
    error[E0658]: use of unstable library feature 'alloc': this library is unlikely to be stabilized in its current form or name (see issue #27783)
      --> /root/.cargo/registry/src/github.com-1ecc6299db9ec823/ring-0.16.9/src/lib.rs:69:1
       |
    69 | extern crate alloc;
       | ^^^^^^^^^^^^^^^^^^^
....
```

0xc0170, 2019-11-13:
cc @theotherjimmy

mark-edgeworth, 2019-11-13:
Seen.

theotherjimmy, 2019-11-13:
Huh, I thought ring complied on stable. What's your rustc version?

korjaa, 2019-11-14:
`(56/61) Installing rust (1.34.2-r1)`

korjaa, 2019-11-14:
Full build log from Docker
```
Sending build context to Docker daemon  417.3kB
Step 1/6 : FROM alpine
 ---> 965ea09ff2eb
Step 2/6 : RUN apk add libgcc
 ---> Using cache
 ---> bf08ef557b1e
Step 3/6 : RUN apk add libstdc++
 ---> Using cache
 ---> 1aefee8f0620
Step 4/6 : RUN apk add --no-cache --virtual build-deps     build-base     cmake     git     mercurial     python3     python3-dev     py-pip     py-virtualenv     libffi-dev     cargo     openssl-dev
 ---> Running in 51713c324ce0
fetch http://dl-cdn.alpinelinux.org/alpine/v3.10/main/x86_64/APKINDEX.tar.gz
fetch http://dl-cdn.alpinelinux.org/alpine/v3.10/community/x86_64/APKINDEX.tar.gz
(1/61) Installing binutils (2.32-r0)
(2/61) Installing libmagic (5.37-r1)
(3/61) Installing file (5.37-r1)
(4/61) Installing gmp (6.1.2-r1)
(5/61) Installing isl (0.18-r0)
(6/61) Installing libgomp (8.3.0-r0)
(7/61) Installing libatomic (8.3.0-r0)
(8/61) Installing mpfr3 (3.1.5-r1)
(9/61) Installing mpc1 (1.1.0-r0)
(10/61) Installing gcc (8.3.0-r0)
(11/61) Installing musl-dev (1.1.22-r3)
(12/61) Installing libc-dev (0.7.1-r0)
(13/61) Installing g++ (8.3.0-r0)
(14/61) Installing make (4.2.1-r2)
(15/61) Installing fortify-headers (1.1-r0)
(16/61) Installing build-base (0.5-r1)
(17/61) Installing libacl (2.2.52-r6)
(18/61) Installing libbz2 (1.0.6-r7)
(19/61) Installing expat (2.2.8-r0)
(20/61) Installing lz4-libs (1.9.1-r0)
(21/61) Installing xz-libs (5.2.4-r0)
(22/61) Installing libarchive (3.3.3-r1)
(23/61) Installing ca-certificates (20190108-r0)
(24/61) Installing nghttp2-libs (1.39.2-r0)
(25/61) Installing libcurl (7.66.0-r0)
(26/61) Installing ncurses-terminfo-base (6.1_p20190518-r0)
(27/61) Installing ncurses-terminfo (6.1_p20190518-r0)
(28/61) Installing ncurses-libs (6.1_p20190518-r0)
(29/61) Installing rhash-libs (1.3.8-r0)
(30/61) Installing libuv (1.29.1-r0)
(31/61) Installing cmake (3.14.5-r0)
(32/61) Installing pcre2 (10.33-r0)
(33/61) Installing git (2.22.0-r0)
(34/61) Installing libffi (3.2.1-r6)
(35/61) Installing gdbm (1.13-r1)
(36/61) Installing readline (8.0.0-r0)
(37/61) Installing sqlite-libs (3.28.0-r1)
(38/61) Installing python2 (2.7.16-r1)
(39/61) Installing mercurial (4.9.1-r0)
(40/61) Installing python3 (3.7.5-r1)
(41/61) Installing pkgconf (1.6.1-r1)
(42/61) Installing python3-dev (3.7.5-r1)
(43/61) Installing py-setuptools (40.8.0-r1)
(44/61) Installing py3-setuptools (40.8.0-r1)
(45/61) Installing py2-pip (18.1-r0)
(46/61) Installing py-virtualenv (16.0.0-r1)
(47/61) Installing py2-virtualenv (16.0.0-r1)
(48/61) Installing py3-virtualenv (16.0.0-r1)
(49/61) Installing linux-headers (4.19.36-r0)
(50/61) Installing libffi-dev (3.2.1-r6)
(51/61) Installing llvm-libunwind (8.0.0-r0)
(52/61) Installing llvm-libunwind-dev (8.0.0-r0)
(53/61) Installing rust-stdlib (1.34.2-r1)
(54/61) Installing libxml2 (2.9.9-r2)
(55/61) Installing llvm8-libs (8.0.0-r1)
[... 306 lines cut ...]
  
  error: aborting due to previous error
  
  For more information about this error, try `rustc --explain E0658`.
  error: Could not compile `ring`.
  warning: build failed, waiting for other jobs to finish...
  error: build failed
  ----------------------------------------
  ERROR: Failed building wheel for cmsis-pack-manager
  Running setup.py clean for cmsis-pack-manager
  Building wheel for pyyaml (setup.py): started
  Building wheel for pyyaml (setup.py): finished with status 'done'
  Created wheel for pyyaml: filename=PyYAML-5.1.2-cp37-cp37m-linux_x86_64.whl size=44103 sha256=5c127f3bc53c05a511a20f92987dfb0697816b000f8a06e766dbbceca2a912a6
  Stored in directory: /root/.cache/pip/wheels/d9/45/dd/65f0b38450c47cf7e5312883deb97d065e030c5cca0a365030
  Building wheel for cffi (setup.py): started
  Building wheel for cffi (setup.py): finished with status 'done'
  Created wheel for cffi: filename=cffi-1.13.2-cp37-cp37m-linux_x86_64.whl size=324708 sha256=87bfffcd54eaf6d0ccd57481aae040a487eb9b00b3e80499abfc19a7d2439d78
  Stored in directory: /root/.cache/pip/wheels/ad/b9/f0/9ba5c3aa263f1330e69ef220c0187a8e7192b38e95b8093e95
  Building wheel for pycparser (setup.py): started
  Building wheel for pycparser (setup.py): finished with status 'done'
  Created wheel for pycparser: filename=pycparser-2.19-py2.py3-none-any.whl size=111029 sha256=fca05c098d469603a269459641059b8bfe45b1078fbff77419091a3be8527090
  Stored in directory: /root/.cache/pip/wheels/f2/9a/90/de94f8556265ddc9d9c8b271b0f63e57b26fb1d67a45564511
Successfully built pyyaml cffi pycparser
Failed to build cmsis-pack-manager
```