ARMmbed/mbed-os #11863: Running unit tests on Windows machine

mprse, 2019-11-14:
### Description of defect

Tried to run unit-tests on the windows machine using the following manual:
https://os.mbed.com/docs/mbed-os/v5.14/tools/unit-testing.html

When running the command: 
`mbed test --unittests`

1. Configuration issue

```
[  2%] Building CXX object CMakeFiles/features-cellular-framework-AT-athandler.MbedOS.dir/C_/Users/int_szyk/Documents/GitHub/mbed-os/features/cellular/framework/AT/ATHandler.cpp.obj
C:\Users\int_szyk\Documents\GitHub\mbed-os\features\cellular\framework\AT\ATHandler.cpp: In member function 'void mbed::ATHandler::write_int(int32_t)':
C:\Users\int_szyk\Documents\GitHub\mbed-os\features\cellular\framework\AT\ATHandler.cpp:1397:48: error: expected ')' before 'PRIi32'
     int32_t result = sprintf(number_string, "%" PRIi32, param);
                             ~                  ^~~~~~~
                                                )
mingw32-make[2]: *** [CMakeFiles\features-cellular-framework-AT-athandler.MbedOS.dir\build.make:63: CMakeFiles/features-cellular-framework-AT-athandler.MbedOS.dir/C_/Users/int_szyk/Documents/GitHub/mbed-os/features/cellular/framework/AT/ATHandler.cpp.obj] Error 1
mingw32-make[1]: *** [CMakeFiles\Makefile2:1980: CMakeFiles/features-cellular-framework-AT-athandler.MbedOS.dir/all] Error 2
mingw32-make: *** [Makefile:140: all] Error 2
```

The same problem was already reported here:
https://github.com/ARMmbed/mbed-os/issues/11485#issuecomment-531793499

This issue can be solved by adding the following lines to the `CMakeLists.txt` file:
```
set(CMAKE_C_FLAGS "${CMAKE_C_FLAGS} -D__STDC_FORMAT_MACROS")
set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -D__STDC_FORMAT_MACROS")
```

(solved by @maciejbocianski )

There is no information in the docs that such a symbol should be added.

2. Build error

```
CMakeFiles\features-cellular-framework-device-cellularstatemachine.dir/objects.a(Semaphore_stub.cpp.obj):Semaphore_stub.cpp:(.text+0x0): multiple definition of `rtos::Semaphore::Semaphore(int)'
CMakeFiles\features-cellular-framework-device-cellularstatemachine.dir/objects.a(AT_CellularContext_stub.cpp.obj):AT_CellularContext_stub.cpp:(.text$_ZN4rtos9SemaphoreC1Ei[_ZN4rtos9SemaphoreC1Ei]+0x0): first defined here
collect2.exe: error: ld returned 1 exit status
mingw32-make[2]: *** [CMakeFiles\features-cellular-framework-device-cellularstatemachine.dir\build.make:394: features-cellular-framework-device-cellularstatemachine.exe] Error 1
mingw32-make[1]: *** [CMakeFiles\Makefile2:1327: CMakeFiles/features-cellular-framework-device-cellularstatemachine.dir/all] Error 2
libfeatures-cellular-framework-AT-at_cellularcontext.MbedOS.a(AT_CellularContext.cpp.obj):AT_CellularContext.cpp:(.text$_ZN4rtos9SemaphoreC1Ei[_ZN4rtos9SemaphoreC1Ei]+0x0): multiple definition of `rtos::Semaphore::Semaphore(int)'
CMakeFiles\features-cellular-framework-AT-at_cellularcontext.dir/objects.a(Semaphore_stub.cpp.obj):Semaphore_stub.cpp:(.text+0x0): first defined here
libfeatures-cellular-framework-AT-at_cellularcontext.MbedOS.a(AT_CellularContext.cpp.obj):AT_CellularContext.cpp:(.text$_ZN4rtos9Semaphore15try_acquire_forEj[_ZN4rtos9Semaphore15try_acquire_forEj]+0x0): multiple definition of `rtos::Semaphore::try_acquire_for(unsigned int)'
CMakeFiles\features-cellular-framework-AT-at_cellularcontext.dir/objects.a(Semaphore_stub.cpp.obj):Semaphore_stub.cpp:(.text+0x80): first defined here
libfeatures-cellular-framework-AT-at_cellularcontext.MbedOS.a(AT_CellularContext.cpp.obj):AT_CellularContext.cpp:(.text$_ZN4rtos9Semaphore7releaseEv[_ZN4rtos9Semaphore7releaseEv]+0x0): multiple definition of `rtos::Semaphore::release()'
CMakeFiles\features-cellular-framework-AT-at_cellularcontext.dir/objects.a(Semaphore_stub.cpp.obj):Semaphore_stub.cpp:(.text+0xaa): first defined here
collect2.exe: error: ld returned 1 exit status
mingw32-make[2]: *** [CMakeFiles\features-cellular-framework-AT-at_cellularcontext.dir\build.make:474: features-cellular-framework-AT-at_cellularcontext.exe] Error 1
mingw32-make[1]: *** [CMakeFiles\Makefile2:1600: CMakeFiles/features-cellular-framework-AT-at_cellularcontext.dir/all] Error 2
mingw32-make: *** [Makefile:140: all] Error 2
```

Looks like there are duplicated symbols in the tests. 

#### Target(s) affected by this defect ?

N/A

#### What version of Mbed-os are you using (tag or sha) ?

Master

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

OS: Windows 10
MinGW: x86_64-8.1.0-posix
cmake: 3.15.3
python: 2.7.13
mbed-cli: 1.10.1
pip: 19.2.3
gcovr: 4.1

#### How is this defect reproduced ? 

On Windows machine run:
`mbed test --unittests`

0xc0170, 2020-02-24:
@ARMmbed/mbed-os-test Please review

maciejbocianski, 2020-02-24:
I think it was already fixed
See:
https://github.com/ARMmbed/mbed-os/pull/12287
https://github.com/ARMmbed/mbed-os/pull/12254

mprse, 2020-02-25:
I think we can close this one.

maciejbocianski, 2020-03-10:
ping @ARMmbed/mbed-os-maintainers

closed 2020-10-15 by adbridge