ARMmbed/mbed-os #11876: Cannot use Filesystem storage type with FlashIAP for both internal and external KV

amq, 2019-11-16:
### Description of defect

As a developer, I need:
- DeviceKey for ROT to derive the device ID
- NVStore or KVStore to persist small chunks of data
- filesystem to store the firmware candidate and other files
- while I only have the internal flash at my disposal

I can't, because:
- DeviceKey is hard-dependent on the KVStore. If it wasn't, I could use NVStore and a self-managed filesystem, as a less tangled solution that the whole storage system
- kv_config (called by DeviceKey) initializes FlashIAPBlockDevice in a way which makes it is impossible to have multiple instances of it. As a result, configuration which tries to have a FlashIAP-based filesystem with RBP will fail

Sample memory layout:

```
+--------------------------+
|        Internal KV       | 16
+--------------------------+
|                          |
|                          |
|        Filesystem        | 440
|                          |
|                          |
+--------------------------+
|                          |
|                          |
|        Active App        | 440
|                          |
|                          |
+--------------------------+
|                          |
|        Bootloader        | 128
|                          |
+--------------------------+
```

#### Target(s) affected by this defect ?

All

#### Toolchain(s) (name and version) displaying this defect ?

All

#### What version of Mbed-os are you using (tag or sha) ?

5.14.1

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

All

#### How is this defect reproduced ? 

Use this storage configuration in `mbed_app.json`:

```
"storage.storage_type": "FILESYSTEM",
"storage_filesystem.blockdevice": "FLASHIAP",
"storage_filesystem.filesystem": "LITTLE",
```

0xc0170, 2019-11-19:
cc @ARMmbed/mbed-os-storage

SeppoTakalo, 2020-01-10:
This is a non-standard and unsupported configuration, but you might be able to try.
In realidy, I don't think `FlashIAP` prevents multiple instances of it. So in theory, you could instantiate another `FlashIAPBlockDevice` on your application. Just make sure that your addresses do no overlap.

As long as application does not cause two `FlashIAP` instances to write, it might work.

0xc0170, 2020-02-24:
@amq I believe this was answered above, I will close this.

closed 2020-02-24 by 0xc0170