ARMmbed/mbed-os #11883: MAX32630FTHR : SYNC FAILED error for greentea tests

Emre-Eken, 2019-11-18:
### Description of defect

Hello,

When I run this example Greentea test, I get SYNC_FAILED error:

[image]

The test is located under TESTS/Group1/TestCase1/

Here are the logs :

[image]

[image]

I saw some related tickets like [#10032](https://github.com/ARMmbed/mbed-os/issues/10032)
but it seems it was not solved for all boards.

#### Target(s) affected by this defect ?

MAX32630FTHR (May also be affecting the other Maxim Integrated boards too)

#### Toolchain(s) (name and version) displaying this defect ?

GCC_ARM

#### What version of Mbed-os are you using (tag or sha) ?

2e96145b7607de430235dd795ab5350c1d4d64d7 (11/16/2019 9:55:53 AM)

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

mbed-cli 1.1.1
mbedgt 1.6.1
mbedhtrun 1.5.1

#### How is this defect reproduced ? 

Run the above Greentea test on a MAX32630FTHR board and check the result.

40Grit, 2019-11-18:
@trowbridgec @millerc-ep

0xc0170, 2019-11-19:
cc @ARMmbed/team-maximintegrated

maclobdell, 2019-11-20:
@Emre-Eken - Can you confirm that other standard tests such as _tests-events-queue_ are passing?  I usually use that one to verify that the test infrastructure including serial interface is working.  The Sync failure can result if the serial connection is not working, flashing is not working, or the device crashes before the application runs.  Normally, the device will respond to the sync signal and then start running its test.  if the sync response doesn't happen then something fundamental went wrong.

Emre-Eken, 2019-11-21:
@maclobdell - I investigated it more and saw that it takes so long to be syncronized. For example, in the example above, it waited 97.23 sec and reached timeout duration. I tried a few more times and saw that sometimes it can be syncronized before timeout as one below.

[image]

Still it took long time (72.59 sec) but succeeded before the timeout. 

I also tried the test that you suggested. It passed as below but still the time was long(83.78 sec).

[image]

I do not know if this is an acceptable time to be syncronized but I think it should not be since sometimes it reaches the default timeout duration. 

Do you have any guess why it takes so long to be syncronized?

timovanasten, 2020-01-02:
Also having issues with greentea on macOS Catalina 10.15 with the NUCLEO-F767ZI. Seems that the first test after plugging in the device will work, but any test after that will fail due to SYNC_FAILED.

0xc0170, 2020-01-02:
> Also having issues with greentea on macOS Catalina 10.15 with the NUCLEO-F767ZI. Seems that the first test after plugging in the device will work, but any test after that will fail due to SYNC_FAILED.

Not certain if this is in greentea or firmware. If you create a full bug report, ST team can have a look.