ARMmbed/mbed-os #11893: serial_clear() does not work for STM32F7

ArturoEsqCR, 2019-11-19:
### Description of defect

#### Target(s) affected by this defect ?
At least STM32F7, didn't check other targets

#### Toolchain(s) (name and version) displaying this defect ?
ARM Compiler v5

#### What version of Mbed-os are you using (tag or sha) ?
mbed-os-5.13.4 (commit 1bf6b20df9d3cd5f29f001ffc6f0d0fcbbb96118)

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
None

#### How is this defect reproduced ? 
By opening a serial port, and counting received messages: Simply poll it and attempt to clear the received bit in the RXNE register as such:

`uint32_t messagesReceived = 0;`
`serial_t serialPort;`
`serial_init(&serialPort, PA_9, PA_10);`

`// Data is polled continuously here`
`while (1) {`
`if (serial_readable(&serialPort)) {`
        `messagesReceived++;`
        `serial_clear(&serialPort); // Fails to clear RXNE bit that serial_readable() is polling`
    `}`
`}`

0xc0170, 2019-11-19:
cc @ARMmbed/team-st-mcd Please review

LMESTM, 2019-11-19:
Hello @ArturoEsqCR ,

Thanks for reporting this. 
I'd be interested to understand how you've encountered this issue as serial_clear is does not seem to be used by any Serial class in mbed.

Could you confirm if a patch like proposed below would fix your issue ? 

```
$ git diff targets/TARGET_STM/TARGET_STM32F7/serial_device.c
diff --git a/targets/TARGET_STM/TARGET_STM32F7/serial_device.c b/targets/TARGET_STM/TARGET_STM32F7/serial_device.c
index 66ba5c0369..f070d81f6d 100644
--- a/targets/TARGET_STM/TARGET_STM32F7/serial_device.c
+++ b/targets/TARGET_STM/TARGET_STM32F7/serial_device.c
@@ -252,6 +252,7 @@ void serial_clear(serial_t *obj)
     struct serial_s *obj_s = SERIAL_S(obj);
     UART_HandleTypeDef *huart = &uart_handlers[obj_s->index];

+    volatile uint32_t tmpval __attribute__((unused)) = huart->Instance->RDR; // Clear RXNE
     __HAL_UART_CLEAR_IT(huart, UART_FLAG_TXE);
     _
```

ArturoEsqCR, 2019-11-19:
Hi @LMESTM,

I was looking for a simple way to poll the serial port for new messages and found serial_clear. I didn't access it through any additional mbed OS layer, but directly through the STM32F7 library files. I was just trying to clear the RXNE bit, but at the moment such function attempts to write to a reserved bit of the ICR register.

And yes, such patch fixes the issue.

LMESTM, 2019-11-25:
Proposed fix pushed in #11938 ... @ArturoEsqCR - your feedback is welcome

ArturoEsqCR, 2019-11-25:
This worked fine for the use case I intended to have. Since there are no defined requirements of what this function is supposed to do, I guess it is better to have it clearing Rx only.

LMESTM, 2019-11-25:
@ArturoEsqCR I think it makes sense to clear errors as well if they're not yet cleared, otherwise the IP would be blocked. Especially if RX was not read yet, there are good chances that Overrun has been flagged as well ...

ArturoEsqCR, 2019-11-25:
Well, yeah.. that's true. Maybe it's better that way.

closed 2019-11-27 by 0xc0170