ARMmbed/mbed-os #11906: ARMC6 seems to ignore pre-processor directive

amitchone, 2019-11-20:
### Description of defect
I'm attempting to conditionally include a file using pre-processor directives as below:

    #if BOARD == TP_EARHART_V1_0_0
        #include "xyz.h"
    #endif

My issue is that the file is being included and, as such, causing a ton of errors. It appears that the statement is just being completed ignored. Now the first thing that popped into my mind was _"Well you're obviously redefining the BOARD macro somewhere else and it actually should evaluate to true"_. So I decided to test that:

    #define STRING2(x) #x
    #define STRING(x) STRING2(x)
    #pragma message "BOARD = " STRING(BOARD)

    #if BOARD ==TP_EARHART_V1_0_0
        #include "LorawanTP.h"
    #endif

This yields the message: 
    
    [Warning] node_flow.h@23,9: BOARD = WRIGHT_V1_0_0 [-W#pragma-messages]

I've even verified that, within the `#if` statement, the value of `BOARD` is definitely not equal to `TP_EARHART_V1_0_0`. I'm not really sure where to look now. 

I should note that I have also attempted, to no avail, the following:

    #if defined BOARD && (BOARD == TP_EARHART_V1_0_0)

#### Target(s) affected by this defect ?
I'm using a custom target (defined in `custom_targets.json`) based on the Murata Type ABZ. 

#### Toolchain(s) (name and version) displaying this defect ?
ARM Compiler v6.11
Clangd 8.0.0

#### What version of Mbed-os are you using (tag or sha) ?
Mbed OS 5.14.1

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
Mbed Studio 0.7.0

#### How is this defect reproduced ? 
As in the description of the defect.

0xc0170, 2019-11-20:
I would start with setting a board to a constant like 1 or 2 and see if pre-processor is being invoked and values are correct - that would prove that preprocessor runs prior compilation.
Based on what you described, you want to compare strings. Did it work in any other compiler - have you tried? I don't think it works the way you had it described.

See for example gcc ref what it supports: https://gcc.gnu.org/onlinedocs/cpp/If.html and of course C FAQ: http://www.c-faq.com/cpp/ifstrcmp.html

There are tricks you can find to achieve it if the c faq suggestion above does not work for you. I can see similar answer was given to your question on stackoverflow.

amitchone, 2019-11-20:
Yes, as I expected I just fundamentally misunderstood what was happening. All is well, thanks for the response however!

closed 2019-11-20 by amitchone