ARMmbed/mbed-os #11912: NRF52840 USB device tests failures

maclobdell, 2019-11-20:
### Description of defect
When running Mbed OS USB Device tests, there are several failures.  Some seem intermittent, such as tests-usb_device-basic, which on some instances will all pass.  However, mostly the results are very consistent.  It also seems consistent across all toolchains.  Full logs are attached.  

#### Target(s) affected by this defect ?
NRF52840_DK, EP_AGORA

#### Toolchain(s) (name and version) displaying this defect ?
ARM, IAR
Didn't test GCC_ARM, but suspect same result

#### What version of Mbed-os are you using (tag or sha) ?
Mbed OS 5.14.2

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
Windows 7 test machine
Zadig `libusb-win32 (v1.2.6.0)` driver
mbed-cli (1.10.0)
mbed-clitest (0.21.0)
mbed-cloud-sdk (2.0.8)
mbed-flasher (0.10.1)
mbed-greentea (1.7.2)
mbed-host-tests (1.5.8)
mbed-ls (1.7.8)
mbed-os-tools (0.0.9)

#### How is this defect reproduced ? 
1. Clone Mbed OS
1. Connect usb cable for debug interface, connect usb cable for target usb
1. Set up environment (installer Zadig drivers) pr mbed-os\TESTS\usb_device\README.md
1. Run tests with the following command

```
mbed test -m NRF52840_DK -t ARM -n tests-usb_device*
```

Results

```
| target            | platform_name | test suite              | test case                                       | passed | failed | result  | elapsed_time (sec) |
|-------------------|---------------|-------------------------|-------------------------------------------------|--------|--------|---------|--------------------|
| NRF52840_DK-ARMC6 | NRF52840_DK   | tests-usb_device-basic  | endpoint test abort                             | 0      | 0      | SKIPPED | 0.0                |
| NRF52840_DK-ARMC6 | NRF52840_DK   | tests-usb_device-basic  | endpoint test data correctness                  | 1      | 0      | OK      | 1.39               |
| NRF52840_DK-ARMC6 | NRF52840_DK   | tests-usb_device-basic  | endpoint test data toggle reset                 | 0      | 0      | SKIPPED | 0.0                |
| NRF52840_DK-ARMC6 | NRF52840_DK   | tests-usb_device-basic  | endpoint test halt                              | 0      | 1      | FAIL    | 8.58               |
| NRF52840_DK-ARMC6 | NRF52840_DK   | tests-usb_device-basic  | endpoint test parallel transfers                | 0      | 0      | SKIPPED | 0.0                |
| NRF52840_DK-ARMC6 | NRF52840_DK   | tests-usb_device-basic  | endpoint test parallel transfers ctrl           | 0      | 0      | SKIPPED | 0.0                |
| NRF52840_DK-ARMC6 | NRF52840_DK   | tests-usb_device-basic  | usb control basic test                          | 1      | 0      | OK      | 3.93               |
| NRF52840_DK-ARMC6 | NRF52840_DK   | tests-usb_device-basic  | usb control sizes test                          | 1      | 0      | OK      | 0.47               |
| NRF52840_DK-ARMC6 | NRF52840_DK   | tests-usb_device-basic  | usb control stall test                          | 1      | 0      | OK      | 0.38               |
| NRF52840_DK-ARMC6 | NRF52840_DK   | tests-usb_device-basic  | usb control stress test                         | 1      | 0      | OK      | 0.53               |
| NRF52840_DK-ARMC6 | NRF52840_DK   | tests-usb_device-basic  | usb device reset test                           | 1      | 0      | OK      | 1.95               |
| NRF52840_DK-ARMC6 | NRF52840_DK   | tests-usb_device-basic  | usb repeated construction destruction test      | 1      | 0      | OK      | 2.3                |
| NRF52840_DK-ARMC6 | NRF52840_DK   | tests-usb_device-basic  | usb soft reconnection test                      | 1      | 0      | OK      | 1.78               |
| NRF52840_DK-ARMC6 | NRF52840_DK   | tests-usb_device-hid    | Configuration descriptor, generic               | 0      | 0      | ERROR   | 0.0                |
| NRF52840_DK-ARMC6 | NRF52840_DK   | tests-usb_device-hid    | Configuration descriptor, keyboard              | 0      | 0      | SKIPPED | 0.0                |
| NRF52840_DK-ARMC6 | NRF52840_DK   | tests-usb_device-hid    | Configuration descriptor, mouse                 | 0      | 0      | SKIPPED | 0.0                |
| NRF52840_DK-ARMC6 | NRF52840_DK   | tests-usb_device-hid    | HID class descriptors, generic                  | 0      | 0      | SKIPPED | 0.0                |
| NRF52840_DK-ARMC6 | NRF52840_DK   | tests-usb_device-hid    | HID class descriptors, keyboard                 | 0      | 0      | SKIPPED | 0.0                |
| NRF52840_DK-ARMC6 | NRF52840_DK   | tests-usb_device-hid    | HID class descriptors, mouse                    | 0      | 0      | SKIPPED | 0.0                |
| NRF52840_DK-ARMC6 | NRF52840_DK   | tests-usb_device-hid    | Raw input/output, 1-byte reports                | 0      | 0      | SKIPPED | 0.0                |
| NRF52840_DK-ARMC6 | NRF52840_DK   | tests-usb_device-hid    | Raw input/output, 20-byte reports               | 0      | 0      | SKIPPED | 0.0                |
| NRF52840_DK-ARMC6 | NRF52840_DK   | tests-usb_device-hid    | Raw input/output, 64-byte reports               | 0      | 0      | SKIPPED | 0.0                |
| NRF52840_DK-ARMC6 | NRF52840_DK   | tests-usb_device-msd    | mount/unmount and data test - Heap block device | 0      | 0      | SKIPPED | 0.0                |
| NRF52840_DK-ARMC6 | NRF52840_DK   | tests-usb_device-msd    | mount/unmount test - Heap block device          | 0      | 1      | FAIL    | 49.95              |
| NRF52840_DK-ARMC6 | NRF52840_DK   | tests-usb_device-msd    | storage initialization                          | 1      | 0      | OK      | 0.04               |
| NRF52840_DK-ARMC6 | NRF52840_DK   | tests-usb_device-serial | CDC RX multiple bytes                           | 0      | 0      | SKIPPED | 0.0                |
| NRF52840_DK-ARMC6 | NRF52840_DK   | tests-usb_device-serial | CDC RX multiple bytes concurrent                | 0      | 0      | SKIPPED | 0.0                |
| NRF52840_DK-ARMC6 | NRF52840_DK   | tests-usb_device-serial | CDC RX single bytes                             | 0      | 0      | SKIPPED | 0.0                |
| NRF52840_DK-ARMC6 | NRF52840_DK   | tests-usb_device-serial | CDC RX single bytes concurrent                  | 0      | 0      | SKIPPED | 0.0                |
| NRF52840_DK-ARMC6 | NRF52840_DK   | tests-usb_device-serial | CDC USB reconnect                               | 0      | 0      | ERROR   | 0.0                |
[... 68 lines cut ...]
| EP_AGORA-ARMC6 | EP_AGORA      | tests-usb_device-hid    | Configuration descriptor, keyboard              | 0      | 0      | SKIPPED | 0.0                |
| EP_AGORA-ARMC6 | EP_AGORA      | tests-usb_device-hid    | Configuration descriptor, mouse                 | 0      | 0      | SKIPPED | 0.0                |
| EP_AGORA-ARMC6 | EP_AGORA      | tests-usb_device-hid    | HID class descriptors, generic                  | 0      | 0      | SKIPPED | 0.0                |
| EP_AGORA-ARMC6 | EP_AGORA      | tests-usb_device-hid    | HID class descriptors, keyboard                 | 0      | 0      | SKIPPED | 0.0                |
| EP_AGORA-ARMC6 | EP_AGORA      | tests-usb_device-hid    | HID class descriptors, mouse                    | 0      | 0      | SKIPPED | 0.0                |
| EP_AGORA-ARMC6 | EP_AGORA      | tests-usb_device-hid    | Raw input/output, 1-byte reports                | 0      | 0      | SKIPPED | 0.0                |
| EP_AGORA-ARMC6 | EP_AGORA      | tests-usb_device-hid    | Raw input/output, 20-byte reports               | 0      | 0      | SKIPPED | 0.0                |
| EP_AGORA-ARMC6 | EP_AGORA      | tests-usb_device-hid    | Raw input/output, 64-byte reports               | 0      | 0      | SKIPPED | 0.0                |
| EP_AGORA-ARMC6 | EP_AGORA      | tests-usb_device-msd    | mount/unmount and data test - Heap block device | 0      | 0      | SKIPPED | 0.0                |
| EP_AGORA-ARMC6 | EP_AGORA      | tests-usb_device-msd    | mount/unmount test - Heap block device          | 0      | 1      | FAIL    | 49.64              |
| EP_AGORA-ARMC6 | EP_AGORA      | tests-usb_device-msd    | storage initialization                          | 1      | 0      | OK      | 0.04               |
| EP_AGORA-ARMC6 | EP_AGORA      | tests-usb_device-serial | CDC RX multiple bytes                           | 0      | 0      | SKIPPED | 0.0                |
[... cut ...]
| EP_AGORA-ARMC6 | EP_AGORA      | tests-usb_device-serial | Serial printf/scanf                             | 0      | 0      | SKIPPED | 0.0                |
| EP_AGORA-ARMC6 | EP_AGORA      | tests-usb_device-serial | Serial terminal reopen                          | 0      | 0      | SKIPPED | 0.0                |
```
[nrf52840_usb_tests_log_20Nov2019_3.txt](https://github.com/ARMmbed/mbed-os/files/3871850/nrf52840_usb_tests_log_20Nov2019_3.txt)
[nrf52840_usb_tests_log_20Nov2019.txt](https://github.com/ARMmbed/mbed-os/files/3871851/nrf52840_usb_tests_log_20Nov2019.txt)

40Grit, 2019-11-20:
@AGlass0fMilk

40Grit, 2019-11-20:
@maclobdell  take a look at Georges PR/Issues that enabled USB support for NRF. There were some issues on the PC driver side that caused intermittent failures.

maclobdell, 2019-11-21:
@40Grit & @AGlass0fMilk  - I don't see any recent PRs or issues that seem related to this topic.  Have you guys seen these specific issues before?

40Grit, 2019-11-22:
Result of our conversation yesterday was to next try the tests using a couple different host PCs and or operations systems.

maciejbocianski, 2019-11-23:
There were some changes see: https://github.com/ARMmbed/mbed-os/pull/11458

maciejbocianski, 2019-11-25:
Also noticed another fail on this target:
```
[1574427582.77][CONN][RXD] >>> Running case #7: 'usb repeated construction destruction test'...
[1574427582.85][CONN][INF] found KV pair in stream: {{__testcase_start;usb repeated construction destruction test}}, queued...
[1574427583.53][CONN][RXD]
[1574427583.53][CONN][RXD]
[1574427583.55][CONN][RXD] ++ MbedOS Error Info ++
[1574427583.59][CONN][RXD] Error Status: 0x8015010E Code: 270 Module: 21
[1574427583.65][CONN][RXD] Error Message: Device state is "Powered" or "Detached"
[1574427583.67][CONN][RXD] Location: 0x90D9
[1574427583.69][CONN][RXD] Error Value: 0x0
[1574427583.81][CONN][RXD] Current Thread: rtx_idle Id: 0x20002F0C Entry: 0xE091 StackSize: 0x200 StackMem: 0x20002F98 SP: 0x2003FF28
[1574427583.89][CONN][RXD] For more info, visit: https://mbed.com/s/error?error=0x8015010E&tgt=NRF52840_DK
[1574427583.91][CONN][RXD] -- MbedOS Error Info --
[1574427583.91][CONN][RXD]
[1574427583.97][CONN][RXD] = System will be rebooted due to a fatal error =
```

0xc0170, 2020-02-24:
> I'm looking into it ...

@maciejbocianski Was there any update

closed 2020-06-05 by ciarmcom