ARMmbed/mbed-os #11922: Mbed default Bootloader for NUCLEO_WB55RG

felipeLeast, 2019-11-21:
Hi guys,

I'm starting to use the board NUCLEO_WB55RG, and I saw that there's no bootloader default on the mbed->feature->FEATURE_BOOTLOADER for this board. 

My question is if those .bin in this folder are from this project: 
https://github.com/ARMmbed/mbed-bootloader-extended/
Or from another one that is closed of.

Does anyone know that? And if I can just build this project with another board and if it will work.

Thank you in advance,

teetak01, 2019-11-22:
Hi @felipeLeast https://github.com/armmbed/mbed-bootloader is the repository from which you can compile bootloaders. As that is a new target, you probably need to write a new configuration for that board.

0xc0170, 2019-11-22:
Hi @felipeLeast , if this is not a bug (as the default template suggests, you removed entire template), but rather a question, better place to ask is on our forums: http://forum.mbed.com/

Thanks @teetak01 for the answer, a good starting point.

I'll close this question as answered.

closed 2019-11-22 by 0xc0170