ARMmbed/mbed-os #11924: [LittleFS] Meet bad block on master

ccli8, 2019-11-22:
### Description of defect

#### Target(s) affected by this defect ?
K64F

#### Toolchain(s) (name and version) displaying this defect ?
ARM Compiler 6.11

#### What version of Mbed-os are you using (tag or sha) ?
4b41a6f164 Merge pull request #11915 from AriParkkila/tests-sendtimeout

#### mbed-os-example-filesystem
17cbb12 Merge pull request #106 from hugueskamba/hk-iotcore-1318-remove-fp-usage-in-example-projects

#### How is this defect reproduced ? 

Without modification, compile `mbed-os-example-filesystem` with:
```sh
mbed compile -m K64F -t ARMC6
```

Flash and see error with LittleFS:
```
--- Mbed OS filesystem example ---
Mounting the filesystem... lfs error:494: Corrupted dir pair at 0 1
lfs error:2222: Invalid superblock at 0 1
Fail :(
No filesystem found, formatting... lfs debug:617: Bad block at 3
lfs debug:617: Bad block at 4
lfs debug:617: Bad block at 5
lfs debug:617: Bad block at 6
lfs debug:617: Bad block at 7
lfs debug:617: Bad block at 8
```

If I change mbed-os version to mbed-os-5.14.2, LittleFS is OK:
```
--- Mbed OS filesystem example ---
Mounting the filesystem... OK
Opening "/fs/numbers.txt"... OK
Incrementing numbers (10/10)... OK
Closing "/fs/numbers.txt"... OK
Opening the root directory... OK
root directory:
    .
    ..
    numbers.txt
Closing the root directory... OK
Opening "/fs/numbers.txt"... OK
numbers:
    3
    4
    5
    6
    7
    8
    9
    10
    11
    12
Closing "/fs/numbers.txt"... OK
Unmounting... OK
Mbed OS filesystem example done!

```

0xc0170, 2019-11-22:
cc @ARMmbed/mbed-os-storage

kjbracey-arm, 2019-11-28:
Duplicate of #11879

ccli8, 2019-12-04:
According to https://github.com/ARMmbed/mbed-os/issues/11879#issuecomment-559405791, check master and it's OK.

With #11879 open, close this one.

closed 2019-12-04 by ccli8