ARMmbed/mbed-os #11929: Ethernet: devices with LAN8742 and using STM32 HAL work only accidentaly

JojoS62, 2019-11-22:
### Description of defect

Ethernet: devices with LAN8742 and using STM32 HAL work only accidentaly

I've tried to use an external LAN8720 board with a STM32F407. This did not work and after tracing the problem I found the cause in the auto negotiation. When I turn it off, it works.
Further tracing revealed that the HAL code is using a register map that fits to DP83848 PHY, but register addresses in LAN87xx are different.
The result of the auto-negotiation is checked in register PHY_SR = 16 dez. This registert does not exist in my LAN8720. In other STM boards, LAN8742 ist used, e.g. DISCO_F746NG, there the register exists but has a different function. So tests with this board may succeed, but accidentaly.
As a workaround I can set a fixed link speed and mode, that is skipping the autonegotiation.

https://github.com/ARMmbed/mbed-os/blob/bd0cbdbad2120823b4b2e5f3b58cd1a94035cc5b/targets/TARGET_STM/TARGET_STM32F4/device/stm32f4xx_hal_eth.c#L369
https://github.com/ARMmbed/mbed-os/blob/bd0cbdbad2120823b4b2e5f3b58cd1a94035cc5b/targets/TARGET_STM/TARGET_STM32F7/device/stm32f7xx_hal_eth.c#L412

#### Target(s) affected by this defect ?
custom target with LAN8720

#### Toolchain(s) (name and version) displaying this defect ?
gcc-arm

#### What version of Mbed-os are you using (tag or sha) ?
5.14.0
5.14.2

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
mbed-cli

#### How is this defect reproduced ?

0xc0170, 2019-11-25:
cc @ARMmbed/team-st-mcd

kjbracey-arm, 2019-11-28:
I've been aware of the "hard-coded for the PHY on the original dev kit" factor in various platforms' Ethernet drivers for a while, but I've not seen it cause any issues yet. 

It would certainly be useful in general to ensure that Ethernet drivers by default didn't use proprietary registers, and the target had to explicitly name its PHY to use proprietary ones.

JojoS62, 2019-11-28:
I agree. There is also not too much special in the PHY init, in this case it is just the checking for the link status. There are standardized common registers and a few special ones. The PHY has also a vendor specific ID that can be used to check for the necessary handling, the guys from NXP do this in their driver. The code could handle standards like DP83848 and LAN8720/8742 by default and provide a callback for unhandled PHYs.

jeromecoutant, 2020-02-19:
Hi
I think #12464 should correct issue ?
Regards

jeromecoutant, 2020-03-11:
As #12464 is merged, can we close this one ?
Thx

JojoS62, 2020-03-11:
Yes, fixed.

closed 2020-03-11 by JojoS62