ARMmbed/mbed-os #11931: Filesystem tests failing with MbedCRC changes

trowbridgec, 2019-11-22:
### Description of defect

Since the MbedCRC changes were added in https://github.com/ARMmbed/mbed-os/pull/11559 the filesystem tests are failing for the `EP_AGORA` target.

Testing with https://github.com/ARMmbed/mbed-os/commit/aae04b2516903a1862afdc6c3a815111c6181556 the tests pass, but testing with https://github.com/ARMmbed/mbed-os/commit/88119722012863e38207fbc3d587ef93a7b91b75, the DUT hits a hard fault state and the test fails.
[image]

Log of the test failure:
```
mbed test -t GCC_ARM -m EP_AGORA -n features-storage-tests-filesystem* --run -v
[mbed] WARNING: Python 3 is not yet fully supported: Python errors may occur when compiling, testing and exporting
---
[mbed] Working path "/home/chris/mbed-os" (program)
[mbed] WARNING: Could not find mbed program in current path "/home/chris/mbed-os".
       You can fix this by calling "mbed new ." in the root of your program.
---
[mbed] Exec "mbedgt --test-spec ./BUILD/tests/EP_AGORA/GCC_ARM/test_spec.json -n features-storage-tests-filesystem* -V" in "/home/chris/mbed-os"
mbedgt: greentea test automation tool ver. 1.7.2
mbedgt: test specification file './BUILD/tests/EP_AGORA/GCC_ARM/test_spec.json' (specified with --test-spec option)
mbedgt: using './BUILD/tests/EP_AGORA/GCC_ARM/test_spec.json' from current directory!
mbedgt: detecting connected mbed-enabled devices...
mbedgt: detected 1 device
        | platform_name | platform_name_unique | serial_port  | mount_point          | target_id                                        |
        |---------------|----------------------|--------------|----------------------|--------------------------------------------------|
        | EP_AGORA      | EP_AGORA[0]          | /dev/ttyACM0 | /media/chris/DAPLINK | 2600360248824e4500127008dd89002c8761000097969900 |
mbedgt: processing target 'EP_AGORA' toolchain 'GCC_ARM' compatible platforms... (note: switch set to --parallel 1)
        | platform_name | platform_name_unique | serial_port       | mount_point          | target_id                                        |
        |---------------|----------------------|-------------------|----------------------|--------------------------------------------------|
        | EP_AGORA      | EP_AGORA[0]          | /dev/ttyACM0:9600 | /media/chris/DAPLINK | 2600360248824e4500127008dd89002c8761000097969900 |
mbedgt: test case filter (specified with -n option)
        features-storage-tests-filesystem*
        test filtered in 'features-storage-tests-filesystem-general_filesystem'
mbedgt: running 1 test for platform 'EP_AGORA' and toolchain 'GCC_ARM'
        use 1 instance of execution threads for testing
mbedgt: checking for 'host_tests' directory above image directory structure
        'host_tests' directory not found: two directory levels above image path checked
mbedgt: selecting test case observer...
        calling mbedhtrun: mbedhtrun -m EP_AGORA -p /dev/ttyACM0:9600 -f "BUILD/tests/EP_AGORA/GCC_ARM/features/storage/TESTS/filesystem/general_filesystem/general_filesystem.hex" -d /media/chris/DAPLINK -c default -t 2600360248824e4500127008dd89002c8761000097969900 -r default -C 6 --sync 5 -P 60
mbedgt: mbed-host-test-runner: started
[1574456537.54][HTST][INF] host test executor ver. 0.0.9
[1574456537.54][HTST][INF] copy image onto target...
[1574456537.55][COPY][INF] Waiting up to 60 sec for '2600360248824e4500127008dd89002c8761000097969900' mount point (current is '/media/chris/DAPLINK')...
[1574456550.08][mbedls.lstools_base]MBED with target id '2600360248824e4500127008dd89002c8761000097969900' is connected, but not mounted. Use the '-u' flag to include it in the list.
[1574456552.78][HTST][INF] starting host test process...
[1574456552.79][CONN][INF] starting connection process...
[1574456552.79][CONN][INF] notify event queue about extra 60 sec timeout for serial port pooling
[1574456552.79][CONN][INF] initializing serial port listener...
[1574456552.79][PLGN][INF] Waiting up to 60 sec for '2600360248824e4500127008dd89002c8761000097969900' serial port (current is '/dev/ttyACM0')...
[1574456552.79][HTST][INF] setting timeout to: 60 sec
[1574456552.80][SERI][INF] serial(port=/dev/ttyACM0, baudrate=9600, read_timeout=0.01, write_timeout=5)
[1574456552.80][SERI][INF] reset device using 'default' plugin...
[1574456552.91][SERI][INF] waiting 1.00 sec after reset
[1574456553.91][SERI][INF] wait for it...
[1574456553.91][SERI][TXD] mbedmbedmbedmbedmbedmbedmbedmbedmbedmbed
[1574456553.91][CONN][INF] sending up to 5 __sync packets (specified with --sync=5)
[1574456553.91][CONN][INF] sending preamble '57aa69a0-6ecf-4c63-9734-61d4a0655f40'
[1574456553.91][SERI][TXD] {{__sync;57aa69a0-6ecf-4c63-9734-61d4a0655f40}}
[1574456554.05][CONN][RXD] mbedmbedmbedmbedmbedmbedmbedmbed
[1574456554.09][CONN][INF] found SYNC in stream: {{__sync;57aa69a0-6ecf-4c63-9734-61d4a0655f40}} it is #0 sent, queued...
[1574456554.09][HTST][INF] sync KV found, uuid=57aa69a0-6ecf-4c63-9734-61d4a0655f40, timestamp=1574456554.090096
[1574456554.12][CONN][INF] found KV pair in stream: {{__version;1.3.0}}, queued...
[1574456554.12][HTST][INF] DUT greentea-client version: 1.3.0
[1574456554.13][CONN][INF] found KV pair in stream: {{__timeout;300}}, queued...
[1574456554.13][HTST][INF] setting timeout to: 300 sec
[1574456554.18][CONN][INF] found KV pair in stream: {{__host_test_name;default_auto}}, queued...
[1574456554.18][HTST][INF] host test class: '<class 'mbed_os_tools.test.host_tests.default_auto.DefaultAuto'>'
[1574456554.18][HTST][INF] host test setup() call...
[1574456554.18][HTST][INF] CALLBACKs updated
[1574456554.18][HTST][INF] host test detected: default_auto
[1574456554.20][CONN][INF] found KV pair in stream: {{__testcase_count;75}}, queued...
[... 269 lines cut ...]
       Code: 1
       Path: "/home/chris/mbed-os"
       Command: "mbedgt --test-spec ./BUILD/tests/EP_AGORA/GCC_ARM/test_spec.json -n features-storage-tests-filesystem* -V"
       Tip: You could retry the last command with "-v" flag for verbose output
---
```

#### Target(s) affected by this defect ?
`EP_AGORA`

#### Toolchain(s) (name and version) displaying this defect ?
GCC_ARM

#### What version of Mbed-os are you using (tag or sha) ?
master

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
1.10.1

#### How is this defect reproduced ? 
```
git clone git@github.com:ARMmbed/mbed-os.git
cd mbed-os
mbed test -t GCC_ARM -m EP_AGORA -n features-storage-tests-filesystem*
```

trowbridgec, 2019-11-22:
@maclobdell @ARMmbed/team-embeddedplanet

0xc0170, 2019-11-25:
@kjbracey-arm

0xc0170, 2019-11-25:
@trowbridgec Have you tested these PRs : #11896  and #11897 ? They will land once we have 5.15rc1 out.

kjbracey-arm, 2019-11-25:
Probably duplicate of issue #11879, should be fixed by #11896.

This issue's got better red pointy blame arrows on the log though.

0xc0170, 2019-11-25:
> Probably duplicate of issue #11879, should be fixed by #11896.

Lets close this , @trowbridgec reopen with a reason if this differs

closed 2019-11-25 by 0xc0170

trowbridgec, 2019-11-25:
@0xc0170 @kjbracey-arm I tested with #11896, and it passed. Thanks and sorry for the duplicate issue.

Also, glad you appreciated my art skills!