ARMmbed/mbed-os #11934: Add the ability to disable a peripheral

embeddedteam103, 2019-11-24:
### Description of defect

Hello!

I am using some SPIs in my projects in the following way - a user sends some bytes to the MCU and then the MCU sends the given bytes via the chosen SPI peripheral.
The thing is,  that this project suppose to be low power and I dont want the SPI peripherals so stay active all the time. 
Is there a way to disable the SPI peripheral clock with the SPI driver?
Is there even a way to disable any peripheral after initiated via `mbed`?

thanks!

#### Target(s) affected by this defect ?
All

#### Toolchain(s) (name and version) displaying this defect ?
IAR

#### What version of Mbed-os are you using (tag or sha) ?
5.14

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
mbed-cli

#### How is this defect reproduced ?

40Grit, 2019-11-24:
Technically this is supposed to be posted in the forum. However it is deffinitely am issue.

If I had the ability to join the Mbed working group I would emphasize the responsibility of the silicon vendors to ensure their API implementations minimize power consumption when a peripheral is "de-initialized".

The API's may need to be updated as well to provide more configurable peripheral and clock states.

0xc0170, 2019-11-25:
It would definitely be better to post on the forum as this is enhancement (we will need to add this functionality).

There has been already done for some drivers, for instance #10924.

> If I had the ability to join the Mbed working group I would emphasize the responsibility of the silicon vendors to ensure their API implementations minimize power consumption when a peripheral is "de-initialized".

Noted. cc @bulislaw

embeddedteam103, 2019-11-25:
OK, I'll move it to the forum.

closed 2019-11-26 by 0xc0170