ARMmbed/mbed-os #11940: Implement I2C Slave functionality for TARGET_NRF52

kevingre, 2019-11-25:
### Description of defect

I2C Slave functionality is not implemented in the TARGET_NRF52 target.  It is stubbed out and contains debug printf statements only.

#### Target(s) affected by this defect ?
TARGET_NRF52 

#### Toolchain(s) (name and version) displaying this defect ?
All

#### What version of Mbed-os are you using (tag or sha) ?
5.14

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
mbed-cli

#### How is this defect reproduced ? 
Examine Lines938-1006 in i2c_api.c located at https://github.com/ARMmbed/mbed-os/blob/master/targets/TARGET_NORDIC/TARGET_NRF5x/TARGET_NRF52/i2c_api.c

0xc0170, 2019-11-26:
`DEVICE_I2CSLAVE` should not be defined for this target. We'll send a fix. 

cc @desmond-blue 

The implementation still needs to be done. PR welcome!

0xc0170, 2019-12-04:
I've checked, the target does not support slave functionality, it is neither enabled.

To have this enabled, please send PR implementing the new functionality. Or request it via forum as a new feature request.

closed 2019-12-04 by 0xc0170