ARMmbed/mbed-os #11951: Stack overflow on Wi-SUN QUECTEL EC2x initialization

felipeLeast, 2019-11-26:
[debug.error.txt](https://github.com/ARMmbed/mbed-os/files/3893556/debug.error.txt)

**Description of defect**
Intermitent stack overflow when runing default code from master on same configuration as describe on README file. When I start runing the code some times I get the following stack overflow right after initiating the cellular module. It takes a few resets before I the code start running, but eventually does.

**Step to reproduce:**
I just compile the master code from nanostack-border-router to F64K, and I let it run, eventually you will get that stack overflow

**Target(s) affected by this defect ?**
F64K with S2LP shield and EC25 demo board connected with uart, power key and reset pins to the device.

**Toolchain(s) (name and version) displaying this defect ?**
ARM_GCC

**What version of Mbed-os are you using (tag or sha) ?**
mbed-os-5.14.2

**What version(s) of tools are you using. List all that apply (E.g. mbed-cli)**
N/A

Can anyone shed a light on this issue please?

**Issue request type**

```
[ ] Question
[ ] Enhancement
[x] Bug
```

0xc0170, 2019-11-27:
cc @ARMmbed/mbed-os-wan

AnttiKauppila, 2019-12-04:
cc @ARMmbed/mbed-os-mesh

felipeLeast, 2019-12-10:
Hi @AnttiKauppila. I believe it has. It is something develop for that purpouse. Follows the link for the application that uses it. https://github.com/ARMmbed/nanostack-border-router

If you see in the configs folder there's a config for Wisun_Stm_s2lp_RF_cell.json which does exactly that. 

But if you think is just a stack overflow because is running out of memory I'll try to increase my heap memory to see if the stack overflow continious. 

But I still don't understand why it only happens sometimes.

mikaleppanen, 2020-01-08:
Cannot repeat the problem here locally, but I detected that some of the PPP debug traces were enabled also without debug flags. This resulted that PPP used more stack than needed.
Corrected that on:

https://github.com/ARMmbed/mbed-os/pull/12195

@felipeLeast Could you check whether the stack overflow happens anymore with the above modification.

mikter, 2020-01-14:
Closed as issue is resolved

closed 2020-01-14 by mikter