ARMmbed/mbed-os #11953: STM32 CAN 'filter handle' operation and documentation incorrect

bikeNomad, 2019-11-27:
### Description of defect

The mbed documentation of the CAN api refers to _filter handle_ in two places:
  * `int CAN::read(CANMessage &msg, int handle)`
  * `int CAN::filter(unsigned int id, unsigned int mask, CANFormat format, int handle)`

However, the meaning of the `handle` parameter in the STM32 code is different than the mbed documentation, and means two different, incompatible things.

The meaning of `handle` in the STM32 `CAN::read` is actually  a FIFO number (0 or 1) to read from.
However, there is no way in the mbed API to use FIFO 1, because the STM32 `CAN::filter` always sets filters to use FIFO 0. In the STM32 `CAN::filter`, `handle` is actually referring to a filter bank number.

Further, the default `CANFormat` for `CAN::filter` is `CANAny`, but the STM32 code can't configure filters for this. But instead of returning 0 if its `format` parameter is `CANAny`, the STM32 `can_filter()` returns 1 and does nothing (it should return 0 in this case).

#### Target(s) affected by this defect ?

All STM32 targets that have CAN peripherals.

See `mbed-os/targets/TARGET_STM/can_api.c`, lines 989-991 (`can_filter()`) and lines 825-865 (`can_read()`)

#### Toolchain(s) (name and version) displaying this defect ?

All

#### What version of Mbed-os are you using (tag or sha) ?

mbed-os-5.14

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

mbed-cli 1.10.1

#### How is this defect reproduced ?

  * Passing a non-zero `handle` parameter to `CAN::read` will never return a message
  * Passing a `format` parameter of `CANAny` to `CAN::filter` will not do anything but will return 1 anyway

0xc0170, 2019-11-27:
cc @ARMmbed/team-st-mcd

jeromecoutant, 2020-03-11:
Hi
I agree TARGET_STM/can_api.c is not using handle parameter as MBED API specification.
But I am sorry, but I don't have any CAN setup.
If you have some proposition to share, we will review it, and integrate the correction.

MubeenHCLite, 2021-05-25:
A PR is raised correcting the read and filter implementations of STM CAN API.
https://github.com/ARMmbed/mbed-os/pull/14677

jeromecoutant, 2021-06-25:
Let's close this one thanks to #14677 ?

closed 2021-06-25 by 0xc0170