ARMmbed/mbed-os #11955: STM32L486RG/MTB_ADV_WISE_1570 LPUART not functioning after wake up from deep sleep

soleilplanet, 2019-11-27:
### Description of defect
LPUART does not work when the device wakes up from deep sleep (MBED_TICKLESS=1).
It can be recovered by setting the baudrate again when the device wakes up, so probably LPUART was not properly re-initialized after wakeup from deep sleep.

#### Target(s) affected by this defect ?
STM32L486RG
MTB_ADV_WISE_1570

#### Toolchain(s) (name and version) displaying this defect ?
GCC
ARMC6
(IAR is not tested)

#### What version of Mbed-os are you using (tag or sha) ?
mbed-os-5.14.1

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
mbed-cli 1.10.0

#### How is this defect reproduced ? 
just write a simple UART echo program, and write a logic to put the device into deep sleep when receiving a certain key.
remember to disable UART traffic for both the LPUART and STDIO UART before entering deep sleep, and make sure they are enabled again after wake-up.
For example:
UARTSerial lpdev(LPUART1_RX, LPUART1_TX, 115200);
...
...
void uart_suspend(void)
{
    lpdev.enable_output(false);
    lpdev.enable_input(false);
    mbed_file_handle(STDIN_FILENO)->enable_output(false);
    mbed_file_handle(STDIN_FILENO)->enable_intput(false);
}
void uart_resume(void)
{
    mbed_file_handle(STDIN_FILENO)->enable_output(true);
    mbed_file_handle(STDIN_FILENO)->enable_intput(true);
    lpdev.enable_output(true);
    lpdev.enable_input(true);
}

Or, if you are using WISE-1570, you may run the cellular example, and after 1 iteration, make the device enter deep sleep for a while, then when it wakes up, do the 2nd iteration, and you will see that the modem can not receive anything after wake up.
The issue can be recovered when I add a lpdev.set_baud(baudrate) in uart_resume();

MarceloSalazar, 2019-11-27:
@ARMmbed/team-st-mcd please have a look

jeromecoutant, 2019-11-27:
Hi

Maybe you could first try to align MTB_ADV_WISE_1570 configuration in targets.json with NUCLEO_L486RG?
- clock_source = MSI ?
- enable LPTICKER ?
- enable MBED_TICKLESS, and other config like lpticker_delay_ticks ?

Thx

LMESTM, 2019-11-27:
1 further question: how do you wake-up from deep sleep ?

kjbracey-arm, 2019-11-28:
The relevant code has changed on master - as far as I can see in the current code, when you call `enable_input`, the serial port gets reinitialised by `SerialBase` and the baud/format settings are reset. So I think it should be okay now - on master and in the upcoming 5.15.

I'm not sure what happened in 5.14. It's possible we were relying on HAL UART retaining settings during deep sleep if you disabled input.

0xc0170, 2019-12-04:
@soleilplanet Please review the responses

soleilplanet, 2019-12-11:
@jeromecoutant low power clock source is using the default (I think it's HSI?), LPTICKER/MBED_TICKLESS has been enabled, and lpticker_delay_ticks was set to 4 and deep_sleep_latency was set to 3 
@kjbracey-arm Since we need to call enable_input() before and after deep sleep anyway, I think the way we are doing now in Mbed-OS 5.15 will get the problem disappeared. However, this problem does not happen on normal UART, and is only seen on LPUART, so I thought it might be better to have the re-initialization done in LPUART driver?
@LMESTM you may refer to the code I shared in another ticket: https://github.com/ARMmbed/mbed-os/issues/11954. I just disabled UART traffic and called ThisThread::sleep_for() to let the device enter deep sleep, and it wakes up because the time's up.

jeromecoutant, 2019-12-16:
> @jeromecoutant low power clock source is using the default (I think it's HSI?), LPTICKER/MBED_TICKLESS has been enabled, and lpticker_delay_ticks was set to 4 and deep_sleep_latency was set to 3

Please check in https://github.com/ARMmbed/mbed-os/blob/f77f4ea9110b4690628b9a46cd355cdce76467a8/targets/targets.json#L3737

            "clock_source": {
                "value": "USE_PLL_MSI",
            "lpticker_lptim": {
                "value": 1
        "overrides": { "lpticker_delay_ticks": 0 },
        "macros_add": [
            "MBED_TICKLESS",
            "EXTRA_IDLE_STACK_REQUIRED",
            "MBED_SPLIT_HEAP"

jeromecoutant, 2020-01-21:
Should we close this one ?
Thx

soleilplanet, 2020-02-07:
I'm ok with the baud/format reset in uart input enable as a solution since anyway we need to do it every time the device goes into deep sleep and then wake up. @kjbracey-arm should we close this one?
Thanks!

closed 2020-02-20 by 0xc0170