ARMmbed/mbed-os #11960: Bug in mbed_alloc_wrappers.cpp

DBS06, 2019-11-27:
### Description of defect
Setting
`"platform.heap-stats-enabled": false,`
`"platform.memory-tracing-enabled": false,`
in `mbed_app.json` does not deactivate tracing in `mbed_alloc_wrappers.cpp`

#### Target(s) affected by this defect ?
every target

#### How is this defect reproduced ? 
Set 
`"platform.heap-stats-enabled": false,`
`"platform.memory-tracing-enabled": false,`
in `mbed_app.json`.
Compile with debug profile and set a break point i.e. in `mbed_alloc_wrappers.cpp` inside `malloc_wrapper()`.

#### Solution:
All
`#ifdef MBED_MEM_TRACING_ENABLED`
and
`#ifdef MBED_HEAP_STATS_ENABLED`
should be replaced by:
`#if MBED_MEM_TRACING_ENABLED`
`#if MBED_HEAP_STATS_ENABLED`

0xc0170, 2019-11-28:
Thanks for the report, we will review proposed fix.

evedon, 2019-11-28:
These settings are disabled by default by being set to "null" not `false` see https://github.com/ARMmbed/mbed-os/blob/master/platform/mbed_lib.json/#L69-L115

kjbracey-arm, 2019-11-28:
There is clearly a problem in the implementation. All boolean options should be settable to `true` or `false`. 

This particular set of options has come up a number of times now, but we've not got around to fixing them.

Last time I talked about it was here: https://github.com/ARMmbed/mbed-os/issues/10987

0xc0170, 2019-11-28:
> These settings are disabled by default by being set to "null" not false see https://github.com/ARMmbed/mbed-os/blob/master/platform/mbed_lib.json/#L69-L115

Even with default ones, an app can do false and this should disable it in Mbed OS. It might be better if we start using false (also null supported) and do a check on its value rather.

Is this something we can clean up for 6?

DBS06, 2019-11-28:
> Even with default ones, an app can do false and this should disable it in Mbed OS. It might be better if we start using false (also null supported) and do a check on its value rather.

Is also my opinion.
Should I close this issue?

evedon, 2019-11-29:
> Even with default ones, an app can do false and this should disable it in Mbed OS. It might be better if we start using false (also null supported) and do a check on its value rather.
>
> Is this [...]

I agree that this is confusing and needs to be cleaned up. 
If we want to maintain backwards compatibility then these macros will need `#ifdef` as well.

kjbracey-arm, 2019-11-29:
> If we want to maintain backwards compatibility then these macros will need #ifdef as well.

I said that myself in the issue I linked to above, but now I can't think why I said that.

Just `#if MBED_CONF_XXX` should be fine - that is treated as `#if 0` if it's not defined (ie `null` in the JSON).

But maybe I've forgotten a subtlety.

0xc0170, 2019-12-03:
PR fixing it was merged, I'll close this one.

closed 2019-12-03 by 0xc0170