ARMmbed/mbed-os #11965: mbed assertation failed: status == osOK during external interrupt

slawekmarczak, 2019-11-27:
### assertion after triggering external interrupt

#### NUCLEO_F091RC

#### Toolchain(s) (name and version) displaying this defect ?
Mbed Studio 0.7.0

#### What version of Mbed-os are you using (tag or sha) ?
5.10.4

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
1.10.1

#### How is this defect reproduced ? 
By triggering falling edge during normal program operations

0xc0170, 2019-11-28:
We might need more details about your application. Looking at the assert you shared. It comes from :

(using 5.10 release branch to have the same code base as you)

```
void ThisThread::sleep_for(uint32_t millisec)
{
    osStatus_t status = osDelay(millisec);
    MBED_ASSERT(status == osOK);
}
```

osDelay should be functional, it is most of the time return OK status, just if you are in IRQ handler or IRQ are masked, returns `osErrorISR`. 

Check what it returns and why it is. Where is your app calling `ThisThread::sleep_for` from interrupt if that is the case?

slawekmarczak, 2019-11-29:
> Check what it returns and why it is. Where is your app calling ThisThread::sleep_for from interrupt if that is the case?

I have no idea why this app even could call `ThisThread::sleep_for` as I have bare-metal looks like architecture one infinite loop and some timer interrupt. I expect the event then it is working in one thread due to not disabled RTOS in .json config which I was not successful to proceed. 
Any `Ticker`, `Timer` or `Serial` can use it? How to investigate it under which function this call is hidden?

kjbracey-arm, 2019-12-03:
If you're using bare-metal in Mbed OS 5.10, then `ThisThread::sleep_for` will never work, as it relies on the RTOS being present. So the assert going to come from any call.

I'm not even sure what form of bare-metal support there was in 5.10 - I don't think there was anything officially supported. If you're going to stick with 5.10, I would suggest removing the `rtos` directory from your build - that would rapidly identify any users of the functions.

5.14 is a much better bet for bare-metal though.  It makes a subset of RTOS APIs, including `ThisThread::sleep_for`, available for use without the RTOS. That makes a bunch of stuff a lot easier.

0xc0170, 2019-12-03:
@slawekmarczak Have you progressed ? As this seems to be rather an application error than a bug, I would suggest closing and for more help, use http://forum.mbed.com/

0xc0170, 2019-12-09:
I'll close this as resolved.

closed 2019-12-09 by 0xc0170

imaniASC4, 2020-12-10:
Its the ticker. I just got the same error.