ARMmbed/mbed-os #11982: DNS error (-3009) when using ESP8266 with NUCLEO-091RC on some Wi-Fi router

zhiyong80, 2019-11-28:
### Description of defect
The problem seems about compatibility of mbed+ESP8266 with some WiFi router setup. When tested on one particular WiFi router, mbed+ESP8266 could connect to wifi, get IP address, network mask, gateway etc. But then stuck when trying connect to any http URLs. The error code is: -3009.
However, this doesn't happen with other Wi-Fi routers. The only thing particular about the problematic wifi router is, it has a large black list of URL such as youtbe/facebook etc. But test URLs in mbed-http-example are not in that blacklist, and can be accessed from a computer on the same network.
#### Target(s) affected by this defect ?
NUCLEO-091RC
#### Toolchain(s) (name and version) displaying this defect ?
mbed-cli(1.8.3)
#### What version of Mbed-os are you using (tag or sha) ?
5.13.4
#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
mbed-cli, 1.8.3 from windows installer 0.4.10
#### How is this defect reproduced ? 
This happens only with one particular WiFi router so far, cannot be repeated on other routers.

zhiyong80, 2019-11-29:
Some other information that might be relevant: 1) Hardware flow control was not enabled. 2) When esp8266.debug was set to true and stdio.baud-rate was set to 9600 instead of 115200, I got a different error 3012 with a previously working wifi router when debug was not turned on.

zhiyong80, 2019-11-29:
Visited the Wifi router again with hardware flow controll and debug more turned on, still got the same error. This seems to have something to do with google's 8.8.8.8 DNS server:
```
![image](https://user-images.githubusercontent.com/13650939/69887113-2f891d00-129a-11ea-9292-d5c11ad86ddd.png)
```

zhiyong80, 2019-11-30:
The problem seems to be ESP8266 not receiving anything from the DNS server. When URLs are replaced to IP address in HTTP request, the http example code runs just fine.
When a direct request to the same DNS server on the same network and PC, response via UDP, correct response is received.

0xc0170, 2019-12-03:
cc @ARMmbed/mbed-os-wan

AnttiKauppila, 2019-12-04:
Maybe HTTPExample does not need so much memory?
And when debug is enabled it affects significantly to memory usage
++ MbedOS Error Info ++
Error Status: 0x8001011F Code: 287 Module: 1
Error Message: Operator new out of memory

zhiyong80, 2019-12-05:
@AnttiKauppila this part actually puzzles me as well. In the example, there are two HTTP requests, 1st for GET, then a POST. If the 1st request fails somehow, http_req would be deleted, then a new request will be created, and mbed shouldn't crash. But oddly, as in log, once that 3009 error shows up, mbed os just crashes.

michalpasztamobica, 2019-12-31:
Hi @zhiyong80 , I wonder - when you disable the blacklisting in your router, are you still able to reproduce the issue?
Can you also try the same router with a device that has more memory, like K64F+ESP8266? 32kB of RAM that NUCLEO_F091RC has is not a lot...

zhiyong80, 2019-12-31:
@michalpasztamobica I tried STM32F401RE which has 64KB ram, and got the same results, also with mbed-os-5.14.2 and 5.15. The only notable difference with the newer version of mbed-os(5.15?) is, the app will try 3 different DNS servers before giving up. So I don't think this is a ram related problem.

More likely, this might be something related to ESP8266 itself. I dug up [some discussion](https://bbs.espressif.com/viewtopic.php?t=123) about DNS over UDP failures from years ago. This may still exist on some routers. 

I will test the router later today with SparkFun ESP8266 Thing which has nothing to do with Mbed and see what I get. 

About app crashing, it is unrelated. I accidentally added an extra line.
```
        HttpRequest* post_req = new HttpRequest(network, HTTP_POST, "SOME URL");
        post_req->set_header("Content-Type", "application/x-www-form-urlencoded");

        const char body[] = "DUMMY POST FORM DATA";

        HttpResponse* post_res = post_req->send(body, strlen(body));
        if (!post_res) {
            printf("HttpRequest failed (error code %d)\n", post_req->get_error());
            //return 1;
        } else {
           printf("\n----- HTTP POST response -----\n");
           dump_response(post_res);
        }
        delete post_res;   // This line crashes app when (!post_res) or causes memory leak when the code runs in a loop.
        delete post_req;
```

zhiyong80, 2020-01-01:
Just tested the same wifi hotspot with [Sparkfun ESP8266 Thing Dev](https://www.sparkfun.com/products/13231), HTTP GET to http://httpbin.org/status/418 worked just fine. So the problem is either in the wifi module (esp-wroom-02 + 1.70 at firmware) or ESP8266 driver on mbed side.

How can i further debug or test?

michalpasztamobica, 2020-01-02:
@AnttiKauppila , @VeijoPesonen , @SeppoTakalo , have you had a chance to try ESP8266's built-in DNS mechanism (AT+CIPDOMAIN)? Would you recommend that we try to use it instead of mbed's `nsapi_dns`? It could be made optional, dsiabled by default and only enable in mbed_app.json.
I don't see any better way to address this issue.

SeppoTakalo, 2020-01-02:
I don't think we have tried that.
But the linked forum post is about DNS issues when using ESP's own DNS resolver, so it makes me wonder whether it helps at all in this case.

If the device does not receive anything from the given DNS server, then I don't know how switching a resolver would make this issue disappear.  Maybe the WiFi router, in this case, does not route the DNS queries back to ESP.

This probably have to be verified with Wireshark, but to me, it sounds like network issue.
However, the mentioned memory leak or crash have to be studied.

michalpasztamobica, 2020-01-02:
@SeppoTakalo , further down in the forum thread the ESP team claims to have fixed the issue in their AT command set ("long time ago" they said in March 2018).
 Also - according to @zhiyong80 , the SparkFun device is using the built-in DNS resolver, so I think switching to it could fix the issue.

At the end of his previous post @zhiyong80 explained that he accidentally introduced the crash/memory leak, so no work needed for that.

@zhiyong80 , can you run wireshark, as @SeppoTakalo suggested and see if the packet towards ESP is actually visible?

zhiyong80, 2020-01-02:
@michalpasztamobica any instruction on how to tap communication between esp8266 and mbed? My plan is solder the RX line of USB-UART chip to TX line of ESP8266 and see what is actually received or not by mbed after sending AT+CIPDOMAIN.

michalpasztamobica, 2020-01-02:
Your suggestion is one way to go, but I suggest setting up Wireshark and connecting to the same network that ESP8266 is using. [Here](https://wiki.wireshark.org/CaptureSetup/WLAN) you can read about the basics. I think it would be good to sniff UDP on port 53 (typically used for DNS communication). If only ESP, the router and your host machine are on the network I don't expect much difficulty figuring out ingoing and outgoing packets. Not sure how well Wireshark is able to decode DNS packets structure, but I'd assume it is quite a usual use case. I didn't read the details, but perhaps [this](https://wiki.wireshark.org/DNS) helps? 
If you see the DNS packets going from the router to the ESP and still see that mbed doesn't get it - then it's time to either use soldering iron or try the built-in DNS resolution.

zhiyong80, 2020-01-03:
@michalpasztamobica Unfortunately I don't have much control of the wifi hotspot, I have password and can access it, but not much beyond. Spent a good part of today trying to figure out how to capture packets on windows with no luck so far. None of the laptops and USB wifi adapters I have supports monitor mode.

On a side note, I was able to confirm that AT+CIPDOMAIN does work on this router. I connecte a esp-wroom-02 evaluation board that has 1.70 firmware and the command works just fine.

```
AT+CWMODE=1
//0D0A
OK
AT+CWJAP="SSID","PASSWORD"
WIFI CONNECTED
WIFI GOT IP
//0D0A
OK
AT+CIPDOMAIN="iot.espressif.cn"
+CIPDOMAIN:115.29.202.58
//0D0A
OK
```

michalpasztamobica, 2020-01-03:
@zhiyong80 , I doubt that we have a direct line to Espressif, but I'll forward the question to @AnttiKauppila , perhaps he knows more...

It seems like switching to built-in DNS handling improved your situation and is a valid option to resolve this task? I'll try to add it to mbed's driver and come back to you with a PR to try out.

zhiyong80, 2020-01-03:
@michalpasztamobica Yes switching to built-in DNS command can improve my situation with that particular wifi router. I tried to implement gethostbyname() in ESP8266Interface but that task seems beyond me due to layers of abstraction. 

My teammate came up with a bandage by tweaking dns_lookup() in ESP8266 driver:
```
bool done = _parser.send("AT+CIPDOMAIN=\"%s\"", name) && _parser.recv("+CIPDOMAIN:%15[^\n]\n", ip) && _parser.recv("OK\n"); // This line as of 5.15.1 seems not working because ESP8266 sends out extra 0x0D0x0AOK.
```
The following is code that works on Sparkfun ESP8266 Thing, included in case it will help. I assumed ESP8266 addon for Arduino uses builtin AT command to resolve DNS, but I am not sure. 
```
#include <Arduino.h>
#include <ESP8266WiFi.h>
#include <ESP8266WiFiMulti.h>
#include <ESP8266HTTPClient.h>
#include <WiFiClient.h>
ESP8266WiFiMulti WiFiMulti;

void setup() {
Serial.begin(115200);
Serial.setDebugOutput(true);

Serial.println();
Serial.println();
Serial.println();

for (uint8_t t = 4; t > 0; t--) {
Serial.printf("[SETUP] WAIT %d...\n", t);
Serial.flush();
delay(1000);
}

WiFi.mode(WIFI_STA);
WiFiMulti.addAP("SSID", "PASSWORD");
}

void loop() {
// wait for WiFi connection
if ((WiFiMulti.run() == WL_CONNECTED)) {

WiFiClient client;
HTTPClient http;

Serial.printf("\n----- HTTP GET response -----\n");
if (http.begin(client, "http://httpbin.org/status/418")) { // HTTP

Serial.printf("[HTTP] GET...\n");
// start connection and send HTTP header
int httpCode = http.GET();

// httpCode will be negative on error
if (httpCode > 0) {
// HTTP header has been send and Server response header has been handled
Serial.printf("[HTTP] GET... code: %d\n", httpCode);

// file found at server
if (httpCode == HTTP_CODE_OK || httpCode == HTTP_CODE_MOVED_PERMANENTLY || httpCode == 418 /* I'm a teapot */) {
String payload = http.getString();
Serial.println(payload);
}
} else {
Serial.printf("[HTTP] GET... failed, error: %s\n", http.errorToString(httpCode).c_str());
}

http.end();
} else {
Serial.printf("[HTTP] Unable to connect\n");
}
}
else
{
Serial.print(".");
delay(500);
}

delay(1000);
}
```
I also tried to NTP on esp-wroom-02_EVM, the following commands worked as expected while UDP socket for DNS didn't work.
```
AT+CIPSTART="UDP","us.pool.ntp.org",123
AT+CIPSEND=48
1B 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00(hex string)
```

michalpasztamobica, 2020-01-07:
Ad 1) (closing when opening)
@zhiyong80 , I see your point, but I can't agree with the solution. You are right that we may fail to close the connection with a `close()` call, because the device is busy. But this was also the case before our current changes to `send()`. I think this exactly why [this check was added](https://github.com/ARMmbed/mbed-os/blob/244422d470bbd0444eceacaae5b0d4d924e97449/components/wifi/esp8266-driver/ESP8266/ESP8266.cpp#L570). If the open() call returns an `ERROR_PARAMETER` it means user is trying to reopen an open socket (or has passed an incorrect id or address). User should therefore try to close the socket and then open it again (if he is sure he wants to reuse this socket). What we could do is have `ESP8266::open_xxx()` return `NSAPI_ERROR_ALREADY` to indicate an open (or non-closed) socket. Would that make sense?
If we add the `close()` call to the `open_tcp/udp()` call, as you suggested, it is possible that user unknowingly closes one of his sockets by mistake, with absolutely no information about this from the driver. `open()` should open the socket, without silently closing it, in my opinion.
The examples of not checking for `close()` are coming from destructors. Users should better `close()` sockets conciously if they suspect that `close()` may fail.
I am also afraid that adding a `close()` on every open will slow down the opening procedure (bear in mind: `close()` unconditionally sends the AT command over serial).

Ad 2) From what I see [your code also returns DEVICE_ERROR](https://github.com/ccli8/mbed-os/commit/9e6503d6740ab12721ad8f8ae0d655ccbd239d01#diff-7d97bec406177b07ad47337a8b01fcedR726). If the return type is an issue, let's change it to NO_SOCKET, but I would insist that keeping status per-socket does not make sense here, exactly because we want to have one-send policy (as opposed to send-per-socket policy).

Ad 3) Double registration is not an issue, parser will just ignore it. Not receiving `SEND OK/SEND FAIL` or not deregistering the handlers is just as problematic as handling them for no reason. But the simplification of the send's flow control convinces me, we can keep the oobs registered all the time.

I am awaiting your feedback on 1 & 2 and I agree with 3 & 4.
If you aslo agree and revert 1 & 2 from your commit I could cherry pick it to my branch, if that's OK with you? 
Obviously, if you are not convinced I am open for further discussion.

zhiyong80, 2020-01-10:
The AT command works but I still get -3009 error code.
```
AT> AT+CIPDOMAIN="httpbin.org"
AT? +CIPDOMAIN:%*15[^
]
%n
AT= +CIPDOMAIN:34.193.212.251

AT? OK
%n
AT= OK

HttpRequest failed (error code -3009)
```

closed 2020-01-15 by 0xc0170