ARMmbed/mbed-os #11993: Some filesystem tests have a too long path name

jeromecoutant, 2019-12-02:
### Description of defect

Got issue with one filesystem test:
features/storage/filesystem/littlefs/TESTS/filesystem_recovery/resilience_functional/main.cpp

Path seems to be too long :-(

WindowsError: [Error 206] The filename or extension is too long: 'BUILD\\tests\\DISCO_L475VG_IOT01A\\ARMC6\\features\\storage\\filesystem\\littlefs\\TESTS\\filesystem_recovery\\resilience_functional\\features\\storage\\filesystem\\littlefs\\TESTS\\filesystem_recovery\\resilience_functional'

WindowsError: [Error 206] The filename or extension is too long: 'BUILD\\tests\\DISCO_L475VG_IOT01A\\GCC_ARM\\features\\storage\\filesystem\\littlefs\\TESTS\\filesystem_recovery\\resilience_functional\\features\\storage\\filesystem\\littlefs\\TESTS\\filesystem_recovery\\resilience_functional'

WindowsError: [Error 206] The filename or extension is too long: 'BUILD\\tests\\DISCO_L475VG_IOT01A\\IAR\\features\\storage\\filesystem\\littlefs\\TESTS\\filesystem_recovery\\resilience_functional\\features\\storage\\filesystem\\littlefs\\TESTS\\filesystem_recovery\\resilience_functional'

#### Target(s) affected by this defect ?

#### Toolchain(s) (name and version) displaying this defect ?

All

#### What version of Mbed-os are you using (tag or sha) ?

5.14.2

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

#### How is this defect reproduced ?

0xc0170, 2019-12-02:
I hope 6.0 will fix tools to get this paths no issue.

cc @ARMmbed/mbed-os-storage FYI

0xc0170, 2019-12-03:
@ARMmbed/mbed-os-storage Is there anything to help or rather wait for:

> I hope 6.0 will fix tools to get this paths no issue.

?

SeppoTakalo, 2019-12-03:
This is not storage related.
Should be fixed by @ARMmbed/mbed-os-tools

0xc0170, 2019-12-03:
> This is not storage related.
Should be fixed by @ARMmbed/mbed-os-tools

Correct, tools will be fixed in 6.0, this is known issue so if it is just in tools we will close this as duplicate. Based on the answer, I lean towards having this as duplicate.

@jeromecoutant 6.0 will bring a tools update. Until then this is known issue, older issue #7129 (contains references to others)

closed 2019-12-03 by 0xc0170

jeromecoutant, 2020-07-03:
Hi @0xc0170 

Back on this old closed issue :-)
 
> > This is not storage related.
> > Should be fixed by @ARMmbed/mbed-os-tools

Yes, but maybe, as a short term solution, test could be moved to another directory ?
- from features/storage/filesystem/littlefs/TESTS/filesystem_recovery/resilience_functional/main.cpp
- to TESTS/filesystem_recovery/resilience_functional/main.cpp ?

> Correct, tools will be fixed in 6.0, this is known issue so if it is just in tools we will close this as duplicate. Based on the answer, I lean towards having this as duplicate.

...

@JeanMarcR