ARMmbed/mbed-os #11994: QSPIF: issue with is_qpi_mode ?

jeromecoutant, 2019-12-02:
### Description of defect

Hi

I am currently debugging som QSPIF issue, and I could see one issue reviewing the code:

https://github.com/ARMmbed/mbed-os/blob/48f544f9e44a52ea8a628bffd06814d8343cc3a6/components/storage/blockdevice/COMPONENT_QSPIF/QSPIFBlockDevice.cpp#L956-L957

is_qpi_mode will be always FALSE...
It can be set to TRUE line 972, but set immediatly back to FALSE line 981

Maybe a break is missing after line 978 ?

Regards

#### Target(s) affected by this defect ?

#### Toolchain(s) (name and version) displaying this defect ?

#### What version of Mbed-os are you using (tag or sha) ?

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

#### How is this defect reproduced ?

0xc0170, 2019-12-02:
cc @ARMmbed/mbed-os-storage

jeromecoutant, 2019-12-09:
@ARMmbed/mbed-os-storage

Thx

michalpasztamobica, 2020-01-17:
Hi @jeromecoutant , thanks for pointing this out.
It seems that this behavior was intentionally introduced in https://github.com/ARMmbed/mbed-os/pull/9057, by @offirko . You were one of the approving reviewers of that PR. Has something changed since that time (over a year) and do you think it now makes sense to support the 4-4-4 version or do you still want the 1-4-4 version to be favoured?

jeromecoutant, 2020-01-17:
OK...

@kyle-cypress

kyle-cypress, 2020-01-21:
I don't have a strong opinion on whether QPI is enabled (compatibility with some controller hardware aside, #9057 makes a reasonable case that there is questionable benefit). 

However, the way the change was implemented does make it appear that the current behavior is the result of a typo rather than intentional. 

Should the entire block of code related to QSPIF_BASIC_PARAM_TABLE_QPI_READ_SUPPORT_BYTE be removed to make the intent more clear?

michalpasztamobica, 2020-01-22:
Hi @kyle-cypress , how about we move the 1-4-4 option "up" in front of the 4-4-4. It has a break clause, so if it is found it will be used. Then we can move the unwanted 4-4-4 "down", but still let it be used in case other options were not found.

The question I find hard to answer is - where exactly should the 4-4-4 option land: right after 1-4-4? at the very bottom, right before the 1-1-1? Somewhere in the middle? @kyle-cypress , @jeromecoutant , your opinion would be much appreciated.

kyle-cypress, 2020-01-22:
My vote would be for either after the 1-4-4 or the 1-1-4. Operating on the premise that the QPI setup overhead may or may not be higher than overhead of sending the address in single mode (hence ambivalence about whether it comes before or after 1-1-4), but QPI will probably be faster than dropping to dual or single mode for data transfer.

closed 2020-01-30 by kjbracey-arm