ARMmbed/mbed-os #12000: build issue with cy8cproto_062_4343w (ARMC5, Mbed OS 5.14)

acabarbaye, 2019-12-02:
### Description of defect

Following build error appears when compiling blinky example for the following target:

> 
> "./mbed-os/features/netsocket/emac-drivers/TARGET_Cypress/COMPONENT_WHD/interface/WhdAccessPoint.cpp", line 25: Error:  #20: identifier "malloc" is undefined
> "./mbed-os/features/netsocket/emac-drivers/TARGET_Cypress/COMPONENT_WHD/interface/WhdAccessPoint.cpp", line 63: Error:  #20: identifier "free" is undefined
> ./mbed-os/features/netsocket/emac-drivers/TARGET_Cypress/COMPONENT_WHD/interface/WhdAccessPoint.cpp: 0 warnings, 2 errors

#### Target(s) affected by this defect ?
cy8cproto_062_4343w

#### Toolchain(s) (name and version) displaying this defect ?
Armc5

#### What version of Mbed-os are you using (tag or sha) ?
mbed-os-5.14.2

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
latest

#### How is this defect reproduced ? 
`mbed compile` of mbed os blinky

0xc0170, 2019-12-03:
@ARMmbed/team-cypress Please review

MarceloSalazar, 2020-04-14:
Arm Compiler 5 is not supported
https://os.mbed.com/blog/entry/Removal-of-ARM-Compiler-5-and-uARM-toolc/

closed 2020-04-14 by MarceloSalazar