ARMmbed/mbed-os #12001: Build issue with cy8ckit_062 (Armc5, Mbed os 5.14)

acabarbaye, 2019-12-02:
### Description of defect

Following build error appears when compiling blinky example for the following target:

> "./mbed-os/platform/cxxsupport/mstd_type_traits", line 1034: Error:  #40: expected an identifier
> "./mbed-os/platform/cxxsupport/mstd_type_traits", line 1034: Error:  #40: expected an identifier
> "./mbed-os/platform/cxxsupport/mstd_type_traits", line 1035: Error:  #79: expected a type specifier
> "./mbed-os/platform/cxxsupport/mstd_type_traits", line 1035: Error:  #79: expected a type specifier
> "./mbed-os/platform/cxxsupport/mstd_type_traits", line 1037: Error:  #40: expected an identifier
> "./mbed-os/platform/cxxsupport/mstd_type_traits", line 1037: Error:  #40: expected an identifier
> "./mbed-os/platform/cxxsupport/mstd_type_traits", line 1038: Error:  #79: expected a type specifier
> "./mbed-os/platform/cxxsupport/mstd_type_traits", line 1038: Error:  #79: expected a type specifier
> "./mbed-os/platform/cxxsupport/mstd_type_traits", line 1037: Warning:  #842-D: constant "<unnamed>"  is not used in or cannot be deduced from the template argument list of class template "std::impl::ct2<<error-type>, <error-type>, <error-constant>, <error-constant>>" 
> "./mbed-os/platform/cxxsupport/mstd_type_traits", line 1037: Warning:  #842-D: constant "<unnamed>"  is not used in or cannot be deduced from the template argument list of class template "std::impl::ct2<<error-type>, <error-type>, <error-constant>, <error-constant>>" 
> "./mbed-os/platform/cxxsupport/mstd_type_traits", line 1038: Error:  #79: expected a type specifier
> "./mbed-os/platform/cxxsupport/mstd_type_traits", line 1038: Error:  #79: expected a type specifier
> "./mbed-os/platform/cxxsupport/mstd_type_traits", line 1053: Error:  #442: too few arguments for class template "std::impl::ct2" 
> "./mbed-os/platform/cxxsupport/mstd_type_traits", line 1053: Error:  #262: not a class or struct name
> ./mbed-os/components/802.15.4_RF/atmel-rf-driver/source/at24mac.cpp: 2 warnings, 12 errors

#### Target(s) affected by this defect ?
cy8ckit_062_ble
cy8ckit_062_wifi_bt

#### Toolchain(s) (name and version) displaying this defect ?
Armc5

#### What version of Mbed-os are you using (tag or sha) ?
mbed-os-5.14.2

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
latest

#### How is this defect reproduced ? 
`mbed compile` of mbed os blinky

0xc0170, 2019-12-03:
@kjbracey-arm Please review

0xc0170, 2019-12-03:
cc @ARMmbed/team-cypress

kjbracey-arm, 2019-12-03:
I think this has come up before. Odd `#define`s of pin names like `D1` can cause problems. We prefer them to be enums. If you are going to have them be defines, at least make them be "plain" so they can serve as identifiers in places like this. That's how this is working on most platforms.

Failure here is `D1` is defined as `CYBSP_D1` and `CYBSP_D1` is defined as `(P5_1)`. `P5_1` is then an enum. Remove the `()`, and it would be fine.

Certainly `mstd_type_traits` could be more defensive, but platforms are dumping their entire stuff into application visibility, so they should take case to limit the effect of the exposure.

I'm going to request that Cypress remove the `()` at a minimum. To help avoid future problems, having a mechanism to include the `D1 = P5_1` type aliases inside the `PinNames` enum would be best.

0xc0170, 2019-12-03:
> I think this has come up before. Odd #defines of pin names like D1 can cause problems. We prefer them to be enums. If you are going to have them be defines, at least make them be "plain" so they can serve as identifiers in places like this. That's how this is working on most platforms.

I recall similar issue after reading this as well ! 

@ARMmbed/team-cypress Please review this issue

0xc0170, 2019-12-09:
> @ARMmbed/team-cypress Please review this issue

Can cy8ckit_062 be fixed not to cause this error?

morser499, 2019-12-09:
Yes, this is something that can, and will be, fixed.  We are currently discussing what the best way to fix it is as we would like to do away with the #defines since they are known to be problematic.

MarceloSalazar, 2020-04-14:
Arm Compiler 5 is not supported
https://os.mbed.com/blog/entry/Removal-of-ARM-Compiler-5-and-uARM-toolc/

closed 2020-04-14 by MarceloSalazar