ARMmbed/mbed-os #12003: Build error with MTB_MXCHIP_EMW3166 (Armc5, Mbed os 5.14)

acabarbaye, 2019-12-02:
### Description of defect

Following build error appears when compiling blinky example for the following target:

> Link: /home/monty-bot/toolchains/ARMC5/5.06u6/bin/armlink --via ./BUILD/MTB_MXCHIP_EMW3166/ARMC5/.link_options.txt
> [DEBUG] Return: 1
> [DEBUG] Errors: Error: L6218E: Undefined symbol WICED_EMAC::get_instance() (referred from BUILD/MTB_MXCHIP_EMW3166/ARMC5/mbed-os/targets/TARGET_WICED/wiced_interface/default_wifi_interface.o).
> [DEBUG] Errors: Error: L6218E: Undefined symbol WicedInterface::WicedInterface(EMAC&, OnboardNetworkStack&) (referred from BUILD/MTB_MXCHIP_EMW3166/ARMC5/mbed-os/targets/TARGET_WICED/wiced_interface/default_wifi_interface.o).
> [DEBUG] Errors: Error: L6218E: Undefined symbol VTT for WicedInterface (referred from BUILD/MTB_MXCHIP_EMW3166/ARMC5/mbed-os/targets/TARGET_WICED/wiced_interface/default_wifi_interface.o).

#### Target(s) affected by this defect ?
MTB_MXCHIP_EMW3166 

#### Toolchain(s) (name and version) displaying this defect ?
Armc5

#### What version of Mbed-os are you using (tag or sha) ?
mbed-os-5.14.2

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
latest

#### How is this defect reproduced ? 
`mbed compile` of mbed os blinky

0xc0170, 2019-12-03:
Similar to #12004 (binary incompatibility when updating from ARMC5 to ARMC6).

MarceloSalazar, 2020-04-14:
Arm Compiler 5 is not supported
https://os.mbed.com/blog/entry/Removal-of-ARM-Compiler-5-and-uARM-toolc/

closed 2020-04-14 by MarceloSalazar