ARMmbed/mbed-os #12004: Build error with ublox odin w2 (Armc5, Mbed os 5.14)

acabarbaye, 2019-12-02:
### Description of defect

Following build error appears when compiling blinky example for the following target:

> 
> Errors: Warning: L6418W: Tagging symbol __tagsym$$used.3 defined in ./mbed-os/targets/TARGET_STM/TARGET_STM32F4/TARGET_STM32F439xI/TARGET_MODULE_UBLOX_ODIN_W2/sdk/TOOLCHAIN_ARM/libublox-odin-w2-driver.ar(cb_version.o)(.rodata.cbWLAN_versionStringGit) is not recognized.
> [DEBUG] Errors: Error: L6200E: Symbol rtos::Thread::Thread__sub_object(osPriority_t, unsigned, unsigned char*, const char*) multiply defined (by ./mbed-os/targets/TARGET_STM/TARGET_STM32F4/TARGET_STM32F439xI/TARGET_MODULE_UBLOX_ODIN_W2/sdk/TOOLCHAIN_ARM/libublox-odin-w2-driver.ar(cb_main.o) and BUILD/MTB_UBLOX_ODIN_W2/ARMC5/mbed-os/drivers/source/usb/USBCDC_ECM.o).
> [DEBUG] Errors: Not enough information to list the image map.
> Error: L6200E: Symbol rtos::Thread::Thread__sub_object(osPriority_t, unsigned, unsigned char*, const char*) multiply defined (by ./mbed-os/targets/TARGET_STM/TARGET_STM32F4/TARGET_STM32F439xI/TARGET_MODULE_UBLOX_ODIN_W2/sdk/TOOLCHAIN_ARM/libublox-odin-w2-driver.ar(cb_main.o) and BUILD/UBLOX_EVK_ODIN_W2/ARMC5/mbed-os/drivers/source/usb/USBCDC_ECM.o).
> Not enough information to list the image map.

#### Target(s) affected by this defect ?
ublox_evk_odin_w2
MTB_UBLOX_ODIN_W2

#### Toolchain(s) (name and version) displaying this defect ?
Armc5

#### What version of Mbed-os are you using (tag or sha) ?
mbed-os-5.14.2

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
latest

#### How is this defect reproduced ? 
`mbed compile` of mbed os blinky

0xc0170, 2019-12-03:
This is one of the cases when we update the compiler version (for ARM in this case), there is a binary breakage, like this one. 

We do not have `TOOLCHAIN_ARM` for ARMC5 , do we ? cc @ARMmbed/mbed-os-tools

0xc0170, 2019-12-09:
@bulislaw How shall we handle this one? This is  "known-issue" for targets with binaries in this tree, won't be able to support ARMCC5

bulislaw, 2019-12-10:
We'll add a overal note in the know issues for 5.15 that ARMC5 may not work.

MarceloSalazar, 2020-04-08:
Issue documented on ODIN W2 platform page here:
https://os.mbed.com/platforms/ublox-EVK-ODIN-W2/#known-limitations

closed 2020-04-08 by MarceloSalazar