ARMmbed/mbed-os #12005: Build error with NUCLEO_H743ZI2 (Armc5, Mbed os 5.14)

acabarbaye, 2019-12-02:
### Description of defect

Following build error appears when compiling blinky example for the following target:

> ./mbed-os/features/netsocket/emac-drivers/TARGET_STM/stm32xx_emac.cpp", line 89: Error:  #20: identifier "ETH_RX_BUFFER_SIZE" is undefined

#### Target(s) affected by this defect ?
NUCLEO_H743ZI2

#### Toolchain(s) (name and version) displaying this defect ?
Armc5

#### What version of Mbed-os are you using (tag or sha) ?
mbed-os-5.14.2

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
latest

#### How is this defect reproduced ? 
`mbed compile` of mbed os blinky

jeromecoutant, 2019-12-03:
Hi

I confirm that ETH_RX_BUFFER_SIZE is not defined :-(

Why issue has not been found before?
Seems it is because used toolchain is Armc5, which is no more tested in CI.

Loooking at features/netsocket/emac-drivers/TARGET_STM/stm32xx_emac.cpp
We got:

#if defined ( __ICCARM__ ) /*!< IAR Compiler */
...
#elif defined ( __CC_ARM )  /* MDK ARM Compiler */
...
__attribute__((section(".RxArraySection"))) uint8_t Rx_Buff[ETH_RX_DESC_CNT][ETH_RX_BUFFER_SIZE] 
...
#elif defined ( __GNUC__ ) /* GNU Compiler */ 
...
#endif

Seems that ARMC6 doesn't set __CC_ARM and set __GNUC__ ?

@kjbracey-arm any sugestion ?

Should we implement:

#if defined ( __ICCARM__ ) /*!< IAR Compiler */
#else
#endif

or

#if defined ( TOOLCHAIN_IAR ) /*!< IAR Compiler */
#elif defined ( TOOLCHAIN_ARM )  /* MDK ARM Compiler */
#elif defined ( TOOLCHAIN_GCC_ARM ) /* GNU Compiler */ 
#endif

0xc0170, 2019-12-03:
>Seems it is because used toolchain is Armc5, which is no more tested in CI.

It's been active in the recent nightlies thus the report.

> Seems that ARMC6 doesn't set __CC_ARM and set GNUC ?

That is correct, `__CC_ARM` is ARMC5. Use that one if you want to have it only used by ARMC5 (or negative for skipping ARMC5).

kjbracey-arm, 2019-12-03:
You can use `__CC_ARM` to check specifically for the ARM (formerly Norcroft) compiler core. So ARMC5, but not ARMC6.

You can use ` __clang__` to look for Clang-based compilers, including ARMC6.

If you want to look for either ARMC5 or ARMC6, then you can test `__ARMCC_VERSION`. That's what you want to do here, I think - your attribute syntax works on both compilers, and you're sharing the linker and linker map.

I'd rarely recommend doing a `>=6000000` test on  `__ARMCC_VERSION` though - if you're doing that you probably really mean "which compiler family is this", for which the above two tests are more appropriate, IMO.

ARMC5, ARMC6 and GCC all set `__GNUC__`, helpfully(!), so you need to take care on that.

Yes, there are the Mbed-specific `TOOLCHAIN_xxx` defines, but I've not really used them myself.

PR #12015 (2019-12-05): STM EMAC: compilation issue with ARMC5

### Description 
#### Summary of change 

STM32H7 boards couldn't compile with ARMC5

Fix #12005 

#### Documentation 

----------------------------------------------------------------------------------------------------------------
### Pull request type 

    [x] Patch update (Bug fix / Target update / Docs update / Test update / Refactor)
    [] Feature update (New feature / Functionality change / New API)
    [] Major update (Breaking change E.g. Return code change / API behaviour change)

-------------------------------------------------------------- [... cut]
closed 2019-12-05 by 0xc0170