ARMmbed/mbed-os #12009: Custom target PinNames

pilotak, 2019-12-03:
### Description of defect
I have a custom target boards which inherit from dev boards (NUCLEO_F412ZG and NUCLEO_F303RE) with a custom pin names specified in `PinNames.h` which is working fine because it overrides `"mbed-os\targets\TARGET_STM\TARGET_STM32F4\TARGET_STM32F412xG\TARGET_NUCLEO_F412ZG\PinNames.h` and my definition is used instead.

When i do the same with NUCLEO_F091RC it doesn't work, `PinNames.h` is included from `mbed-os\targets\TARGET_STM\TARGET_STM32F0\TARGET_NUCLEO_F091RC` not from `TARGET_TEST123` under root of project like with F412 or F303. The only workaround is to delete `PinNames.h` under `mbed-os` dir.

File tree:
```
|   main.cpp
|   custom_targets.json
+---TARGET_TEST123
|       device.h
|       PinNames.h
|       reserved_pins.h
```

custom_targets.json
```json
{
    "TEST123": {
        "inherits": ["NUCLEO_F091RC"],
        "supported_form_factors": [],
        "overrides": {
            "clock_source": "USE_PLL_HSE_XTAL"
        },
        "components_add": ["I2CEE"]
    }
}
```

#### Target(s) affected by this defect ?
NUCLEO_F091RC

#### Toolchain(s) (name and version) displaying this defect ?
GCC_ARM

#### What version of Mbed-os are you using (tag or sha) ?
a8ee2d8c7db578132e26b15e6115ad1fc78b9182

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
Python 3.7.4
mbed-cli 1.10.1

#### How is this defect reproduced ? 
When creating custom target which inherits from NUCLEO_F091RC

0xc0170, 2019-12-03:
cc @ARMmbed/team-st-mcd Please review

jeromecoutant, 2019-12-03:
Hi

Do you mean that we need a create an intermediate directory ?

Like this: https://github.com/jeromecoutant/mbed/tree/PR_F091 ?

pilotak, 2019-12-03:
No, i'm unable to override `PinNames.h` located in mbed with the one located in "TARGET_TEST123" when inheriting from NUCLEO_F091RC but it works with F412 and F303

jeromecoutant, 2019-12-03:
Maybe you should propose your need in a PR

Thx

pilotak, 2019-12-03:
that's a problem, because i don't know how to fix this on mbed side, other than delete the file. I will create a repo where it should be clearer what the problem is.

pilotak, 2019-12-03:
@jeromecoutant https://github.com/pilotak/mbed-custom-target

```sh
mbed import https://github.com/pilotak/mbed-custom-target
cd mbed-custom-target
mbed compile -m TESTF091 -t GCC_ARM
```

you will see an error 
```
[Error] main.cpp@3,17: 'TEST_PIN' was not declared in this scope
[ERROR] .\main.cpp:3:17: error: 'TEST_PIN' was not declared in this scope
 DigitalOut test(TEST_PIN);
```

but when you compile with second target `TEST_PIN` is defined
```
mbed compile -m TESTF412 -t GCC_ARM
```

PR #12029 (2019-12-05): STM32F0: create STM32F091xC target

### Summary of changes 

In order to ease custom targets with STMF091xC MCU,
a intermediate directory is created in TARGET_STM32F0

Fix #12009 

@pilotak 

#### Impact of changes 

#### Migration actions required 

### Documentation 

--------
closed 2019-12-05 by 0xc0170