ARMmbed/mbed-os #12016: [BLE][Cordio] Manufacturer Specific Data Truncated to 8 Bytes

AGlass0fMilk, 2019-12-03:
### Description of defect

I am trying to do beacon-style advertising using the nRF52840 and Cordio BLE stack. When I try to set a Manufacturer Specific payload entry longer than 8 bytes, it either gets truncated or disappears entirely from the advertising data.

I am essentially using the [BLE_GattServer](https://github.com/ARMmbed/mbed-os-example-ble/tree/c2931924ab830f99cbcefa4d5ce281a899c7fc9b/BLE_GattServer) example code.

I am adding no other fields to the advertisement payload (31 bytes) and the buffer is not overflowing.

Is this a configuration parameter in Cordio or something?

#### Target(s) affected by this defect ?

NRF52840_DK

#### Toolchain(s) (name and version) displaying this defect ?
arm-none-eabi-gcc (GNU Tools for ARM Embedded Processors 6-2017-q2-update) 6.3.1 20170620 (release) [ARM/embedded-6-branch revision 249437]

#### What version of Mbed-os are you using (tag or sha) ?

cf4f12a123c05fcae83fc56d76442015cb8a39e9

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

mbed-cli 1.8.2

#### How is this defect reproduced ? 

Attempt to set manufacturer specific data field in advertising payload longer than 8 bytes.

0xc0170, 2019-12-03:
cc @ARMmbed/mbed-os-pan

paul-szczepanek-arm, 2019-12-03:
Can you print the payload before and after you add the data to the advertising payload builder and check the return code of setManufacturerSpecificData?

paul-szczepanek-arm, 2019-12-03:
And can you check the buffer size of the buffer you pass to AdvertisingDataBuilder when creating it?

AGlass0fMilk, 2019-12-03:
My mistake... it appears iOS was filtering the data from the apps I use to scan.

It's strange that the desktop nRF Connect also failed to show the manufacturer data. I'm not sure what fixed it, but it's working now.

Sorry for crying wolf @paul-szczepanek-arm

closed 2019-12-03 by AGlass0fMilk