ARMmbed/mbed-os #12021: Problems exporting to keil uvision

alejandroaps, 2019-12-04:
I have a NXP LPC1768 and I've exported a project from mbed online compiler to keil uvision but I can't build becuase there are several erros:
```

Rebuild started: Project: ___USBMIDI_HelloWorld
*** Using Compiler 'V6.12', folder: 'C:\Keil_v5\ARM\ARMCLANG\Bin'
Rebuild target '___USBMIDI_HelloWorld'
error: invalid value '' in '-std='
note: use 'c++98' or 'c++03' for 'ISO C++ 1998 with amendments' standard
note: use 'gnu++98' or 'gnu++03' for 'ISO C++ 1998 with amendments and GNU extensions' standard
note: use 'c++11' for 'ISO C++ 2011 with amendments' standard
note: use 'gnu++11' for 'ISO C++ 2011 with amendments and GNU extensions' standard
note: use 'c++14' for 'ISO C++ 2014 with amendments' standard
note: use 'gnu++14' for 'ISO C++ 2014 with amendments and GNU extensions' standard
note: use 'c++17' for 'ISO C++ 2017 with amendments' standard
note: use 'gnu++17' for 'ISO C++ 2017 with amendments and GNU extensions' standard
note: use 'c++2a' for 'Working draft for ISO C++ 2020' standard
note: use 'gnu++2a' for 'Working draft for ISO C++ 2020 with GNU extensions' standard
compiling main.cpp...
error: invalid value '' in '-std='
note: use 'c++98' or 'c++03' for 'ISO C++ 1998 with amendments' standard
note: use 'gnu++98' or 'gnu++03' for 'ISO C++ 1998 with amendments and GNU extensions' standard
note: use 'c++11' for 'ISO C++ 2011 with amendments' standard
note: use 'gnu++11' for 'ISO C++ 2011 with amendments and GNU extensions' standard
note: use 'c++14' for 'ISO C++ 2014 with amendments' standard
note: use 'gnu++14' for 'ISO C++ 2014 with amendments and GNU extensions' standard
note: use 'c++17' for 'ISO C++ 2017 with amendments' standard
note: use 'gnu++17' for 'ISO C++ 2017 with amendments and GNU extensions' standard
note: use 'c++2a' for 'Working draft for ISO C++ 2020' standard
note: use 'gnu++2a' for 'Working draft for ISO C++ 2020 with GNU extensions' standard
compiling USBMIDI.cpp...
USBMIDI/usbdevice.c(23): warning: In file included from...
USBMIDI/usbcore.h(26): warning: In file included from...
mbed\mbed.h(22): warning: In file included from...
mbed/DigitalIn.h(12): warning: In file included from...
mbed/Base.h(12): error: 'cstdlib' file not found
#include <cstdlib>
         ^~~~~~~~~
1 error generated.
compiling usbdevice.c...
USBMIDI/usbcore.c(23): warning: In file included from...
USBMIDI/usbcore.h(26): warning: In file included from...
mbed\mbed.h(22): warning: In file included from...
mbed/DigitalIn.h(12): warning: In file included from...
mbed/Base.h(12): error: 'cstdlib' file not found
#include <cstdlib>
         ^~~~~~~~~
1 error generated.
compiling usbcore.c...
".\BUILD\___USBMIDI_HelloWorld.axf" - 4 Error(s), 0 Warning(s).
Target not created.
Build Time Elapsed:  00:00:01

```

Thanks
Regards

0xc0170, 2019-12-04:
Please use the template for reporting the bug. It provides sections that are very helpful. Without these details, we won't be able to help you.

What uvision do you have locally, what code are you cloning (version of Mbed OS) ?

0xc0170, 2019-12-05:
I'll close this , please reopen with an update.

closed 2019-12-05 by 0xc0170

alejandroaps, 2019-12-06:
> 
> 
> I'll close this meanwhile, please reopen with an update.

Sorry I have been a little busy and I didn't know about the template. I updated with the datea that you told me, Is the template on any website?

alejandroaps, 2019-12-10:
Can you reopen? Or how can I reopen?