ARMmbed/mbed-os #12042: All netsocket tests fail on EP_AGORA with IPv6 disabled

linlingao, 2019-12-05:
### Description of defect
These tests fail on EP_AGORA with IPv6 disabled. They pass with IPv6 enabled.

Test | Branch/Tag | lwip.ipv6-enabled | Result | Comments
-- | -- | -- | -- | --

tests-netsocket-tcp | master | TRUE | Pass |  
-- | -- | -- | -- | --
tests-netsocket-tcp | master | FALSE | Fail | TIMEOUT
tests-netsocket-tcp | mbed-os-5.14.2 | TRUE | Pass |  
tests-netsocket-tcp | mbed-os-5.14.2 | FALSE | Fail | TIMEOUT
tests-netsocket-udp | master | TRUE | Pass |  
tests-netsocket-udp | master | FALSE | Fail | TIMEOUT
tests-netsocket-udp | mbed-os-5.14.2 | TRUE | Pass |  
tests-netsocket-udp | mbed-os-5.14.2 | FALSE | Fail | TIMEOUT
tests-netsocket-dns | master | TRUE | Pass |  
tests-netsocket-dns | master | FALSE | Fail | TIMEOUT
tests-netsocket-dns | mbed-os-5.14.2 | TRUE | Pass |  
tests-netsocket-dns | mbed-os-5.14.2 | FALSE | Fail | TIMEOUT
tests-netsocket-tls | master | TRUE | Pass |  
tests-netsocket-tls | master | FALSE | Fail | TIMEOUT
tests-netsocket-tls | mbed-os-5.14.2 | TRUE | Pass |  
tests-netsocket-tls | mbed-os-5.14.2 | FALSE | Fail | TIMEOUT

#### Target(s) affected by this defect ?
EP_AGORA

#### Toolchain(s) (name and version) displaying this defect ?
GCC_ARM

#### What version of Mbed-os are you using (tag or sha) ?
5.14.2 and master
#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
N/A

#### How is this defect reproduced ? 
mbed test -m ep_agora -t gcc_arm -n mbed-os-tests-netsocket-*

The json file is attached.
[mbed_app.json.zip](https://github.com/ARMmbed/mbed-os/files/3928927/mbed_app.json.zip)

linlingao, 2019-12-05:
@ARMmbed/team-embeddedplanet @maclobdell

0xc0170, 2020-02-24:
> @ARMmbed/team-embeddedplanet @maclobdell

Please review

loverdeg, 2020-11-22:
Is this still valid?

adbridge, 2021-06-07:
As there has been no response on this for nearly 8 months I am going to close it. It can be re-opened if it is still an issue.

closed 2021-06-07 by adbridge