ARMmbed/mbed-os #12049: NXP - support for MIMXRT1050-EVKB

JanneKiiskila, 2019-12-09:
### Description of defect

Could NXP add support for MIMXRT1050-EVKB, because it seems MIMXRT1050-EVK is not available or vendors are not stocking it. I.e.
- [digikey](https://www.digikey.fi/product-detail/en/nxp-usa-inc/MIMXRT1050-EVK/568-13517-ND/7646293?utm_adgroup=General&utm_source=google&utm_medium=cpc&utm_campaign=Dynamic%20Search_FI_Product&utm_term=&mkwid=s08a401Mf&pcrid=399610958126&pkw=&pmt=b&pdv=c&productid=&slid=&gclid=Cj0KCQiA_rfvBRCPARIsANlV66OGNSG2t6_0sYs77IK7HmqncMoFgtULfVYZj6aK2BV9SazABqLnJ34aAtuPEALw_wcB) - states it's discontinued
- [mouser.fi](https://www.mouser.co.uk/ProductDetail/NXP-Semiconductors/MIMXRT1050-EVK?qs=sGAEpiMZZMtw0nEwywcFgJjuZv55GFNmTyRZx6s3qIUZD31ixq11dA%3D%3D) - non-stocked.

Both have the EVKB available in stock.

Attn: @ARMmbed/team-nxp 

#### Target(s) affected by this defect ?

#### Toolchain(s) (name and version) displaying this defect ?
N/A

#### What version of Mbed-os are you using (tag or sha) ?

master

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

N/A.

#### How is this defect reproduced ?

mmahadevan108, 2019-12-09:
The current support is for EVKB

0xc0170, 2019-12-09:
@mmahadevan108 To understand, the EVKB is supported , it was just a target name confusion ?

mmahadevan108, 2019-12-09:
Thats correct.

JanneKiiskila, 2019-12-09:
So, fix is then easy - either rename the current existing target or create a secondary one that has the EVKB-name and just inherits the existing one?

https://github.com/ARMmbed/mbed-os/blob/mbed-os-5.15/targets/targets.json#L1900

Just out of personal curiosity, what does the `inherits: "Target"` actually do? I don't understand what `Target` it inherits, but then again without that line the target fails to compile completely.

0xc0170, 2019-12-10:
> Just out of personal curiosity, what does the inherits: "Target" actually do? I don't understand what Target it inherits, but then again without that line the target fails to compile completely.

Get the defaults.

@mmahadevan108 should we fix the name - create a new one that just inherits ?

JanneKiiskila, 2020-01-09:
@mmahadevan108  - any takes on this?

mmahadevan108, 2020-01-09:
My preference would be to leave this unchanged as EVKB board is the default EVK. 

Changing name here would also require changing the mbedls database,

closed 2020-01-10 by JanneKiiskila