ARMmbed/mbed-os #12054: arm math:

0xc0170, 2019-12-09:
### Description of defect

File `arm_math.h` is part of this repository. It was added `106f34e2edd89dc385bd501d8d180d2f53d51e87` but I believe it was in the tree longer as we used to have DSP library part of Mbed 2.  I recall this header contained some implementation we used (CLZ for example) but that was removed in this header. We should not need this header file.

It causes a problems, as CMSIS DSP is a component (providing implementation), not any API we could overwrite. We should review this header file, and possibly remove it (it is outdated, our import script ignores it). Plus it causes a conflicts, for applications that use DSP, our outdated header file gets in and causes errors.

Ideally, DSP gets in via separate module as we do not provide this functionality in Mbed OS.

#### Target(s) affected by this defect ?

Cortex-M targets (all)

#### Toolchain(s) (name and version) displaying this defect ?

All, generic file

#### What version of Mbed-os are you using (tag or sha) ?

5.14.0

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

Not relevant

#### How is this defect reproduced ? 

Not relevant

cc @ARMmbed/mbed-os-core

kjbracey-arm, 2019-12-09:
We have had a small non-zero number of queries about why we don't include the CMSIS DSP/NN libraries.

Would be nice to bring them in as a feature/component/whatever to make people's life easy.

But certainly we shouldn't be supplying this old fragment that interferes with someone bringing them in themselves.

loverdeg-ep, 2019-12-09:
@AGlass0fMilk

AGlass0fMilk, 2019-12-09:
My solution up to now was to find which old version of CMSIS DSP that header was from and import the appropriate source 😅.

It's probably easier to just have people import CMSIS DSP/NN themselves (using `mbed add` or something). That way Mbed-OS doesn't have to maintain a copy/link to another repo. If they're using DSP/NN (fairly advanced) they will likely be familiar with adding a library.

What I would like to see is a C++ DSP framework for Mbed built with CMSIS DSP :)

loverdeg-ep, 2019-12-09:
yeah would be nice if CMSIS-packs were mature enough to be used for dependency management. Or if an official and well specified C/C++ dependency management tool were  selected to be used with Mbed...

0xc0170, 2019-12-10:
> It's probably easier to just have people import CMSIS DSP/NN themselves (using mbed add or something). That way Mbed-OS doesn't have to maintain a copy/link to another repo. If they're using DSP/NN (fairly advanced) they will likely be familiar with adding a library.

That is the plan, this header file was there from times when we had DSP in the tree. It causes issues now.

> yeah would be nice if CMSIS-packs were mature enough to be used for dependency management. Or if an official and well specified C/C++ dependency management tool were selected to be used with Mbed...

Wish the same 👍

PR #12055 (2020-01-08): cmsis: remove arm math

### Summary of changes 

Not needed file, should be part of DSP. The functionality is part of the library provided
by CMSIS.

I would like to test this, for all devices. This header file should be removed in 6.0.0 as we do not provide DSP library. Thus I set this as Major update.

Fixes #12054

#### Impact of changes 

Applications should use CMSIS DSP package
commit 5a985a7d (2020-01-08): cmsis: remove arm math

Fixes #12054

Not needed file, should be part of DSP. The functionality is part of the library provided
by CMSIS.
closed 2020-01-08 by 0xc0170