ARMmbed/mbed-os #12062: Mbed OS download failure

kpyush, 2019-12-10:
### Description of defect
The following download link is not working: https://developer.mbed.org/users/mbed_official/code/mbed/archive/0f02307a0877.zip

#### What version of Mbed-os are you using (tag or sha) ?
Mbed OS version: 5.12.0

#### How is this defect reproduced ? 
Follow the link: https://developer.mbed.org/users/mbed_official/code/mbed/archive/0f02307a0877.zip

0xc0170, 2019-12-10:
You are trying to download Mbed 2 release. You can download Mbed Os release on Github: https://github.com/ARMmbed/mbed-os/releases

Although the mbed should work, how did you get that archive link? I am trying now to download latest mbed (172 revision).

0xc0170, 2019-12-10:
https://os.mbed.com/users/mbed_official/code/mbed/export - works for me if I select a board and a toolchain.

kpyush, 2019-12-10:
@0xc0170 We are shipping this link information with a product, there it is failing. Should we migrate to new link?

0xc0170, 2019-12-10:
Let me verify that one with web team. Was it previously working? The link above is getting an archive for specific SHA of Mbed 2 correct?

kpyush, 2019-12-11:
Thanks @0xc0170, yes it was working previously.

kpyush, 2019-12-11:
Can you please give me the new link to download the same archive of Mbed 2 ?

kpyush, 2019-12-16:
@0xc0170 Any update ??

0xc0170, 2020-01-03:
It was internally triaged. The cause is not yet found but confirmed (there were other similar reports).

0xc0170, 2020-01-03:
Similar report to get mbed 2 lib here #12174.

micque01, 2020-01-07:
Hi @kpyush, the Mbed web team have fixed the issue, could you please try the https://developer.mbed.org/users/mbed_official/code/mbed/archive/0f02307a0877.zip link again?

closed 2020-01-10 by ciarmcom