ARMmbed/mbed-os #12066: STM32L4 AnalogIn not working after deepsleep

marisabele, 2019-12-10:
### Description of defect

Reading analog input on STM32L4 will return 0 value after the deep sleep.
Failing example:

```
#include "mbed.h"

Serial pc(PA_2, PA_3,115200);
AnalogIn bat(PC_1);
// main() runs in its own thread in the OS
int main() {
    while (true) {
        pc.printf("ADC1:%f\n",bat.read());
        ThisThread::sleep_for(5*1000);
    }
}
```

If deepsleep locked then ADC works correctly:
```
#include "mbed.h"

Serial pc(PA_2, PA_3,115200);

AnalogIn bat(PC_1);
// main() runs in its own thread in the OS
int main() {
    DeepSleepLock lock;
    while (true) {
        pc.printf("ADC1:%f\n",bat.read());
        ThisThread::sleep_for(5*1000);
    }
}
```

#### Target(s) affected by this defect ?
STM32L4 
#### Toolchain(s) (name and version) displaying this defect ?

#### What version of Mbed-os are you using (tag or sha) ?
mbed-os 6691:cf4f12a
#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
Online IDE

#### How is this defect reproduced ? 

```
#include "mbed.h"

Serial pc(PA_2, PA_3,115200);
AnalogIn bat(PC_1);
// main() runs in its own thread in the OS
int main() {
    while (true) {
        pc.printf("ADC1:%f\n",bat.read());
        ThisThread::sleep_for(5*1000);
    }
}
```

jeromecoutant, 2019-12-10:
Hi @marisabele 

I couldn't currently reproduce your issue...
Maybe something is missing ?
Which board are you using ?

marisabele, 2020-01-06:
This issue was specific to one hardware board. (Maybe ESD damage or something else)

closed 2020-01-06 by marisabele