ARMmbed/mbed-os #12072: Linker fails when using std::string and Softdevice

melvinvdb, 2019-12-10:
### Description of defect

When targeting NRF52_DK with the Softdevice enabled, it fails linking when using std::string.
We still require the use of a softdevice instead of Cordio because otherwise there is no support for FOTA.

`Link: BLE_Beacon
arm-none-eabi/bin/ld: .ARM.exidx has both ordered and unordered sections
arm-none-eabi/bin/ld: final link failed: Bad value
collect2: error: ld returned 1 exit status
[ERROR] arm-none-eabi/bin/ld: .ARM.exidx has both ordered and unordered sections
arm-none-eabi/bin/ld: final link failed: Bad value
collect2: error: ld returned 1 exit status`

#### Target(s) affected by this defect ?
NRF52_DK and maybe others

#### Toolchain(s) (name and version) displaying this defect ?
GCC_ARM

#### What version of Mbed-os are you using (tag or sha) ?
This happens from 5.13.0 up to the latest (5.14.2).

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
mbed-cli 1.8.3
mbed-os 5.14.2

#### How is this defect reproduced ? 
I forked mbed-os-example-ble and created a demo: 
https://github.com/melvinvdb/mbed-os-example-ble/commit/37e2e165f2b822d6bfa5cb29d51c0acebe3bbfc9

0xc0170, 2019-12-10:
When you updated, you use linker script unchanged? I would compare how the linker script changed - mainly for .ARM.exidx ?

Or it was not the problem previously but after the update, the section is at the different address and misaligned? would adding ` . = ALIGN(4); ` help there?

melvinvdb, 2019-12-10:
Well I didn't change any mbed-os sources.
When I compare mbed-os 12 linker script with mbed-os 14 I don't see any difference in the exidx.
Also the current version is not as mentioned above. It is:
`
    __exidx_start = .;
    .ARM.exidx :
    {
        *(.ARM.exidx* .gnu.linkonce.armexidx.*)
        . = ALIGN(8);
    } > FLASH
    __exidx_end = .;
`
As can be seen at mbed-os/targets/TARGET_NORDIC/TARGET_NRF5x/TARGET_NRF52/TARGET_MCU_NRF52832/device/TOOLCHAIN_GCC_ARM/NRF52832.ld

melvinvdb, 2019-12-10:
Well actually changing 
```
. = ALIGN(8);
```
To
```
. = ALIGN(4);
```
in: https://github.com/ARMmbed/mbed-os/blob/master/targets/TARGET_NORDIC/TARGET_NRF5x/TARGET_NRF52/TARGET_MCU_NRF52832/device/TOOLCHAIN_GCC_ARM/NRF52832.ld
fixed the issue!

0xc0170, 2019-12-11:
cc @desmond-blue

0xc0170, 2019-12-11:
I chose the wrong linker above (another NRF5x I assume), that does not even have the sections aligned.

I think we found the offender: https://github.com/ARMmbed/mbed-os/commit/61404e62d882aac53afeb0b40de22119d787a927

This changed it to be aligned on 8 byte boundary. I am now wondering what is causing this alignment problem, related to softdevice?

desmond-blue, 2019-12-11:
I don't think it's related to SoftDevice, I also tried to use `_ALIGN(4)` in `.ARM.exidx` section, and got this map.

```
.ARM.extab     0x000418dc       0x18 c:/program files (x86)/gnu tools arm embedded/9 2019-q4-major/bin/../lib/gcc/arm-none-eabi/9.2.1/thumb/v7e-m+fp/softfp\libgcc.a(pr-support.o)
                0x000418f8                . = ALIGN (0x8)
 *fill*         0x000418f4        0x4 
                0x000418f8                __exidx_start = .

.nrf_queue      0x000418f8       0x14
 .nrf_queue     0x000418f8       0x14 BUILD/BLE_Button/mbed-os/targets/TARGET_NORDIC/TARGET_NRF5x/TARGET_SDK_15_0/integration/nrfx/legacy/nrf_drv_rng.o
                0x000418f8                m_rand_pool

.ARM.exidx      0x0004190c      0x1d8
 *(.ARM.exidx* .gnu.linkonce.armexidx.*)
```

The `.nrf_queue` changes the alignment, but I have no idea why this be placed here. Perhaps some code size optimization?

0xc0170, 2020-02-24:
@melvinvdb Any new findings? Or you stayed with align4?

melvinvdb, 2020-02-25:
I just stayed with align(4). 
Seems to me that not many people use a NRF chip with MBed OS otherwise this would have been found earlier. Unfortunately I have to many problems with using MBed OS with the NRF52 so I'm thinking of moving away...

0xc0170, 2020-08-21:
As there has been no further update since February, I'll close this as "won't fix".

closed 2020-08-21 by 0xc0170