ARMmbed/mbed-os #12077: DNS fails with Gemalto EMS31-US modem due to Mbed Gemalto Cellular driver

jporter4, 2019-12-10:
### Description of defect
When the Gemalto modem receives socket data and sends it to the Gemalto cellular driver, the driver attempts to separate modem framing from the socket data.  This fails in the case of DNS.  Single-stepping through the code using the IAR debugger, I saw that the function "nsapi_dns_query_multiple" in file "nsapi_dns.cpp" was attempting to parse the following buffer as if it were a DNS response:
<0d><0a>^SISR: 0,96,0,"8.8.8.8:53"<0d><0a><00><01><81><80><00><01><00><04><00><00><00><00> etc.
In fact, the first 30 bytes are modem framing, and should have been stripped out by the Gemalto cellular driver.  The actual DNS response <00><01><81><80> etc. begins at the 31st byte.

#### Target(s) affected by this defect ?
Any target with a Gemalto EMS31-US cellular modem attached.  I also have a target that has a Gemalto ELS61-US modem, but I did not observe this behavior there.  I suspect strongly that the driver was only tested against ELS61-US.

#### Toolchain(s) (name and version) displaying this defect ?
I'm using IAR EWB for ARM 8.40. The issue should be observed regardless of which toolchain is used however.

#### What version of Mbed-os are you using (tag or sha) ?
5.14.1
#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
IAR EWB for ARM 8.40

#### How is this defect reproduced ? 
Build for GEMALTO_CINTERION (edit "mbed-os\features\cellular\framework\targets\GEMALTO\CINTERION\mbed_lib.json", set "provide-default" value to true). The simple TCP client example "https://os.mbed.com/teams/mbed-os-examples/code/mbed-os-example-sockets/docs/tip/main_8cpp_source.html" works nicely, just replacing Ethernet with Gemalto cellular.

AriParkkila, 2019-12-12:
@jporter4 Can you get a log with `"mbed-trace.enable": true` and `"cellular.debug-at": true` in the mbed_app.json?

jporter4, 2019-12-12:
Here it is.
[dns_log.txt](https://github.com/ARMmbed/mbed-os/files/3958076/dns_log.txt)

jporter4, 2019-12-12:
Hi,
I just uploaded the log file.
Thanks

Confidentiality Notice: This electronic mail transmission is intended for the use of the individual or entity to which it is addressed and may contain confidential and/or proprietary information belonging to the sender. If you are not the intended recipient, you are hereby notified that any disclosure, use, copying, distribution, or the taking of any action in reliance on the contents of this information is strictly prohibited. If you have received this transmission in error, please notify the sender immediately by e-mail and delete the original message. Thank you for your cooperation.

AriParkkila, 2019-12-13:
@jporter4 Thanks for a log. EMS31 response seems incorrect here: _If the currently read datagram is smaller than the number of bytes requested by the host the <remainUdpPacketLength>
parameter is omitted._
```
AT^SISR=1,512
^SISR: 1,204,0,"130.244.127.161:53"
```
Could you check the modem has the latest firmware, that might fix?

jporter4, 2019-12-18:
Hi Ari,
My firmware version is:
Cinterion
EMS31-US
REVISION 5.1.2.0a
A-REVISION 01.001.00

I’ve requested support from Gemalto to identify whether or not this is the latest firmware version, and if not, how to update it.
I’ll report back when I have an answer.
Thanks
Jack

jporter4, 2019-12-26:
@AriParkkila Gemalto support confirmed that the firmware version I reported above is the latest.
Thanks
Jack

AriParkkila, 2019-12-30:
@jporter4 Can you still check if `^SISR` response (in your AT command manual version) was changed to not omit [remainUdpPacketLength](https://github.com/ARMmbed/mbed-os/issues/12077#issuecomment-565398381).

jporter4, 2019-12-30:
@AriParkkila Could you clarify what you're asking me to do ?  Check something in the AT manual ?  in the driver code ?  in my device logs ?  Thanks.

AriParkkila, 2019-12-30:
@jporter4 The current implementation seems to be up to the EMS31 AT command manual v4.9.5, but there might be a more recent version? As a workaround you might try to change [this line to true](https://github.com/ARMmbed/mbed-os/blob/5f495db06bd6c6e0ac0adf001be57f1d724f9aa1/features/cellular/framework/targets/GEMALTO/CINTERION/GEMALTO_CINTERION_CellularStack.cpp#L437).

jporter4, 2019-12-30:
@AriParkkila  My AT manual is EMS31-US_ATC_V5.1.2.0a.  I got it from Gemalto support earlier this year.

AriParkkila, 2019-12-30:
@jporter4 Can you check if it still requires that _If the currently read datagram is smaller than the number of bytes requested by the host the <remainUdpPacketLength>
parameter is omitted._

jporter4, 2019-12-30:
@AriParkkila Yes, that quote appears in section 9.8 "AT^SISR Internet Service Read Data"

AriParkkila, 2019-12-30:
@jporter4 Thanks, I think this should be fixed in EMS31. Can you try if the [workaround](https://github.com/ARMmbed/mbed-os/issues/12077#issuecomment-569632829) works.

jporter4, 2019-12-30:
So you want me to change line 437 of the Gemalto driver from
 "if (len >= (nsapi_size_or_error_t)size) {" to
 "if (true) {" 

I'll try it and report back.

jporter4, 2019-12-31:
@AriParkkila 
I changed the line (it was actually line 434 in version 5.15.0, not line 437) to "if (true) {", 
rebuilt with trace enabled and downloaded to my target and ran it.  
It never even got to that code.  
It still fails earlier for using connection id 0, which is legal for the ELS61 but illegal for the EMS31, as I reported in issue #11760.  That issue has been closed, and I'm not sure why, as it's obviously not been fixed in version 5.15.0.
Do you have a target board with an EMS31 that you can test the driver on ?

[INFO][CELL]: AT TX (20): AT^SISS=1,"conId",0<cr>
[INFO][CELL]: AT RX (18): <cr><ln>
                          +CME ERROR: 50<cr><ln>
[WARN][CELL]: AT error code 50
[DBG ][CELL]: AT error -3012  <== "failure interfacing with the network processor"

jporter4, 2019-12-31:
@AriParkkila 
I jammed "connection_profile_id" to 1 to get past the previous error, rebuilt etc.  Now it times out reading the DNS response.  Here's my trace:

[DBG ][CELL]: Cinterion open 1 (err 0)
[DBG ][CELL]: AT OoB readable 1, len 3
[INFO][CELL]: AT RX (10): ^SIS: 1,5<cr><ln>
[DBG ][CELL]: AT OoB done
[INFO][CELL]: AT TX (34): AT^SISW=1,32,0,"10.105.16.254:53"<cr>
[INFO][CELL]: AT RX (17): <cr><ln>
                          ^SISW: 1,32,0<cr><ln>
[INFO][CELL]: AT TX (32): <00><01><01><00><00><01><00><00><00><00><00><00><03>www<06>google<03>com<00><00><01><00><01>
[INFO][CELL]: AT RX (20): <cr><ln>
                          OK<cr><ln>
                          <cr><ln>
                          ^SISW: 1,1<cr><ln>
[INFO][CELL]: Socket 1 sent 32 bytes to 10.105.16.254 port 53
[DBG ][CELL]: AT OoB readable 0, len 14
[DBG ][CELL]: Data can be written to socket 1
[DBG ][CELL]: AT OoB done
[DBG ][CELL]: CALL process_oob
[DBG ][CELL]: Socekt 1 recv would block
[DBG ][CELL]: CALL process_oob
[DBG ][CELL]: Socekt 1 recv would block
[DBG ][CELL]: AT OoB readable 1, len 0
[INFO][CELL]: AT RX (14): <cr><ln>
                          ^SISR: 1,1<cr><ln>
[DBG ][CELL]: Data is available from socket 1
[DBG ][CELL]: AT OoB done
[INFO][CELL]: AT TX (14): AT^SISR=1,512<cr>
[INFO][CELL]: AT RX (98): <cr><ln>
                          ^SISR: 1,62,0,"10.105.16.254:53"<cr><ln>
                          <00><01><81><80><00><01><00><01><00><00><00><00><03>www<06>google<03>com<00><00><01><00><01><03>www<06>google<03>com<00><00><01><00><01><00><00><00>?<00><04><AC><D9><A8><C4>
[WARN][CELL]: AT timeout
[DBG ][CELL]: AT error -3012
[ERR ][CELL]: Socket 1 recv error -3012  <== "failure interfacing with the network processor"
[DBG ][CELL]: Cinterion close 1
[INFO][CELL]: AT TX (10): AT^SISC=1<cr>

AriParkkila, 2019-12-31:
@jporter4 Unfortunately, I don't have EMS31 here so I need to rely just on its AT specification. The `_cid` problem is fixed in #12131 (your workaround seems fine). AT timeout seems like it has dismissed characters due to tracing (could try without tracing enabled).

jporter4, 2019-12-31:
@AriParkkila I didn't think that you could have since there doesn't seem to be any Arduino shield available with an EMS31.  The nearest thing I could find is the Cinterion Connect which has an ELS61.
Anyway, I turned off AT tracing and rebuilt etc.  It still times out on the DNS response:

[INFO][DNS ]: Host www.google.com
[INFO][CELL]: Socket 1 open
[INFO][DNS ]: Query server 10.105.16.254
[DBG ][CELL]: Cinterion open 1 (err 0)
[DBG ][CELL]: AT OoB readable 1, len 1
[DBG ][CELL]: AT OoB done
[INFO][CELL]: Socket 1 sent 32 bytes to 10.105.16.254 port 53
[DBG ][CELL]: AT OoB readable 1, len 0
[DBG ][CELL]: Data can be written to socket 1
[DBG ][CELL]: AT OoB done
[DBG ][CELL]: CALL process_oob
[DBG ][CELL]: Socekt 1 recv would block
[DBG ][CELL]: CALL process_oob
[DBG ][CELL]: Socekt 1 recv would block
[DBG ][CELL]: AT OoB readable 1, len 0
[DBG ][CELL]: Data is available from socket 1
[DBG ][CELL]: AT OoB done
[WARN][CELL]: AT timeout
[DBG ][CELL]: AT error -3012
[DBG ][CELL]: AT error -3012
[ERR ][CELL]: Socket 1 recv error -3012
[DBG ][CELL]: Cinterion close 1
[INFO][CELL]: Socket 1 closed

Regarding the byte count in the previous DNS response when tracing was enabled:
When I pasted the previous trace into the comment form, the website removed some embedded <cr> and <ln> bytes.
  The "AT RX (98)" indicates 98 bytes to follow.  ^SISR: 1,62,0,"10.105.16.254:53" is

jporter4, 2019-12-31:
@AriParkkila I tried one other thing to provide more info.  I reverted to the original 5.15.0 version of the file, but with "connection_profile_id" jammed to 1 to get past the AT error code 50; i.e. I backed out your "if (true)" change from the previous build.  I've attached the trace, which shows that the driver read the entire 104 byte packet which includes the modem framing ("^SISR: 1,62,0," etc.) and the actual 62-byte DNS response (<00><01><81><80> etc.).  But, the 62 bytes that function "nsapi_dns_query_multiple" gets from the driver starts with the modem framing rather than at the start of the 62-byte DNS response.

[scrap2.txt](https://github.com/ARMmbed/mbed-os/files/4013378/scrap2.txt)

AriParkkila, 2020-01-03:
@jporter4 The first response AT timeout is probably from [read_bytes](https://github.com/ARMmbed/mbed-os/blob/de798c4f379adbcf282199ba06553c0fa644cc8b/features/cellular/framework/targets/GEMALTO/CINTERION/GEMALTO_CINTERION_CellularStack.cpp#L483) expecting to get more bytes than readable.

The second response `^SISR: 1,62,0,"10.105.16.254:53"` still has that `remainUdpPacketLength` (it should be dropped "If the currently read datagram is smaller than the number of bytes requested by the host") so you need to modify the driver to read that in any case or the DNS response has incorrectly AT response.

jporter4, 2020-01-03:
@AriParkkila Regarding your statement "you need to modify the driver ..." - are you planning on fixing the driver for EMS31, are you expecting that I will ?  Do you work for Mbed ?  Or are you and end user ?  I want to make sure I understand the game plan here.  Thanks

AriParkkila, 2020-01-07:
@jporter4 The EMS31 specification seems to differ here from modem functionality. Which one we should follow here? Probably the specification as the firmware update later might fix the modem to comply with the specification. Do you know if the manufacturer has provided some update or errata on the EMS31 AT specification regarding the `remainUdpPacketLength` parameter? Mbed OS is open source and contributions are important part of the ecosystem, so you are most welcome to make a pull request to be reviewed by Mbed OS maintainers (I'm not one).

closed 2020-02-25 by ciarmcom