ARMmbed/mbed-os #12084: Astyle issue for PSA auto-generated files during release

jainvikas8, 2019-12-11:
### Astyle issue for PSA auto-generated files during release

#### Target(s) affected by this defect ?

LPC55S69_S
ARM_MUSCA_A1_S

#### Toolchain(s) (name and version) displaying this defect ?

ARMC6 and GCC_ARM

#### What version of Mbed-os are you using (tag or sha)?

https://github.com/ARMmbed/mbed-os/pull/12082

#### How is this defect reproduced? 

During the release - https://github.com/ARMmbed/mbed-os/commit/5d947032ba7a0c538667a715fd47d08953a92642 was done, to fix the issue.

`python tools/psa/release.py -t <tool> -m <target> --commit`

0xc0170, 2019-12-11:
I was checking the template file and how is generated. I did not find out why only comments were misaligned. The template file has it correct.

jainvikas8, 2019-12-11:
> I was checking the template file and how is generated. I did not find out why only comments were misaligned. The template file has it correct.

Yes, it does, but it seems this is the line which generates that astyle issue - https://github.com/ARMmbed/mbed-os/blob/master/tools/psa/templates/psa_setup.c.tpl#L189

PR #12085 (2019-12-16): PSA setup template changes.
Fixes #12084

### Summary of changes 
Astyle issue for PSA auto-generated files during release therefore removed spaces for the problematic line.
commit 4a84e146 (2019-12-16): PSA setup template changes.

Ideally, there is nothing wrong, but it seems to produce AStyle
error when release script runs for PSA Auto-generated files.
Therefore removed the spaces from the problematic line.

Fixes #12084
closed 2019-12-16 by 0xc0170