ARMmbed/mbed-os #12096: bare metal > sleep function locks need removing?

star297, 2019-12-12:
By virtue of we're not using RTOS here, surely sleep locking functions should be removed?

I hit this problem using RawSerial with an IRQ buffer after that I can no longer enter sleep (deepsleep), just returns immediately.

I have tried using:
 `mbed_file_handle(STDIN_FILENO)->enable_input(false);`

`pc.printf("Deep sleep allowed: %i\r\n\n", sleep_manager_can_deep_sleep());` tells me the Deep sleep is locked out. 
I had this issue in OS2 so tried OS5 bare metal and got the same results. Clearly the Target I'm using is base line. Prefer to use Online Mbed but can move this to Studio if there is anything I can try that can't be handled in the mbed_app.json file

#### Target(s) affected by this defect ?
STM Nucleo-F030R8 and possibly all STM targets.

#### Toolchain(s) (name and version) displaying this defect ?
Mbed online
#### What version of Mbed-os are you using (tag or sha) ?
OS v5.14
#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
Mbed online

star297, 2019-12-13:
Calling `hal_deepsleep();` instead of `sleep();` does put the MCU into deep sleep. However I do have to use the following GPIO initialize code to get it to around 3uA.

`RCC->AHBENR |= (RCC_AHBENR_GPIOAEN | RCC_AHBENR_GPIOBEN | RCC_AHBENR_GPIOCEN |` 
                             `RCC_AHBENR_GPIODEN | RCC_AHBENR_GPIOFEN);`

    GPIO_InitTypeDef GPIO_InitStruct;
    // All other ports are analog input mode
    GPIO_InitStruct.Pin = GPIO_PIN_All;
    GPIO_InitStruct.Mode = GPIO_MODE_ANALOG;
    GPIO_InitStruct.Pull = GPIO_NOPULL;
    GPIO_InitStruct.Speed = GPIO_SPEED_FREQ_LOW;
    HAL_GPIO_Init(GPIOA, &GPIO_InitStruct);
    HAL_GPIO_Init(GPIOB, &GPIO_InitStruct);
    HAL_GPIO_Init(GPIOC, &GPIO_InitStruct);
    HAL_GPIO_Init(GPIOD, &GPIO_InitStruct);
    HAL_GPIO_Init(GPIOF, &GPIO_InitStruct);

    RCC->AHBENR &= ~(RCC_AHBENR_GPIOAEN | RCC_AHBENR_GPIOBEN |RCC_AHBENR_GPIOCEN |
                     RCC_AHBENR_GPIODEN | RCC_AHBENR_GPIOFEN);

kjbracey-arm, 2019-12-19:
> By virtue of we're not using RTOS here, surely sleep locking functions should be removed?

Deep sleep locking does not depend on RTOS or not at all.

The system assumes that a serial interrupt handler will not work in deep sleep (as that's the HAL specification), so deep sleep is locked as long as it's attached.

Deep sleep can be temporarily unlocked by detaching the interrupt handler temporarily, or `enable_input(false)` in 5.15.

If you had a platform where the serial actually did work in deep sleep, then at the minute you'd need to modify `SerialBase` to stop it locking deep sleep. But it's likely wake-up from deep sleep would be slow enough that you'd lose the data that woke you. You'd probably need a special signalling protocol such that there was a "wake" byte after which you locked deep sleep to receive following.

Or you'd need hardware flow control signals that managed to gate the sender correctly to wait for your wake.

star297, 2019-12-20:
Thanks Kevin, I thought this would be a little complex, I was hoping that bare metal would be more 'bare' with little or no automated processes going on.

I understand the UART interrupt situation on 'wakeup', I send a dummy byte over Serial to do that before data transmission (I have worked a bit with Xbee).

I did try with no effect to detach the Serial with:
`esp2.attach(NULL);`
and
 ` __disable_irq();`

> Deep sleep can be temporarily unlocked by detaching the interrupt handler temporarily, or enable_input(false) in 5.15.

I couldn't get this working with the Online (rev. 5.15) and Studio gave me this compile error when upadting from rev. 5.14.2 (using target Nucleo-F030R8)

> [Error] mbed_rtx.h@137,2: "INITIAL_SP is not defined for this target in the mbed_rtx.h file"
> [ERROR] In file included from .\mbed-os\platform\source\mbed_error.c:30:
> ./mbed-os/targets/TARGET_STM\mbed_rtx.h:137:2: error: "INITIAL_SP is not defined for this target 
> in the mbed_rtx.h file"
> 
> #error "INITIAL_SP is not defined for this target in the mbed_rtx.h file"

I would suggest to close this ticket for now. We know that we need bare metal to be more 'bare' so perhaps the sleep locking function could be looked at another day.
For now I can still use earlier versions of Mbed2 which also save around a third of Flash/RAM resources over OS5 bare metal.

kjbracey-arm, 2019-12-20:
> I was hoping that bare metal would be more 'bare' with little or no automated processes going on.

There really isn't much going on. The deep sleep locking is just a reference count that affects the generic `sleep` function. If you really know what you're doing, you can call `hal_deepsleep()` directly, ignoring any lock reference count. That should be called from inside `core_util_critical_section_enter()` (or just `__disable_irq()`, if a simple device).

Still though, I don't know why deep sleep is being locked. `sleep_manager_can_deep_sleep()` should return true if called at the start of `main` (and you've got no static C++ objects inited pre-main that do stuff).

Only thing I can think of is that you may have turned on `platform.stdio-buffered-serial`, so the console is buffered, and has an IRQ handler? That would need its input disabled with `mbed_file_handle(FILENO_STDIN)->enable_input(false)`.

kjbracey-arm, 2019-12-20:
@ARMmbed/team-st-mcd, in case it's a target-specific issue.

MarceloSalazar, 2020-01-10:
@ARMmbed/team-st-mcd can you please have a look and comment?

LMESTM, 2020-01-10:
My 2 cents: on our platforms with lp_ticker support , we need to a deep sleep lock very regularly at low level:
https://github.com/ARMmbed/mbed-os/blob/master/targets/TARGET_STM/lp_ticker.c#L325

You'll need some very short period of inactivity (~100µs) before the lock is released. 
Not sure if this is the lock you're seeing though ... you may want to check the "sleep_manager_locked" variable at the same time ...

LMESTM, 2020-01-21:
@star297 could you find out the root cause of what driver is locking deep sleep ?

If not yet, you should look into this table here: 
https://github.com/ARMmbed/mbed-os/blob/master/platform/source/mbed_power_mgmt.c#L100

Please debug your app and look at this table at the time you try to enter sleep() and see what driver is locking it ...

star297, 2020-01-22:
After a lot of digging around I found I could put:

> "macros": [        
>        "MBED_SLEEP_TRACING_ENABLED=1"
>    ], 

in my mbed_app.json file that pointed to an 'un-stopped' Timer right in the middle of my Serial ring buffer function.
This information is in OS docs but not clear as how to implement it and tbh I wrongly assumed baremetal would exclude this locking process.

Stopping that Timer and calling attach(NULL) for the Serial allows the sleep() function to go to the Deep Sleep level as it should.

However I'm not convinced we need this locking function in baremetal.

closed 2020-01-22 by star297

kjbracey-arm, 2020-01-23:
> However I'm not convinced we need this locking function in baremetal.

This really has nothing to do with RTOS vs bare metal.

The sleep locking is because the generic HAL specification says that the us_ticker (which backs `Timer`) and the serial_api do not function in deep sleep.

Therefore when you request `sleep()` it chooses shallow sleep to keep your active timer and serial operational.

RTOS idle absolutely must obey that lock to avoid serial loss in normal operation. Idle must be transparent.

But it could be that on a certain target you could get away with calling `hal_deepsleep` with a timer and serial port active, and during that the devices merely "freeze" rather than lose state. And then you could do that as an application-initiated "suspend". But that would still be possible in both RTOS and non-RTOS.

But it's better to just acknowledge that you don't need those devices, release them, then request sleep. Then that would work on any Mbed OS platform.

kjbracey-arm, 2020-01-23:
And I guess then given that application behaviour, where you know `sleep` will always be called with no high-power devices active, you could reduce the "lock count" to a develop/debug function only. Simply assert that the lock count is 0 in `sleep` before entering `hal_deepsleep`, and in release builds, eliminate the counter.

Ah, except that wouldn't work for the "internal" shallow sleeps we use on the way into and out of deep sleep, eg where we schedule a timed deep sleep wake early to allow for latency, then go back to shallow sleep until the precise wake time.

kjbracey-arm, 2020-01-23:
Further note - as of current Mbed OS master (not sure if it made 5.15) you can do `enable_input(false)` and/or `enable_output(false)` on a `Serial` to suspend it. They temporarily unhooks any TX/RX interrupt handler and if you do both, the HAL is told to power down the serial hardware, which may save more power than just detaching the interrupt.

star297, 2020-01-23:
Thanks Kevin, makes a lot more sense now, my concern is when Mbed OS absorbs Mbed Classic (v2) at some point. This extra processing is too much for the sub 64k flash MCU's so I've been looking at what can be stripped out to make them usable.