ARMmbed/mbed-os #12102: UBLOX G350 get local ip fails

pilotak, 2019-12-13:
### Description of defect
After connecting to internet on SARA G350, there is command to send to get a local IP address but it fails all the time due to a wrong parameter `-1` 

This will rise a problem when you write into a new socket created straight after cb received that you are connected (same time `CGPADDR` is called) - i'm expecting `NSAPI_ERROR_OK`  but getting a response `-3012` but actually this is not a response from my socket but response to `+CGPADDR` call.
The workaround is to delay opening a socket for 2 sec because this is a timeout of get ip function.

Is this function really necessary on UBLOX AT stack?
```
[INFO][CELL]: AT TX ( 9): AT+UPSDA=
[INFO][CELL]: AT TX ( 1): 0
[INFO][CELL]: AT TX ( 1): ,
[INFO][CELL]: AT TX ( 1): 3
[INFO][CELL]: AT TX ( 1): <cr>
[INFO][CELL]: AT RX ( 5): <cr><ln>OK<cr>
[INFO][CELL]: AT RX ( 1): <ln>
[INFO][CELL]: AT TX ( 9): AT+UPSND=
[INFO][CELL]: AT TX ( 1): 0
[INFO][CELL]: AT TX ( 1): ,
[INFO][CELL]: AT TX ( 1): 8
[INFO][CELL]: AT TX ( 1): <cr>
[INFO][CELL]: AT RX ( 5): <cr><ln>+UP
[INFO][CELL]: AT RX (18): SND: 0,8,1<cr><ln><cr><ln>OK<cr><ln>

[INFO][CELL]: AT TX (11): AT+CGPADDR=
[INFO][CELL]: AT TX ( 2): -1
[INFO][CELL]: AT TX ( 1): <cr>
[INFO][CELL]: AT RX (10): <cr><ln>+CME ERR
[INFO][CELL]: AT RX ( 7): OR: 4<cr><ln>
[WARN][CELL]: AT error code 4
[DBG ][CELL]: AT error -3012
[INFO][CELL]: AT TX (13): AT+CMGL="ALL"
[INFO][CELL]: AT TX ( 1): <cr>
[INFO][CELL]: AT RX ( 4): <cr><ln>OK
[INFO][CELL]: AT RX ( 2): <cr><ln>
[INFO][CELL]: AT TX (11): AT+CGPADDR=
[INFO][CELL]: AT TX ( 2): -1
[INFO][CELL]: AT TX ( 1): <cr>
[INFO][CELL]: AT RX ( 5): <cr><ln>+CM
[INFO][CELL]: AT RX (12): E ERROR: 4<cr><ln>
[WARN][CELL]: AT error code 4
[DBG ][CELL]: AT error -3012
[INFO][CELL]: AT TX (11): AT+CGPADDR=
[INFO][CELL]: AT TX ( 2): -1
[INFO][CELL]: AT TX ( 1): <cr>
[INFO][CELL]: AT RX ( 7): <cr><ln>+CME
[INFO][CELL]: AT RX (10): ERROR: 4<cr><ln>
[WARN][CELL]: AT error code 4
[DBG ][CELL]: AT error -3012
[INFO][CELL]: AT TX (11): AT+CGPADDR=
[INFO][CELL]: AT TX ( 2): -1
[INFO][CELL]: AT TX ( 1): <cr>
[INFO][CELL]: AT RX ( 1): <cr>
[INFO][CELL]: AT RX ( 2): <ln>+
[INFO][CELL]: AT RX (14): CME ERROR: 4<cr><ln>
[WARN][CELL]: AT error code 4
[DBG ][CELL]: AT error -3012
[INFO][CELL]: AT TX (11): AT+CGPADDR=
[INFO][CELL]: AT TX ( 2): -1
[INFO][CELL]: AT TX ( 1): <cr>
[INFO][CELL]: AT RX ( 7): <cr><ln>+CME
[INFO][CELL]: AT RX (10): ERROR: 4<cr><ln>
[WARN][CELL]: AT error code 4
[DBG ][CELL]: AT error -3012
[INFO][CELL]: AT TX (11): AT+CGPADDR=
[INFO][CELL]: AT TX ( 2): -1
[INFO][CELL]: AT TX ( 1): <cr>
[INFO][CELL]: AT RX (10): <cr><ln>+CME ERR
[INFO][CELL]: AT RX ( 7): OR: 4<cr><ln>
[WARN][CELL]: AT error code 4
[DBG ][CELL]: AT error -3012
[INFO][CELL]: AT TX (11): AT+CGPADDR=
[INFO][CELL]: AT TX ( 2): -1
[INFO][CELL]: AT TX ( 1): <cr>
[INFO][CELL]: AT RX ( 8): <cr><ln>+CME E
[INFO][CELL]: AT RX ( 9): RROR: 4<cr><ln>
[WARN][CELL]: AT error code 4
[DBG ][CELL]: AT error -3012
[INFO][CELL]: AT TX (11): AT+CGPADDR=
[INFO][CELL]: AT TX ( 2): -1
[INFO][CELL]: AT TX ( 1): <cr>
[INFO][CELL]: AT RX ( 3): <cr><ln>+
[INFO][CELL]: AT RX (14): CME ERROR: 4<cr><ln>
[WARN][CELL]: AT error code 4
[DBG ][CELL]: AT error -3012
[INFO][CELL]: AT TX (11): AT+CGPADDR=
[INFO][CELL]: AT TX ( 2): -1
[INFO][CELL]: AT TX ( 1): <cr>
[INFO][CELL]: AT RX ( 4): <cr><ln>+C
[INFO][CELL]: AT RX (13): ME ERROR: 4<cr><ln>
[WARN][CELL]: AT error code 4
[DBG ][CELL]: AT error -3012
[INFO][CELL]: AT TX (11): AT+CGPADDR=
[INFO][CELL]: AT TX ( 2): -1
[INFO][CELL]: AT TX ( 1): <cr>
[INFO][CELL]: AT RX ( 2): <cr><ln>
[INFO][CELL]: AT RX (15): +CME ERROR: 4<cr><ln>
[WARN][CELL]: AT error code 4
[DBG ][CELL]: AT error -3012
[WARN][CELL]: Connected but no local ip address
```

#### Target(s) affected by this defect ?
Ublox SARA G350

#### Toolchain(s) (name and version) displaying this defect ?
GCC_ARM

#### What version of Mbed-os are you using (tag or sha) ?
c098bdff184122cd523ff875ba5e2c67cd581bf4

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
CLI 1.10.1

#### How is this defect reproduced ? 
descripbed above

AnttiKauppila, 2020-01-07:
It seems that CellularContext uses the default context ID (=-1) and therefore you will get that error.
CellularStack sets the _cid in do_connect() phase, which is overridden by UBlox driver.
I think this is the culprit: (Lines 55&56) https://github.com/ARMmbed/mbed-os/blob/master/features/cellular/framework/targets/UBLOX/AT/UBLOX_AT_CellularContext.cpp#L55
In 41XM family drivers the _cid is handled correctly.

@ARMmbed/team-ublox Please investigate.

fahim-ublox, 2020-01-07:
Looks like after mbed-os 5.15 release, string-based API usage within mbed-os started to remove(i.e. get_ip_address). Are you getting this problem recently ? 

While PGM_NO_MBD-319 Internal ticket created to fix the issue.

closed 2020-01-09 by 0xc0170