ARMmbed/mbed-os #12103: ARMC5 build is failing mbed-os-5-15-rc2 for k64f in linux

saheerb, 2019-12-13:
### Description of defect

Build is failing in this combination:
mbed-os-5-15-rc2, 
application: mbed-os-example-blink
host: linux
toolchain: ARMC5 Version: 
Target: K64F

Full log file:

https://btools-build-results.s3-us-west-2.amazonaws.com/production/production-nightly/88/logs/compilation_target_k64f_toolchain_armc5_os_version_latest_host_platform_linux_example_baseline_blinky_all_hosts.txt

**This is a regression from mbed-os-5-15-rc1**

#### Target(s) affected by this defect ?
K64F

#### Toolchain(s) (name and version) displaying this defect ?
ARMC5, 

#### What version of Mbed-os are you using (tag or sha) ?
mbed-os-5-15-rc2

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

#### How is this defect reproduced ? 
mbed compile

jamesbeyond, 2019-12-13:
This is because of the static pinmap feature we introduced, it used `constexpr`  keyword in https://github.com/ARMmbed/mbed-os/blob/master/features/storage/kvstore/conf/kv_config.cpp#L54

the `constexpr` is part of C++14 standard which ARMC5 is not supporting, hence resulting in the issue

0xc0170, 2019-12-16:
Just making sure, @jamesbeyond 5.15 release notes contain this info about it : `STATIC_PINMAP_READY` targets wont compile with ARMC5 or similar?

cc @adbridge

kjbracey-arm, 2019-12-20:
Fixed by #12117 - closing.

closed 2019-12-20 by kjbracey-arm