ARMmbed/mbed-os #12104: STM32 can_init_freq() ignores frequency and uses 1000000 instead

bikeNomad, 2019-12-13:
### Description of defect

Attempts to create an `mbed::CAN` object passing in the CAN bit rate don't properly set the frequency. Instead, the CAN port is always opened at 100000 bps.

#### Target(s) affected by this defect ?

All STM32 parts with CAN peripherals.

#### Toolchain(s) (name and version) displaying this defect ?

All

#### What version of Mbed-os are you using (tag or sha) ?

`mbed-os-5.15`

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

mbed-cli 1.10.1

#### How is this defect reproduced ? 

Create an `mbed::CAN` object using the three-argument constructor and pass in a frequency argument other than 100000:

```c++
CAN myCAN(PIN_CAN_RX, PIN_CAN_TX, 500000);
```
This is due to `mbed-os/targets/TARGET_STM/can_api.c` containing this code:

```c++
void can_init_freq(can_t *obj, PinName rd, PinName td, int hz)
{
    CANName can_rd = (CANName)pinmap_peripheral(rd, PinMap_CAN_RD);
    CANName can_td = (CANName)pinmap_peripheral(td, PinMap_CAN_TD);
    int peripheral = (int) pinmap_merge(can_rd, can_td);

    int function_rd = (int)pinmap_find_function(rd, PinMap_CAN_RD);
    int function_td = (int)pinmap_find_function(td, PinMap_CAN_TD);

    const can_pinmap_t static_pinmap = {peripheral, rd, function_rd, td, function_td};

    CAN_INIT_FREQ_DIRECT(obj, &static_pinmap, 100000);
}
```
As you can see, the `hz` argument is ignored and `100000` is used instead.

0xc0170, 2019-12-16:
cc @ARMmbed/team-st-mcd

jeromecoutant, 2019-12-16:
Hi

@mprse

I understand that for CAN, there are 2 API for init:

CAN::CAN(PinName rd, PinName td) :
    can_init(&_can, rd, td);
CAN::CAN(PinName rd, PinName td, int hz) :
    can_init_freq(&_can, rd, td, hz);

Before #11892, in each target API:
- can_init_freq is setting CAN with the expected frequency
- can_init is only calling can_init_freq with the default frequecny

Now with #11892
- can_init is now calling CAN_INIT_DIRECT
- can_init_freq is now calling CAN_INIT_FREQ_DIRECT
We miss the frequency setting

mprse, 2019-12-16:
@jeromecoutant @bikeNomad Yes, this is a bug. 

The fix can be found here: PR https://github.com/ARMmbed/mbed-os/pull/12113. 
Please review.

jeromecoutant, 2020-01-21:
As #12113 is merged, we close close this one 
Thx

closed 2020-02-20 by 0xc0170