ARMmbed/mbed-os #12105: USB on Maple Mini board is not supported

rblaze, 2019-12-15:
### Description of defect
According to https://os.mbed.com/users/hudakz/code/MapleMini_USBSerial/, Maple mini board and clones have USB Serial support. But firmware says there is no driver.

```
$ cat main.cpp 
#include "USBSerial.h"
#include "mbed.h"

DigitalOut myled(LED1);
DigitalOut usbEn(PB_9); // Used for connecting/disconnecting the 1k5 resistor
                        // to/from the USB DP pin

int main() {
  usbEn = 0; // Keep the on-board 1k5 resistor connected to the USB DP pin

  printf("Starting usb\n");
  USBSerial usbSerial;

  while (1) {
    myled = !myled;
    printf("I am a serial port\n");               // 9600 bit/s
    usbSerial.printf("I am a USB serial port\n"); // 12 Mbit/s (USB full-speed)
    ThisThread::sleep_for(1000);
  }
}
```

```
[mbed] Working path "/home/blaze/dev/mbed/mapple_serial" (program)
[mbed] Detecting connected targets/boards to your system...
[mbed] Opening serial terminal to "ST_PLACEHOLDER"
--- Terminal on /dev/ttyACM0 - 9600,8,N,1 ---
Starting usb

++ MbedOS Error Info ++
Error Status: 0x80FF0100 Code: 256 Module: 255
Error Message: Fatal Run-time error
Location: 0x8002BFD
Error Value: 0x0
Current Thread: main  Id: 0x20000E50 Entry: 0x8003F8F StackSize: 0x1000 StackMem: 0x200015E8 SP: 0x200020CC 
For more info, visit: https://mbed.com/s/error?error=0x80FF0100&tgt=NUCLEO_F103RB
-- MbedOS Error Info --
This board does not have a hardware USB driver
```

Adding mbed-MapleMini library and calling `confSysClock()` doesn't help either.

#### Target(s) affected by this defect ?
NUCLEO_F103RB

#### Toolchain(s) (name and version) displaying this defect ?
GCC_ARM_PATH=/home/blaze/.local/gcc-arm-none-eabi-8-2019-q3-update/bin

#### What version of Mbed-os are you using (tag or sha) ?
```
commit cf4f12a123c05fcae83fc56d76442015cb8a39e9 (HEAD, tag: mbed-os-5.14.2, tag: latest, origin/mbed-os-5.14)
Merge: 679d24833a 918a0f5509
Author: Martin Kojtal <martin.kojtal@arm.com>
Date:   Wed Nov 20 14:13:16 2019 +0100

    Merge pull request #11894 from ARMmbed/release-candidate
    
    Release candidate for mbed-os-5.14.2
```

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
mbed-cli 1.10.1, but it hardly matter.

#### How is this defect reproduced ? 
1) Build program
2) Upload it to Maple Mini board
3) See console

0xc0170, 2019-12-16:
@rblaze the board referenced looks like it mentions Mbed 2. You can use their lib: https://os.mbed.com/users/hudakz/code/USBDevice_STM32F103/

Usb driver was updated with a rework for Mbed OS thus it does report this error. `NUCLEO_F103RB`  does not yet have usb support.

cc @ARMmbed/team-st-mcd

jeromecoutant, 2019-12-16:
USB support for mbed2 is no more maintained.

USB device support for STM32F1 and mbed-o is planned
Any help from community is welcome

0xc0170, 2020-02-24:
> USB device support for STM32F1 and mbed-o is planned
Any help from community is welcome

The question was answered as feature update needed. As this is not technical issue but rather feature request, any further requests, please use forum.mbed.com.

I'll close this.

closed 2020-02-24 by 0xc0170